zephyrproject-rtos/zephyr #43647: Bluetooth: LE multirole: connection as central is not totally unreferenced on disconnection

eloisesko, 2022-03-10:
**Describe the bug**
When doing BLE multirole and connected both to a central and a peripheral, if the peripheral disconnects, a reference to the peripheral connection still remains. Next connections to a peripheral will then fail because no more connections are available.

The extra reference is set by a call to `bt_le_adv_resume()` in `bt_conn_unref` in `deferred_work` (conn.c) [here](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.0.0/subsys/bluetooth/host/conn.c#L1577).

The issue does not happen if no central is connected to the device because the advertising flags are not in the same state (BT_ADV_PERSIST and BT_ADV_ENABLED are set, so bt_le_adv_resume returns immediately).

A workaround is to check the role of the connection to call bt_le_adv_resume() only for BT_CONN_ROLE_PERIPHERAL.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable CONFIG_BT_PERIPHERAL and CONFIG_BT_CENTRAL
2. Start advertising
3. Connect a central to the device
4. Connect the device to a peripheral
5. Disconnect the peripheral : check acl_conns, a reference still remains to the peripheral connection
6. Try to reconnect the peripheral : `bt_conn_le_create` fails with error -ENOMEM

**Expected behavior**
There should be no reference remaining to the perihperal connection after disconnection

**Impact**
It is impossible to reconnect to a peripheral.

carlescufi, 2022-03-10:
There are multiple possible causes for this, but this doesn´t look at first sight like a bug in the stack. Can you please post a simple application to reproduce this, in source code, somewhere on GitHub or even as a comment here?

This could be a reference counting bug in the application, please check the reference counting code in your app. Are you calling `bt_unref()` to the connection in 4. ?

eloisesko, 2022-03-10:
Yes, bt_conn_unref is called by the application on disconnection.

Before the call [here](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.0.0/subsys/bluetooth/host/conn.c#L1577), the reference counter equals  to 1. After the call, it is still equal to 1 (it is increased by bt_le_adv_resume())

I'll find some time to provide sample code.

carlescufi, 2022-03-10:
Can you make sure you describe the sequence in a way that is unambiguous?

I understand that:
You enable `CONFIG_BT_PERIPHERAL` and `CONFIG_BT_CENTRAL`, set `CONFIG_BT_MAX_CONN` to `2`.

1. Zephyr starts connectable advertising
2. Zephyr gets connected as a peripheral
3. Zephyr establishes a connection as a central
Now Zephyr is both a central and a peripheral
4. Zephyr disconnects the connection it has as a central
Now Zephyr is only a peripheral
5. Zephyr tries to establish a connection as a central, fails with `-ENOMEM`

Is this correct?

carlescufi, 2022-03-17:
@eloisesko could you please try using `BT_LE_ADV_OPT_ONE_TIME` when you start connectable advertising so that the stack doesn't automatically restart it after connection?

eloisesko, 2022-03-21:
Sorry for the late reply, I finally made a  [sample project](https://github.com/eloisesko/zephyr-multiperipheral) that reproduces the issue.
With `BT_LE_ADV_OPT_ONE_TIME` flag, the issue with connection references does not happen.

cvinayak, 2022-03-21:
When you do not use `BT_LE_ADV_OPT_ONE_TIME` and have `BT_MAX_CONN` > 1, then the broadcaster will restart connectable advertiser after a connection in peripheral role (in the local device) is established, this is to allow multiple peer central devices to be able to connect to the local device.

Such restarted connectable advertising depletes one of the two connection instances in your case, hence the attempt to create a connection to a remote peripheral device would fail with `-ENOMEM` when you have an active connection in peripheral role and a connectable advertising too.

carlescufi, 2022-03-21:
Closing as per the explanation above. If you have more questions feel free to reopen.

closed 2022-03-21 by carlescufi

eloisesko, 2022-10-17:
#51256 fixes the issue