zephyrproject-rtos/zephyr #43655: esp32c3: Connection fail loop

bgreggsmith, 2022-03-10:
**Describe the bug**
The sample code for connecting to wifi (samples/boards/esp32/wifi_station), works fine on esp32 but does not work on esp32c3. Instead of connecting,  a disconnect loop is obtained.

This is on a generic esp32c3 development board. Zephyr version 3.0.99.

This happens on a WPA1, WPA2, and Open access point. Building the same code for esp32 and running on a normal esp32 yields perfectly normal behaviour.

**To Reproduce**
Steps to reproduce the behavior:
1. make copy of sample samples/boards/esp32/wifi_station
2. edit prj.conf with details to connect to AP
3. build (west build -b esp32c3_devkitm -d build_dir wifi_fix --pristine)
5. flash generated image (west flash -d build_dir --esp-device /dev/ttyUSB0)

**Expected behavior**
Board connects to access point and get a DHCP address.

**Impact**
Unable to connect to wifi on esp32c3.

**Logs and console output**
Console output (the loop) on ESP32C3:
`
I (49) boot: ESP-IDF release/v4.3-120-gbcd7565ff 2nd stage bootloader
I (49) boot: compile time 15:06:09
I (50) boot: chip revision: 3
I (53) boot.esp32c3: SPI Speed      : 40MHz
I (58) boot.esp32c3: SPI Mode       : DIO
I (62) boot.esp32c3: SPI Flash Size : 2MB
I (67) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) boot_comm: chip revision: 3, min. application chip revision: 0
I (117) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (126) esp_image: segment 1: paddr=00010044 vaddr=3fcb5f40 size=01470h (  5232) load
I (135) esp_image: segment 2: paddr=000114bc vaddr=3fcb73b0 size=00594h (  1428) load
I (143) esp_image: segment 3: paddr=00011a58 vaddr=3fcb7948 size=0006ch (   108) load
I (151) esp_image: segment 4: paddr=00011acc vaddr=40380000 size=0e56ch ( 58732) load
I (173) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=11a10h ( 72208) map
I (189) esp_image: segment 6: paddr=00031a58 vaddr=4038e56c size=02ae0h ( 10976) load
I (192) esp_image: segment 7: paddr=00034540 vaddr=00000000 size=0bad8h ( 47832) 
I (205) esp_image: segment 8: paddr=00040020 vaddr=42020020 size=5009ch (327836) map
I (280) boot: Loaded app from partition at offset 0x10000
I (280) boot: Disabling RNG early entropy source...
pp rom version: 9387209
net80211 rom version: 9387209
I (12) wifi:wifi driver task: 3fc926c8, prio:2, stack:3584, core=0
I (12) wifi:wifi firmware version: 7e2c914
I (12) wifi:wifi certification version: v7.0
I (12) wifi:config NVS flash: disabled
I (12) wifi:config nano formating: disabled
I (12) wifi:Init data frame dynamic rx buffer num: 32
I (12) wifi:Init management frame dynamic rx buffer num: 32
I (12) wifi:Init management short buffer num: 32
I (12) wifi:Init dynamic tx buffer num: 32
I (12) wifi:Init static tx FG buffer num: 2
I (12) wifi:Init static rx buffer size: 1600
I (12) wifi:Init static rx buffer num: 10
I (12) wifi:Init dynamic rx buffer num: 32
I (12) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
I (12) wifi:set rx active PTI: 0, rx ack PTI: 0, and default PTI: 0
I (12) wifi:mode : softAP (21:58:a1:06:57:50)
I (12) wifi:Total power save buffer number: 16
I (12) wifi:Init max length of beacon: 752/752
I (12) wifi:Init max length of beacon: 752/752
*** Booting Zephyr OS build zephyr-v3.0.0-942-g0e5d438ee2ab  ***
I (12) wifi:mode : sta (21:58:a1:06:57:50)
I (16) wifi:enable tsf
[00:00:00.227,083] <inf> esp32_wifi: WIFI_EVENT_STA_START
I (1231) wifi:new:<11,0>, old:<1,1>, ap:<255,255>, sta:<11,0>, prof:1
I (1232) wifi:state: init -> auth (b0)
I (2232) wifi:state: auth -> init (200)
I (2233) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[00:00:02.441,507] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (4303) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (4303) wifi:state: init -> auth (b0)
[00:00:04.489,562] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (5304) wifi:state: auth -> init (200)
I (5304) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[00:00:05.513,192] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (7375) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (7375) wifi:state: init -> auth (b0)
[00:00:07.561,847] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (8376) wifi:state: auth -> init (200)
I (8377) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[00:00:08.585,222] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (10447) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (10447) wifi:state: init -> auth (b0)
[00:00:10.636,250] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (11448) wifi:state: auth -> init (200)
I (11449) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[00:00:11.657,236] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED

`
Same code, ESP32 output:
`
I (27) boot: ESP-IDF release/v4.3-120-gbcd7565ff 2nd stage bootloader
I (27) boot: compile time 15:01:25
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (111) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=01830h (  6192) load
I (121) esp_image: segment 2: paddr=0001187c vaddr=3ffb1830 size=0054ch (  1356) load
I (128) esp_image: segment 3: paddr=00011dd0 vaddr=40080000 size=0e268h ( 57960) load
I (161) esp_image: segment 4: paddr=00020040 vaddr=3f400040 size=12c9ch ( 76956) map
I (190) esp_image: segment 5: paddr=00032ce4 vaddr=4008e268 size=06364h ( 25444) load
I (201) esp_image: segment 6: paddr=00039050 vaddr=00000000 size=06fc8h ( 28616) 
I (212) esp_image: segment 7: paddr=00040020 vaddr=400d0020 size=407d0h (264144) map
I (322) boot: Loaded app from partition at offset 0x10000
I (323) boot: Disabling RNG early entropy source...
I (686) wifi:wifi driver task: 3ffcb288, prio:2, stack:3584, core=0
I (687) wifi:wifi firmware version: 7e2c914
I (687) wifi:wifi certification version: v7.0
I (689) wifi:config NVS flash: disabled
I (692) wifi:config nano formating: disabled
I (697) wifi:Init data frame dynamic rx buffer num: 32
I (701) wifi:Init management frame dynamic rx buffer num: 32
I (707) wifi:Init management short buffer num: 32
I (711) wifi:Init dynamic tx buffer num: 32
I (715) wifi:Init static rx buffer size: 1600
I (719) wifi:Init static rx buffer num: 10
I (723) wifi:Init dynamic rx buffer num: 32
I (728) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (870) wifi:mode : softAP (24:6f:28:2b:63:d5)
I (870) wifi:Total power save buffer number: 16
I (870) wifi:Init max length of beacon: 752/752
I (872) wifi:Init max length of beacon: 752/752
*** Booting Zephyr OS build zephyr-v3.0.0-942-g0e5d438ee2ab  ***
I (882) wifi:mode : sta (24:6f:28:2b:63:d4)
I (885) wifi:enable tsf
[00:00:00.889,000] <inf> esp32_wifi: WIFI_EVENT_STA_START
I (2102) wifi:new:<11,0>, old:<1,1>, ap:<255,255>, sta:<11,0>, prof:1
I (2102) wifi:state: init -> auth (b0)
I (2109) wifi:state: auth -> assoc (0)
I (2114) wifi:state: assoc -> run (10)
I (2114) wifi:connected with propus_open, aid = 1, channel 11, BW20, bssid = 96:83:c4:02:a1:c0
I (2116) wifi:security: Open Auth, phy: bgn, rssi: -45
I (2121) wifi:pm start, type: 1

I (2181) wifi:AP's beacon interval = 102400 us, DTIM period = 2
[00:00:02.124,000] <inf> esp32_wifi: WIFI_EVENT_STA_CONNECTED
[00:00:04.905,000] <inf> net_dhcpv4: Received: 10.30.1.189
[00:00:04.905,000] <inf> esp32_wifi_sta: Your address: 10.30.1.189
[00:00:04.905,000] <inf> esp32_wifi_sta: Lease time: 43200 seconds
[00:00:04.905,000] <inf> esp32_wifi_sta: Subnet: 255.255.255.0
[00:00:04.905,000] <inf> esp32_wifi_sta: Router: 10.30.1.1

`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (Zephyr 3.0.99 (0e5d438ee2ab38ca79d9c4375a9d9ca56bc8df2e, main), Espressif (1.24.0.123_64eb9ff-8.4.0))

sylvioalves, 2022-03-15:
@bgreggsmith, the log output in ESP32C3 kind of tells that the input password is wrong. It is trying to connect to the SSID but gets disconnected due to wrong password. Have you run `west update` as well?

bgreggsmith, 2022-03-23:
@sylvioalves I did run west update. Here's what I dont understand; the identical code and credentials when compiled for esp32 (as opposed to esp32c3) and flashed to an esp32 (as opposed to an esp32c3) connects just fine.

martinjaeger, 2022-03-23:
I'm experiencing exactly the same issue with ESP32-C3 and I double-checked that my password and SSID are set correctly.

As also mentioned in Discord, the `examples/wifi/getting_started/station` from ESP-IDF works perfectly fine with ESP32-C3, so there must be something wrong in Zephyr.

sylvioalves, 2022-03-23:
@bgreggsmith @martinjaeger I can't reproduce in here at all. Anyway, I'll try with other boards to evaluate it properly.

bgreggsmith, 2022-03-23:
@sylvioalves Took delivery of a new batch of boards today (These are ESP32-C3-WROOM-02, the old boards I was testing on were ESP-C3-32S) and these now work fine with the original binary.

Thanks for your time on this, but I think we can mark this as solved.

martinjaeger, 2022-03-23:
@bgreggsmith no, it's not solved for me. I will double-check with a different board as well.

martinjaeger, 2022-03-26:
I have tried this out with multiple boards now. The issue is very strange.

I've got two different ESP32-C3-DevKitM-1 boards. On one board it works fine, on the other one it fails with above disconnect loop, even after multiple resets, flash erase cycles, etc.

Both boards may be from different batches because I bought one from ebay just after it was released and the other one later from Digikey/Mouser. The working one is the XXN4 variant, the other the XXH4 variant, everything else including the ROM and 2nd stage bootloader is exactly the same:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x1894
load:0x403ce000,len:0x8c8
load:0x403d0000,len:0x2c14
SHA-256 comparison failed:
Calculated: 225bbcf7f9bb04ec6a55c4fbaaabe392f9533aadbb1782278721d172193382c4
Expected: b58235bbf573ed5ddd42c437c09ceea3d6d7ff25e092755761b016c13cdb17e4
Attempting to boot anyway...
entry 0x403ce000
<0x1b>[0;32mI (49) boot: ESP-IDF release/v4.3-151-g3373bdddb 2nd stage bootloader<0x1b>[0m
<0x1b>[0;32mI (50) boot: compile time 11:55:14<0x1b>[0m
...
```

Annoyingly, I've developed some custom boards (XXN4 module variant) which all fail to connect to the access point. They do work perfectly fine and immediately connect with most recent ESP-IDF (git describe: v5.0-dev-2137-g2f9d47c708) and `examples/wifi/getting_started/station`.

Any ideas?

bgreggsmith, 2022-03-26:
@martinjaeger This just got interesting enough for me to open the work laptop on a weekend. I thought it was just me and that I had a bunch of bad boards. 

I'm working on large(r) networks so I have about 20 of these things and I flashed the first 9 boards with the same firmware on Friday. 6 work fine, 3 are unable to connect to anything. Re-flashing them doesn't help, so I was under the impression I had 3 DOA boards or a very weird bug in my test firmware.

Temperatures are just normal bench conditions (about 20C?). Guess I'm not sending off some emails to RMA or tearing apart my work just yet. It did seem weird that espressif would have ~30% failure rate of boards.

bgreggsmith, 2022-03-26:
@martinjaeger So I have 3 sets of boards, one normal ESP32 with a ESP-WROOM-32 module of unknown origin (it's been in the lab a while from someone elses' order). This one works perfectly.

Then I have a pair of boards with the ESP-C3-32S module which has FCC ID 2ATPO-ESP-C3-X and no other markings both of which are affected by this connection bug. These have an Ai logo on them and not much else besides the CMIIT ID and FCC ID. The board the module is on is a "NodeMCU-Series". Being from amazon, it wouldn't surprise me theres some weirdness I'm in the dark about.

Finally I have a batch of 16 boards which have the ESP32-C3-WROOM-02 module on them. These are the only ones with that XXN4 marking. All these boards were bought recently off mouser as a single order. Of these I've unboxed 9 and 6 of them work perfectly, 3 are affected by this bug. These are the boards I really care about since I need 12 working ones and getting them out of the US into Germany has been a multi-month ordeal because apparently shipping a trace antenna to Germany is a threat to US national security so fixing this will be faster than getting more.

sylvioalves, 2022-03-27:
I was able to reproduce the issue as well. I'll check this.

martinjaeger, 2022-04-05:
@sylvioalves do you possibly have any update on this already? I'd be willing to support with narrowing down the issue, but as I'm not too familiar with the exact way how `espressif_hal` is integrated into Zephyr, some pointers where to look at would be helpful. And do you think that the planned upgrade of the HAL you mentioned in another PR could fix the problem?

bgreggsmith, 2022-04-07:
To seconds this, I don't have much experience messing with the insides of zephyr but in the next week or two this might be a blocker for me so I might also have to dig into whats going wrong, so a pointer on where to start for this issue might be helpful.

sylvioalves, 2022-04-07:
@martinjaeger I have started working on this a few days ago, but no success yet. Will ping you once fixed.

martinjaeger, 2022-04-08:
> @martinjaeger I have started working on this a few days ago, but no success yet. Will ping you once fixed.

Excellent, thanks a lot @sylvioalves!

bgreggsmith, 2022-05-27:
@omikronlambda sadly not, its been a blocker for me for a while now, so I'm probably going to have to spend more hours than I'd like figuring out where to even start since I've not had any hints either :(

sylvioalves, 2022-05-27:
@omikronlambda I could figure that out yet. I have been updating hal layer to bring the latest modifications and check if it solves it.

martinjaeger, 2022-05-28:
> I have been updating hal layer to bring the latest modifications and check if it solves it.

Is that HAL update already available upstream or did you only test it locally? I would be interested in testing as well, but don't know how to update the HAL manually.

omikronlambda, 2022-05-30:
I test HAL updates periodcally ('west update' with zephyr's 'revision:  main', then 'west espressif update'), but still no way.
Last days I've investigated this bug more, and for now I have a suspicion, that it's somehow related to pin initialization while boot up.
Possibly pins sleep modes, or pins direction setup at wrong moment. Can't explain, just a suspicion :)
I'll try to experiment with pin settings later.

bgreggsmith, 2022-05-30:
@sylvioalves I have a network of ESP32C3s which I thought were known good on the bench. I'd be happy to help test anything you think might help since I am seeing increasingly erratic behaviour of known good boards as the number of nodes increases.

omikronlambda, 2022-06-28:
Just check updates. Still the same :(

`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x188c
load:0x403ce000,len:0x8c8
load:0x403d0000,len:0x2c14
SHA-256 comparison failed:
Calculated: 366be7785f74dcc12cd6cf1bbdb600921880e5568db77a75db03e7feb364f604
Expected: e1df479d33055978287ab4bf9e101dc0e723c3b48b2e88461adf5f74e8162b60
Attempting to boot anyway...
entry 0x403ce000
I (49) boot: ESP-IDF release/v4.3-190-g0e6ea916f 2nd stage bootloader
I (50) boot: compile time 15:49:56
I (50) boot: chip revision: 3
I (53) boot.esp32c3: SPI Speed      : 40MHz
I (58) boot.esp32c3: SPI Mode       : DIO
I (63) boot.esp32c3: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) boot_comm: chip revision: 3, min. application chip revision: 0
I (117) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)
I (126) esp_image: segment 1: paddr=00010044 vaddr=3fcc5690 size=01464h (  5220) load
I (135) esp_image: segment 2: paddr=000114b0 vaddr=3fcc6af4 size=004fch (  1276) load
I (143) esp_image: segment 3: paddr=000119b4 vaddr=40380000 size=0e684h ( 59012) load
I (165) esp_image: segment 4: paddr=00020040 vaddr=3c000040 size=11848h ( 71752) map
I (181) esp_image: segment 5: paddr=00031890 vaddr=4038e684 size=01658h (  5720) load
I (182) esp_image: segment 6: paddr=00032ef0 vaddr=00000000 size=0d128h ( 53544)
I (198) esp_image: segment 7: paddr=00040020 vaddr=42020020 size=503c8h (328648) map
I (273) boot: Loaded app from partition at offset 0x10000
I (273) boot: Disabling RNG early entropy source...
▒pp rom version: 9387209
net80211 rom version: 9387209
I (3) wifi:wifi driver task: 3fc91348, prio:2, stack:3584, core=0
I (4) wifi:wifi firmware version: 7e2c914
I (4) wifi:wifi certification version: v7.0
I (7) wifi:config NVS flash: disabled
I (10) wifi:config nano formating: disabled
I (14) wifi:Init data frame dynamic rx buffer num: 32
I (19) wifi:Init management frame dynamic rx buffer num: 32
I (24) wifi:Init management short buffer num: 32
I (29) wifi:Init dynamic tx buffer num: 32
I (32) wifi:Init static tx FG buffer num: 2
I (36) wifi:Init static rx buffer size: 1600
I (41) wifi:Init static rx buffer num: 10
I (44) wifi:Init dynamic rx buffer num: 32
I (52) wifi:set rx active PTI: 0, rx ack PTI: 0, and default PTI: 0
I (52) wifi:mode : softAP (a1:f8:a1:03:8f:04)
I (52) wifi:Total power save buffer number: 16
I (52) wifi:Init max length of beacon: 752/752
I (52) wifi:Init max length of beacon: 752/752
I (52) wifi:mode : sta (a1:f8:a1:03:8f:04)
I (52) wifi:enable tsf
*** Booting Zephyr OS build zephyr-v3.1.0-921-g04b7e8ed1530  ***
[00:00:00.194,700] <inf> esp32_wifi: WIFI_EVENT_STA_START
I (537) wifi:new:<5,1>, old:<1,1>, ap:<255,255>, sta:<5,1>, prof:1
I (1052) wifi:state: init -> auth (b0)
I (1056) wifi:state: auth -> assoc (0)
I (1067) wifi:state: assoc -> run (10)
I (11067) wifi:state: run -> init (cc00)
I (11068) wifi:new:<5,0>, old:<5,1>, ap:<255,255>, sta:<5,1>, prof:1
[00:00:11.644,884] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (13129) wifi:new:<5,1>, old:<5,0>, ap:<255,255>, sta:<5,1>, prof:1
I (13130) wifi:state: init -> auth (b0)
I (13136) wifi:state: auth -> assoc (0)
[00:00:13.695,594] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (13146) wifi:state: assoc -> run (10)
I (23146) wifi:state: run -> init (cc00)
I (23147) wifi:new:<5,0>, old:<5,1>, ap:<255,255>, sta:<5,1>, prof:1
[00:00:23.724,196] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (25207) wifi:new:<5,1>, old:<5,0>, ap:<255,255>, sta:<5,1>, prof:1
I (25208) wifi:state: init -> auth (b0)
I (25214) wifi:state: auth -> assoc (0)
[00:00:25.773,162] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (25225) wifi:state: assoc -> run (10)
I (35226) wifi:state: run -> init (cc00)
I (35227) wifi:new:<5,0>, old:<5,1>, ap:<255,255>, sta:<5,1>, prof:1
[00:00:35.805,632] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (37286) wifi:new:<5,1>, old:<5,0>, ap:<255,255>, sta:<5,1>, prof:1
I (37288) wifi:state: init -> auth (b0)
I (37295) wifi:state: auth -> assoc (0)
[00:00:37.854,272] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (37304) wifi:state: assoc -> run (10)
`

sdonnan, 2022-07-10:
I've replicated the issue on an `ESP32-C3-DevKitM-1`. I the exact same build working fine on an `ESP32-C3-DevKitC-02`. FWIW they both carry the `XXN4` marking.

I'm going to poke around a little bit and see if I can identify any differences.

beriberikix, 2022-07-31:
I have a bunch of `ESP32-C3-DevKitC-02` and have been able to reproduce on 6 boards.

martinjaeger, 2022-07-31:
@sylvioalves do you have any estimation when the Espressif team will have time to have a closer look into this or upgrade to next version HAL?

This bug makes the ESP32-C3 pretty much unusable for Zephyr, as WiFi is one of the key features why one would choose this MCU.

sylvioalves, 2022-08-15:
Hi, I managed to get this fixed, will submit the PR soon.

martinjaeger, 2022-08-15:
That's really good news, thanks @sylvioalves!

JeffWelder, 2022-08-22:
@sylvioalves Thanks! Waiting to see how to fix it so I can finally use my dev-kit. 😄

closed 2022-09-01 by fabiobaltieri

spilz87, 2023-01-01:
Hi

I have a similare issue with ESP32C3 WROOM 02 H4 / N4.
I realysed that if I touch the antenna with my finger, it works well, other while it restart when trying to use BLE or WiFi.

I find no information about this issue.

can someone explain me the origin of what you find ?

I thanks you in advance

Spilz

PR #49682 (2022-09-01): soc: esp32: updates to sync with v4.4.1 hal_espressif
This PR brings a few updates to meet `hal_espressif` v4.4.1 baseline.

This also integrates the binary blob documentation, which is now available in `hal_espressif`.

Issues below were also fixed during hal_v4.4.1 bring up:  
Closes #43655
Closes #49762