zephyrproject-rtos/zephyr #43693: LE Audio: Rename enum bt_audio_pac_type

Thalley, 2022-03-11:
**Is your enhancement proposal related to a problem? Please describe.**
The enum is used to describe the direction, or type, or an endpoint and/or stream. Furthermore, it is used for broadcast endpoints as well, so the `PAC` infix is not very fitting. 

**Describe the solution you'd like**
The name should be change to something more generic. 

`bt_audio_type` is a bit too generic IMO, perhaps `bt_audio_ep_type`? 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

MariuszSkamra, 2022-03-11:
`bt_audio_direction` maybe

Thalley, 2022-03-11:
> `bt_audio_direction` maybe

Also a possibility. In that case, I think the values should be renamed `IN` and `OUT` as `SOURCE` and `SINK` are not really directions.