zephyrproject-rtos/zephyr #43696: mgmt/mcumgr: RFC: Standardize Kconfig option names for MCUMGR

de-nordic, 2022-03-11:
**Is your enhancement proposal related to a problem? Please describe.**
There is no common standard for naming MCUMGR related options in Kconfig making it sometimes hard to figure out what they are for or what name should be given to a new one.

**Describe the solution you'd like**
Standardize naming format for Kconfig options
My proposal is to make all Kconfig options related to `MCUMGR` start with `MGMT_MCUMGR`,
the follow it with `TRANSPORT` for transport related options or `GRP` for command groups, for example
```
MGMT_MCUMGR -- mcumgr enabler;
MGMT_MCUMGR_GLBL -- global mcumgr Kconfig options not related to transport or 
MGMT_MCUMGR_TRANSPORT -- transport group prefix;
MGMT_MCUMGR_TRANSPORT_GLBL -- global transport options;
MGMT_MCUMGR_TRANSPORT_<BT|UART|SHELL> -- transport specific options;
MGMT_MCUMGR_GRP_GLBL -- global options for MCUMGR management groups;
MGMT_MCUMGR_GRP_[OS|IMG|FLASH|FS|SHELL|STATS] -- prefix for specific groups
```

Intention of this issue is to collect insight and proposals on vocabulary (keywords) to use in Kconfig options definitions and general ideas onto how to tackle the problem with chaos in naming of options.

**Describe alternatives you've considered**
Leaving the mess as it is and let the naming chaos to widen.

**Additional context**
This  may be messy to do and require a lot of depreciation or intermediate solution that for some time would support old and new Kconfigs with translation layer provided.

de-nordic, 2022-03-11:
FYI @lairdjm @rerickson1 @mniestroj @nandojve

rerickson1, 2022-03-11:
This sounds great. Like the idea.

de-nordic, 2022-03-11:
The question is how we want to define vocabulary, whether we allow shortcuts or not.
For example in my proposal I use `TRANSPORT` for tansport but `GRP` for group: this already may seem misleading because at the same level of, lets say, options tree we have short name for one branch and long for other.
Additionally we already use shortcuts `MGMT` `OS`, `IMG`, `FS` and `STATS` and kinda full words `FLASH` and `SHELL`.
Should we maybe document the naming convention and vocabulary, within mcumgr documentation in Zephyr?

de-nordic, 2023-01-09:
Closed with https://github.com/zephyrproject-rtos/zephyr/pull/52761