zephyrproject-rtos/zephyr #43722: Request to configure microphones MP34DT01 on ST Discovery kit IoT node

dbortoluzzi, 2022-03-12:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to use microphone (MP34DT01) of the B-L475E-IOT01A (ST Discovery kit IoT node) in a zephyr project. This mcu is supported by zephyr, but the .dts doesn't configure the  interface used by microphones.

**Describe the solution you'd like**
According with the hardware specs [1], and a similar example [3], I tried to start from zephyr example using st_mpxxdtyy driver [2] and configure:

- prj.conf:

```
CONFIG_AUDIO=y
CONFIG_AUDIO_DMIC=y
CONFIG_AUDIO_MPXXDTYY=y
CONFIG_DMA=y
```

- app.overlay:

```
&XXXX {
     status = "okay";
     pinctrl-0 = <&YYYY &ZZZZ>;
     pinctrl-names = "default";

     mp34dt01: mp34dt01@0 {
            compatible = "st,mpxxdtyy";
            reg = <0>;
            label = "MP34DT01";
     };
};
```
According with the specs [1] and analyzing the ST function pack FP-AI-SENSING1 [4], I think I should use:
- _DFSDM1_CKOUT_ —» pin PE9
- _DFSDM1_DATIN2_ —» pin PE7
but I don't know how to write the DTS file.

In particular, about the app.overlay, I'm not sure:
- which interface to use (XXXX field)
- which pin variables to use (YYYY, ZZZZ fields).

**Describe alternatives you've considered**

I also tried these configurations, according to .dts specs:

```
/ {
	soc {
		i2c4: i2c@40005800 {
			compatible = "st,stm32-i2c-v2";
			#address-cells = <1>;
			#size-cells = <0>;
			label= "I2C_4";
			
			status = "okay";
			pinctrl-names = "default";
			clock-frequency = <I2C_BITRATE_FAST_PLUS>;

			mp34dt01@0 {
				compatible = "st,mpxxdtyy";
				reg = <0>;
				label = "MP34DT01";
			};
		};
	};

	i2c4_ck_pe9: i2c4_ck_pe9@0 {
		pinmux = <STM32_PINMUX('E', 9, GPIO)>;
		drive-open-drain;
	};
	
	i2c4_da_pe7: i2c4_da_pe7@0 {
		pinmux = <STM32_PINMUX('E', 7, GPIO)>;
		drive-open-drain;
	};
};

&i2c4 {
	pinctrl-names = "default";
	pinctrl-0 = <&i2c4_ck_pe9 &i2c4_da_pe7>;
};
```
But it doesn't compile:
_Error: disco_l475_iot1.dts.pre.tmp:1939.55-56 syntax error
FATAL ERROR: Unable to parse input tree_

**Additional context**

- _Zephyr version: 2.7.0_
- Sources:
[1] https://www.digikey.com/htmldatasheets/production/2263149/0/0/1/b-l475e-iot01a-user-manual.html : electric schema of the iot node
[2] https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/shields/x_nucleo_iks02a1/microphone : st mpxxdtyy example project
[3] https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/96b_argonkey/96b_argonkey.dts : similar dtsi (MP34DT05 module)
[4] https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-ode-function-pack-sw/fp-ai-sensing1.html : ST FP-AI-SENSING1

avisconti, 2022-03-17:
> @avisconti Is there an example tha could be used by @dbortoluzzi ?

The microphone on sensortile.box is analog (MP23ABS1), correct me if I'm wrong.
All the examples available are for digital microphones (like MP34DT01), which provides already a PDM streaming.

To use analog mic we require to attach it to ADC and do the conversion.

avisconti, 2022-03-17:
The digital microphone has to be attached to SPI/I2S, not I2C.
Take a look to x_nucleo_iks02a1, there is an example for microphone.
```
/*                                                                                                                                                       
 * Copyright (c) 2020 STMicroelectronics                                        
 *                                                                              
 * SPDX-License-Identifier: Apache-2.0                                          
 */                                                                             
                                                                                
&arduino_i2s {                                                                  
>       status = "okay";                                                        
                                                                                
>       mp34dt05@0 {                                                            
>       >       compatible = "st,mpxxdtyy";                                     
>       >       reg = <0>;                                                      
>       >       label = "MP34DT05";                                             
>       };                                                                      
};                                                                              
```
And samples:

```
  1 CONFIG_LOG=n                                                                                                                                        
   2 CONFIG_BOOT_BANNER=n                                                            
   3 CONFIG_STDOUT_CONSOLE=y                                                         
   4 CONFIG_PRINTK=y                                                                 
   5 CONFIG_I2S=y                                                                    
   6 CONFIG_GPIO=y                                                                   
   7 CONFIG_AUDIO=y                                                                  
   8 CONFIG_AUDIO_DMIC=y                                                             
   9 CONFIG_AUDIO_MPXXDTYY=y                                                         
```

avisconti, 2022-03-17:
@erwango 
I also noticed that in *boards/arm/nucleo_f411re/arduino_r3_connector.dtsi* there is already a arduino_i2s definition.
I guess that it is necessary also in *boards/arm/disco_l475_iot1/arduino_r3_connector.dtsi*

Would you agree?

erwango, 2022-03-17:
> I also noticed that in boards/arm/nucleo_f411re/arduino_r3_connector.dtsi there is already a arduino_i2s definition.

Is i2s supported on L4 ?
I was under the impression that i2s is F4 only today.

avisconti, 2022-03-17:
> > I also noticed that in boards/arm/nucleo_f411re/arduino_r3_connector.dtsi there is already a arduino_i2s definition.
> 
> Is i2s supported on L4 ? I was under the impression that i2s is F4 only today.

Oh, you are right, there is no I2S support in L4.
Infact, I just took a look to discovery 475 IOT board and the MP34DT01 is attached to DFSDM module, not SPI/I2S. To be honest I don't know much about this module. Do we have support for it in zephyr?

erwango, 2022-03-17:
> Infact, I just took a look to discovery 475 IOT board and the MP34DT01 is attached to DFSDM module, not SPI/I2S. To be honest I don't know much about this module. Do we have support for it in zephyr?

Ok, I missed that as well; no it doesn't seem to be supported..

giatorta, 2022-03-17:
yes, thank you, that's exactly where we are stuck

in IoT node, the microphones  MP34DT01 are attached to DFSDM, not via SPI,
I2C or I2S interfaces

we are trying to exploit code for mic based directly on STM32 native
examples/drivers (e.g. the SENSING1 dev kit from ST), but up to now it
seems to conflict somehow with Zephyr

any further suggestion would be very helpful

Gianluca (working with Daniele)

erwango, 2022-03-17:
> but up to now it seems to conflict somehow with Zephyr. any further suggestion would be very helpful

Well the problem description is too vague for providing helpful directions.
One thing that can help:
https://github.com/zephyrproject-rtos/hal_stm32#use-stm32cube-in-your-application

Then, you can also try to implement a zehyr driver similar using DFSDM hal API.