zephyrproject-rtos/zephyr #43728: esp32 build error while applicaton in T2 topology;

ck-telecom, 2022-03-13:
`**Describe the bug**
build error when create an application in T3 topology based on esp32

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   usually create an application using T3 topology described in https://docs.zephyrproject.org/latest/guides/west/workspaces.html
 - What have you tried to diagnose or workaround this issue?
 - no idea on this

**To Reproduce**
Steps to reproduce the behavior:
1. create an application using T2 topology
   you could follow the instructions bellow:
```
  $ git clone https://gitee.com/fwatch/twatch

  $ cd twatch

  $ west init -l app/

  $ west update # update modules

  $ west espressif update # update submodules
```
2. build samples/hell_word
```
west build -p auto -b esp32 -s app
``` 
6. See error
```
[84/85] Linking C executable bootloader.elf
[85/85] Generating binary image from built executable
esptool.py v3.1-dev
Merged 1 ELF section
Generated D:/workspace/twatch/build/esp-idf/build/bootloader/bootloader.bin
[159/169] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/workspace/twatch/build/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && C:\Users\gouqs\.espressif\tools\zephyr\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=D:/workspace/twatch/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/users/gouqs/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0"  -LD:/workspace/twatch/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -TD:/workspace/twatch/modules/hal/espressif/zephyr/esp32/../../components/esp32/ld/esp32.peripherals.ld && cmd.exe /C "cd /D D:\workspace\twatch\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/users/gouqs/.espressif/tools/zephyr/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/drivers/serial/libdrivers__serial.a(uart_esp32.c.obj):(.literal.uart_esp32_configure+0x0): undefined reference to `uart_periph_signal'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\workspace\twatch\build'
```

**Expected behavior**
A clear and concise description of what you expected to happen.
there is no error when using T2 topology 

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (Windows 11 build successful with T1 topology )

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

gmarull, 2022-03-14:
note: you can take https://github.com/zephyrproject-rtos/example-application as a reference

sylvioalves, 2022-03-14:
@ck-telecom @gmarull This issue happens due to wrong `hal_espressif` version. It does not match Zephyr's main revision.
The `west.yml` points to the revision below:  
```c
- name: hal_espressif
      revision: bcd7565ffa390d5774dc2fbe71a002faa9a7d082
      path: modules/hal/espressif
      west-commands: west/west-commands.yml
      groups:
        - hal
```
However, after `west update`, `modules/ha/espressif` is still at `bd2e2d8dcdc19defdb1bf6e07fb94f62bc99c925`.  
For some reason, `west update` is not pulling correct revision.  
If you run 
```c
cd modules/ha/espressif
git checkout bcd7565ffa390d5774dc2fbe71a002faa9a7d082
```  
build will be successful.

So, any reason for `west update` not to be pulling proper revision?

ck-telecom, 2022-03-14:
@sylvioalves but it happens in my home-computer

ck-telecom, 2022-03-14:
This is due to hal_espressif
``` 
-      west-commands: west/west-commands.yml
+      west-commands: scripts/west-commands.yml
```

sylvioalves, 2022-03-14:
@ck-telecom Did that work? If yes, close the issue please.

ck-telecom, 2022-03-15:
okay, not sure zephyr upstream need to fix?

ck-telecom, 2022-03-15:
> @ck-telecom Did that work? If yes, close the issue please.

can change to west-commands: scripts/west-commands.yml, result in west espressif update failed, it seems a potential issue

sylvioalves, 2022-03-15:
```c
-      west-commands: west/west-commands.yml
+      west-commands: scripts/west-commands.yml
```
That is wrong. You need to keep `west-commands: west/west-commands.yml`.

sylvioalves, 2022-03-15:
Something is wrong with `west update` in your env.  
neither `hal_espressif` and `tinycrypt` are having correct revision pulled.

sylvioalves, 2022-03-15:
@gmarull do you what is the root cause for `west update` not to be pulling revision defined in `west.yml` in this repo?

ck-telecom, 2022-03-15:
> Something is wrong with `west update` in your env. neither `hal_espressif` and `tinycrypt` are having correct revision pulled.

did you try to do some changes bellow:
```
-      west-commands: west/west-commands.yml
+      west-commands: scripts/west-commands.yml
```
that's a workaround for me to update to correct version, after that I revert to this manually :)

gmarull, 2022-03-15:
I'd simply suggest following the `example-application` application structure to see if the problem is gone.

sylvioalves, 2022-03-18:
@ck-telecom Have you checked previous comment regarding the `example-application`?

closed 2022-03-24 by ck-telecom