zephyrproject-rtos/zephyr #43732: esp32: MQTT publisher sample stuck for both TLS and non-TLS sample.

DineshDK03, 2022-03-13:
**Describe the bug**
MQTT publisher sample getting stucked after resolving ip address of hostname which uses **CONFIG_DNS_RESOLVER** . 

1. I am using esp32-wrover-devkit module's wifi for mqtt publisher sample where in this sample I have tried resolving the public mqtt broker **broker.hivemq.com**  using [getaddrinfo()](https://docs.zephyrproject.org/latest/reference/networking/sockets.html#c.zsock_getaddrinfo) function. Though it successfully resolving hostname's IP address, it is randomly stucking here and there  inside this [while loop ](https://github.com/zephyrproject-rtos/zephyr/blob/c31cb0075ad1933da5487db734c118cd5083ae3d/samples/net/mqtt_publisher/src/main.c#L443).

2. With same above hardware setup except for TLS I am using amazon iot core with CA and device certificates, the same issue as said in 1st point that hostname resolved fine but it is getting stucked in the process of writing handshake message.

**To Reproduce**
Steps to reproduce the behavior:
1. I have uploaded both the samples [here](https://github.com/DineshDK03/Zephyr-MQTT) 
2. `west build -p auto -b esp32 && west flash && west espressif monitor`

**Expected behavior**
Expecting mqtt publish response OK without any errors. like below console log
```
*** Booting Zephyr OS build v3.0.0-rc3-1133-gc31cb0075ad1  ***
[00:00:00.969,000] <inf> net_mqtt_publisher_sample: Running dhcpv4 client...
[00:00:00.969,000] <inf> esp32_wifi: WIFI_EVENT_STA_START
uart:~$ I (2174) wifi:new:<11,2>, old:<1,1>, ap:<255,255>, sta:<11,2>, prof:1
I (2819) wifi:state: init -> auth (b0)
I (2824) wifi:state: auth -> assoc (0)
I (2829) wifi:state: assoc -> run (10)
I (2932) wifi:connected with airtel_home, aid = 5, channel 11, 40D, bssid = f8:aa:3f:57:b0:f9
I (2933) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
I (2934) wifi:pm start, type: 1

I (2939) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[00:00:02.938,000] <inf> esp32_wifi: WIFI_EVENT_STA_CONNECTED
uart:~$ W (6465) wifi:<ba-add>idx:0 (ifx:0, f8:aa:3f:57:b0:f9), tid:0, ssn:4, winSize:64
[00:00:06.224,000] <inf> net_dhcpv4: Received: 192.168.1.15
[00:00:06.225,000] <inf> net_mqtt_publisher_sample: Your address: 192.168.1.15
[00:00:06.225,000] <inf> net_mqtt_publisher_sample: Lease time: 86400 seconds
[00:00:06.225,000] <inf> net_mqtt_publisher_sample: Subnet: 255.255.255.0
[00:00:06.225,000] <inf> net_mqtt_publisher_sample: Router: 192.168.1.1
[00:00:06.225,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:06.880,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[00:00:06.880,000] <inf> net_mqtt_publisher_sample: try_to_connect: 0 <OK>
[00:00:06.880,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:06.970,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:07.380,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:07.380,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:07.550,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:07.881,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
```

**Logs and console output**

1. Output for MQTT publisher sample without TLS

```
*** Booting Zephyr OS build v3.0.0-rc3-1133-gc31cb0075ad1  ***
[00:00:00.986,000] <inf> net_mqtt_publisher_sample: Running dhcpv4 client...
uart:~$ I (2191) wifi:new:<11,2>, old:<1,1>, ap:<255,255>, sta:<11,2>, prof:1
I (2836) wifi:state: init -> auth (b0)
I (2839) wifi:state: auth -> assoc (0)
I (2844) wifi:state: assoc -> run (10)
I (2950) wifi:connected with airtel_home, aid = 4, channel 11, 40D, bssid = f8:aa:3f:57:b0:f9
I (2950) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (2952) wifi:pm start, type: 1

I (3004) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (6293) wifi:<ba-add>idx:0 (ifx:0, f8:aa:3f:57:b0:f9), tid:0, ssn:3, winSize:64

Backtrace:0x40091180:0x3FFE7680 0x00000A89:0x3FFE76A0  |<-CORRUPTED
0x40091180: z_swap at /home/dineshdk/zephyrproject/zephyr/kernel/include/kswap.h:170

[00:00:06.308,000] <inf> net_mqtt_publisher_sample: DNS resolved for broker.hivemq.com:1883
[00:00:06.308,000] <inf> net_mqtt_publisher_sample: get_mqtt_broker_addrinfo: 0 <OK>
[00:00:06.308,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:06.693,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[... 111 lines cut ...]

[00:00:07.657,000] <err> os:  **  PS 0x60020
[00:00:07.657,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[00:00:07.657,000] <err> os:  **  A0 (nil)  SP 0x3ffe78a0  A2 (nil)  A3 0x60020
[00:00:07.657,000] <err> os:  **  A4 0x3ffc82f4  A5 0x3ffc62e8  A6 0x60023  A7 0x3ffea090
[00:00:07.657,000] <err> os:  **  A8 0x80091174  A9 0x3ffe7890 A10 0x3ffe9b5c A11 0x3ffc6378
[00:00:07.657,000] <err> os:  ** A12 0x60420 A13 0x3ffec080 A14 (nil) A15 0x3ffc7ab8
[00:00:07.657,000] <err> os:  ** LBEG 0x40087fe9 LEND 0x40088000 LCOUNT 0x4
0x40087fe9: find_context at /home/dineshdk/zephyrproject/zephyr/subsys/net/ip/net_context.c:840
(inlined by) net_context_packet_received at /home/dineshdk/zephyrproject/zephyr/subsys/net/ip/net_context.c:1834

0x40088000: find_context at /home/dineshdk/zephyrproject/zephyr/subsys/net/ip/net_context.c:839
(inlined by) net_context_packet_received at /home/dineshdk/zephyrproject/zephyr/subsys/net/ip/net_context.c:1834

[00:00:07.657,000] <err> os:  ** SAR 0x4
[00:00:07.657,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:07.657,000] <err> os: Current thread: 0x3ffc62e8 (rx_q[0])
[00:00:07.853,000] <err> os: Halting system
uart:~$
```

**Environment (please complete the following information):**
 - OS: Manjaro Linux
 - Toolchain zephyr-sdk-0.13.2
 - Zephyr v3.0.0

grantramsay, 2022-04-01:
Try increasing the "tcp work" stack size `CONFIG_NET_TCP_WORKQ_STACK_SIZE=1280`

Default is 1024, but this sample seems to use ~1168:
`[00:00:45.069,000] <inf> thread_analyzer:  tcp_work            : STACK: unused 112 usage 1168 / 1280 (91 %); CPU: 0 %`

Here's a patch I was using for esp32 wroom dev kit (zephry v3.0.0), just set `CONFIG_NET_CONFIG_PEER_IPV4_ADDR`/`CONFIG_ESP32_WIFI_SSID`/`CONFIG_ESP32_WIFI_PASSWORD`:
```
diff --git a/samples/net/mqtt_publisher/prj.conf b/samples/net/mqtt_publisher/prj.conf
index 9743597039..8f14c8bf47 100644
--- a/samples/net/mqtt_publisher/prj.conf
+++ b/samples/net/mqtt_publisher/prj.conf
@@ -22,8 +22,23 @@ CONFIG_NET_CONFIG_SETTINGS=y
 CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
 CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
 
-CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
-CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
+CONFIG_NET_CONFIG_PEER_IPV4_ADDR="mybrokeripv4"
+CONFIG_ESP32_WIFI_SSID="myssid"
+CONFIG_ESP32_WIFI_PASSWORD="mypassword"
+
+CONFIG_WIFI=y
+CONFIG_WIFI_ESP32=y
+CONFIG_ESP32_WIFI_STA_AUTO=y
+CONFIG_NET_L2_ETHERNET=y
+CONFIG_NET_DHCPV4=y
+
+CONFIG_MBEDTLS=y
+CONFIG_MBEDTLS_BUILTIN=y
+CONFIG_MBEDTLS_ENTROPY_ENABLED=y
+CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
+CONFIG_MBEDTLS_ECP_ALL_ENABLED=y
+
+CONFIG_NET_TCP_WORKQ_STACK_SIZE=1280
 
 CONFIG_MAIN_STACK_SIZE=2048
```

sylvioalves, 2022-04-01:
@grantramsay Hi, thanks. So the issue/crash reported in this PR was due to small `NET_TCP_WORKQ_STACK_SIZE` size. Not related to anything else.

grantramsay, 2022-04-01:
@sylvioalves yes, most likely. Although I have not run OPs sample code or tested with TLS enabled

DineshDK03, 2022-04-04:
@grantramsay  thanks for the solution, it did solve the issue for both MQTT publisher samples. Also closing this issue since it solved my issue.

closed 2022-04-04 by DineshDK03