zephyrproject-rtos/zephyr #43737: Cannot compile `native_posix` targets on Windows using the MinGW compiler for 3.0.0

coder137, 2022-03-14:
**Describe the bug**
Cannot compile `native_posix` targets on Windows using the MinGW compiler for 3.0.0

**MinGW GCC information**

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=D:/Software/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-10.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++ --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts=yes --enable-libstdcxx-time=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --disable-plugin --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev3, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat -Wl,--disable-dynamicbase,--default-image-base-low -static-libstdc++ -static-libgcc' 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--dynamicbase,--high-entropy-va,--nxcompat\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Rev3, Built by MSYS2 project)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Getting started steps for the Zephyr RTOS installation
2. pipenv install [your_packages]
3. `pipenv shell`
4. `[ROOT]/zephyrproject/zephyr/zephyr-env.cmd`
5. cd zephyrproject && `west update`
6. Copy `samples/hello_world` to `[ROOT]/examples/hello_world`
7. cd `[ROOT]/examples/hello_world`
8. `west build -b native_posix .`

**Output**

```
>west build -b native_posix .
-- west build: generating a build system
Including boilerplate (Zephyr base): D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: D:/Repositories/ZephyrProject-Repo/examples/hello_world
-- Zephyr version: 3.0.0 (D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr)
-- Found Python3: C:/Users/naidu/.virtualenvs/ZephyrProject-Repo-t4ZTUPJ2/Scripts/python.exe (found suitable exact version "3.8.5") found components: Interpreter
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/.cache
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/dts.cmake
Parsing D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration 'D:/Repositories/ZephyrProject-Repo/examples/hello_world/prj.conf'
Configuration saved to 'D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/.config'
Kconfig header saved to 'D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Software/msys64/mingw64/bin/gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Repositories/ZephyrProject-Repo/examples/hello_world/build
-- west build: building application
[1/94] Generating include/generated/version.h
-- Zephyr version: 3.0.0 (D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr), build: zephyr-v3.0.0
[6/94] Building C object zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
D:\Software\msys64\mingw64\bin\gcc.exe -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -ID:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/kernel/include -ID:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/arch/posix/include -ID:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/include -Izephyr/include/generated -ID:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/soc/posix/inf_clock -ID:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/boards/posix/native_posix -Os -imacros D:/Repositories/ZephyrProject-Repo/examples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=D:/Repositories/ZephyrProject-Repo/examples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=D:/Repositories/ZephyrProject-Repo/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -include D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -std=c11 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -MF zephyr\CMakeFiles\offsets.dir\arch\posix\core\offsets\offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -c D:/Repositories/ZephyrProject-Repo/zephyrproject/zephyr/arch/posix/core/offsets/offsets.c
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s: Assembler messages:
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:18: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:18: Error: junk at end of line, first unrecognized character is `_'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:24: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:24: Error: junk at end of line, first unrecognized character is `_'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:30: Warning: .type pseudo-op used outside of .def/.endef: ignored.
[... 28 lines cut ...]
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:114: Error: junk at end of line, first unrecognized character is `_'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:120: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:120: Error: junk at end of line, first unrecognized character is `_'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:126: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:126: Error: junk at end of line, first unrecognized character is `K'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:132: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:132: Error: junk at end of line, first unrecognized character is `_'
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:138: Warning: .type pseudo-op used outside of .def/.endef: ignored.
C:\Users\naidu\AppData\Local\Temp\ccptZLOf.s:138: Error: junk at end of line, first unrecognized character is `_'
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'D:\Software\CMake\bin\cmake.EXE' --build 'D:\Repositories\ZephyrProject-Repo\examples\hello_world\build'
```

**Expected behavior**
[... cut ...]  be run on the Windows OS using the MinGW compiler (Native posix)

**Impact**
- Would force me to use WSL or a dedicated Linux machine. (nothing wrong with this, but would limit my current development setup on a Windows machine)
- Testing RTOS / Thread behaviour and running simulations
- Unit testing my code in the future

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Native POSIX with MinGW 10.2.0
 - Commit SHA or Version used: HEAD detached at v3.0.0

stephanosio, 2022-03-14:
I do not think the POSIX arch port was ever meant to work/tested with the MinGW. Maybe @aescolar can further comment on this.

As for resolving this issue, removing `.type` for `CONFIG_ARCH_POSIX` when `__MINGW32__` should do:
https://github.com/zephyrproject-rtos/zephyr/blob/c31cb0075ad1933da5487db734c118cd5083ae3d/include/toolchain/gcc.h#L511-L519

stephanosio, 2022-03-14:
We should document more clearly that the POSIX arch port and the `native_posix` boards are currently not supported on non-Linux systems (macOS support is WIP, no plan for Windows/MinGW at the moment).

coder137, 2022-03-14:
Thank you for the quick reply. 
I would like to deep dive into this issue and see if I can add support for the MinGW toolchain.

Apart from the `include/toolchain/gcc.h` file and `gen_offset_header.py` file are there any additional files that would need to be changed according to the architecture?

I am currently reading more about the build procedure here (https://docs.zephyrproject.org/latest/guides/build/index.html). Any additional information/links regarding toolchain support and changes required would be great to have as well.

aescolar, 2022-03-14:
Indeed native_posix is not expected/meant to work on Windows with MinGW. And even if you get a basic sample compiling and running you will hit walls left and right with more complex apps, as the POSIX arch does expect a POSIX compliant OS, and going further some drivers will expect straight Linux.
This is documented under "Host system dependencies"
https://docs.zephyrproject.org/2.6.0/boards/posix/native_posix/doc/index.html#overview

closed 2022-11-20 by nashif