zephyrproject-rtos/zephyr #43786: [Logging] log context redefined with XCC when use zephyr logging api with SOF

aiChaoSONG, 2022-03-15:
**Describe the bug**
According to the document here: https://docs.zephyrproject.org/latest/reference/logging/index.html, I defined log context with LOG_MODULE_REGISTER or LOG_MODULE_DECLARE, and for inline functions in header file, I used LOG_MODULE_DECLARE within that function. But I can not compile SOF application with XCC, and GCC is good. 

Errors when use XCC to compile:
```
modules/sof/libmodules_sof.a(kpb.c.obj): In function `z_log_printf_arg_checker':
(.bss.__log_current_dynamic_data$24337+0x0): multiple definition of `__log_current_dynamic_data$24337'
modules/sof/libmodules_sof.a(pipeline-stream.c.obj):(.bss.__log_current_dynamic_data$24337+0x0): first defined here
modules/sof/libmodules_sof.a(idc.c.obj): In function `z_log_printf_arg_checker':
(.bss.__log_current_dynamic_data$24137+0x0): multiple definition of `__log_current_dynamic_data$24137'
modules/sof/libmodules_sof.a(pipeline-params.c.obj):(.bss.__log_current_dynamic_data$24137+0x0): first defined here
modules/sof/libmodules_sof.a(idc.c.obj):(.bss.__log_current_dynamic_data$24337+0x0): multiple definition of `__log_current_dynamic_data$24337'
modules/sof/libmodules_sof.a(pipeline-stream.c.obj):(.bss.__log_current_dynamic_data$24337+0x0): first defined here
```

Patch to replace SOF log calls with zephyr API: 
[5411.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/8249814/5411.patch.txt)

If I make some functions not inline, the issue can not be reproduced: https://github.com/thesofproject/sof/pull/5459

**To Reproduce**

1. Clone sof repo: `git clone https://github.com/thesofproject/sof.git`
2. Apply the patch: [5411.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/8249814/5411.patch.txt)
3. Go to SOF directory, clone zephyr repo with: `./script/xtensa-build-zephyr.py -c`
4. export XTENSA_TOOLS_ROOT=xtensa-tool-path.
5. build SOF+Zephyr with `./script/xtensa-build-zephyr.py tgl`

Or with zephyr style, you can apply my patch to SOF module, and build SOF.

**Expected behavior**
SOF replace with zephyr logging api, it should be built.

**Environment (please complete the following information):**
 - Toolchain: XCC
 - Commit SHA: e186ebc314

dcpleung, 2022-03-15:
This is due to the use of `LOG_MODULE_DECLARE()` inside the inline function in `component-ext.h`. AFAICT, XCC emits the same symbol names based on the source file for those static variables inside functions (hence the same number after the dollar sign). Because of this, multiple use of these inline functions (e.g. `comp_params()`) across different C files will result in same symbol names in all the corresponding object files. During linking, all these object files have symbols of the same name, and thus failing. Later version of GCC is smarter about this situation and does not emit these type of symbols anymore. However, due to XCC being horribly outdated, we are stuck with this kind of issue. This is not an issue with the logging subsystem, but an issue with XCC. A workaround would be do create a dedicated logging printf-like function in C file for components, and use `LOG_MODULE_DECLARE()` these instead of having them in all those inline functions.

aiChaoSONG, 2022-03-16:
> A workaround would be do create a dedicated logging printf-like function in C file for components, and use LOG_MODULE_DECLARE() these instead of having them in all those inline functions.

You mean I should create a function which wraps LOG_INF/LOG_ERR/LOG_WRN/LOG_DBG, and use LOG_MODULE_DECLARE() within the function? This looks like four functions, because we have four log levels.  Could you please explain more detailed

dcpleung, 2022-03-16:
Since those inline functions call, e.g., `comp_err()`, you can create some wrappers to functions which eventually call `LOG_*()`, as long as these new functions are in a C file.

dcpleung, 2022-03-29:
Once the PR in SOF is merged, could you close this issue once you have confirmed that it is no longer an issue? Thanks.

dcpleung, 2022-04-18:
Since the PR in SOF has been merged, is this still an issue for you?

aiChaoSONG, 2022-04-19:
Liam suppose this will be fixed later, alright, let's close for now.

closed 2022-04-19 by aiChaoSONG