zephyrproject-rtos/zephyr #43793: Alllow callbacks to CDC_ACM events

holgerschurig, 2022-03-15:
Szenario: Firmware inside a STM32 microcontroller connected via USB to host OS Linux

Wish:

* knowing when host OS has enumerated the CDC_ACM USB device
* knowing when host OS stops using the firmware (e.g. to reset a JSON parsing statemachine)

The events already exists, and one can then see in the console log already:

```
[00:00:13.076,000] <inf> usb_cdc_acm: Device configured
```

But unfortunately I cannot react to these USB events programmatically. So it would be nice if one could register a callback that cdc_acm_do_cb() calls. That would one allow to react programmatically to everything: USB_DC_ERROR, USB_DC_RESET, USB_DC_CONNECTED, USB_DC_DISCONNECTED, ...

jfischer-no, 2022-03-15:
You can pass a callback to usb_enable(), https://docs.zephyrproject.org/latest/reference/usb/uds.html?highlight=usb_enable#c.usb_enable

closed 2022-03-15 by jfischer-no

holgerschurig, 2022-03-15:
Thanks, can be closed. `usb_enable()` does the job for me.