zephyrproject-rtos/zephyr #43794: BMI160 Driver: Waiting time between SPI activation and reading CHIP IP is too low

PeachBug, 2022-03-15:
**Describe the bug**
When resetting the SoC (nrf52840) via jlink, BMI160 stops responding. The code blocks when the driver tries to read the CHIP ID register. I solved the issue by increasing the k_busy_wait from 100 to 200.

https://github.com/zephyrproject-rtos/zephyr/blob/565fc4d29c90e244cb16b5af05d5cc47d1edc059/drivers/sensor/bmi160/bmi160.c#L901

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840dk_nrf52840
2. JLinkRTTClient
3. JLinkExe
3.1. connect
3.2. r
3.3. g
4. See error on JLinkRTTClient

**Impact**
It induces the user to think there is some type of BMI160 misconfiguration issue.

**Environment:**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK
 - Version: 2.7.0

mbolivar-nordic, 2022-03-15:
@anangl -- do you have this sensor?

anangl, 2022-03-16:
> @anangl -- do you have this sensor?

No, unfortunately, I don't have one.

carlescufi, 2022-03-23:
@PeachBug can you please send a Pull Request changing the delay from 100 to 200?

PR #44146 (2022-04-13): drivers: sensor: bmi160: fix waiting time before reading CHIP_ID
Value changed from 100ms to 150ms. Value was not enough upon softreset.

Fixes #43794
commit 70e1f97c (2022-04-13): drivers: sensor: bmi160: fix waiting time before reading CHIP_ID

Value changed from 100us to 150us. Value was not enough upon softreset.
This value was arbitrarily chosen and should be changed if more
information on the subject is provided.

Fixes #43794
closed 2022-04-13 by mbolivar-nordic