zephyrproject-rtos/zephyr #43807: Test "cpp.libcxx.newlib.exception" failed on some x86 platforms.

chen-png, 2022-03-15:
**Describe the bug**
When running test "tests/subsys/cpp/libcxx/ cpp.libcxx.newlib" on some x86 platforms, the test it failed with timeout.

The output "exit" means that it couldn't handle the raised exception, so it invoked the default terminate function.
check and found that **for those platforms, it used bin format "zephyr.bin" file to run tests, but the key .eh_frame section for exception handler has been removed from bin output.**
```
 $<TARGET_PROPERTY:bintools,elfconvert_flag_section_remove>.eh_frame
```

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.0.0-14-gdb1f0289dd31  ***
version 201703
Running test suite libcxx_tests

START - test_array
PASS - test_array in 0.1 seconds

START - test_vector
PASS - test_vector in 0.1 seconds

START - test_make_unique
PASS - test_make_unique in 0.1 seconds

START - test_exception
exit
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: Zephyr-sdk-0.13.2
 - Commit ID:

stephanosio, 2022-03-15:
"some x86 platforms" is really not helpful. Please provide a proper description of the bug.

I am unable to reproduce this on the QEMU platforms:
```
INFO    -  1/14 qemu_x86                  tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (qemu 2.461s)
INFO    -  2/14 acrn                      tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (build)
INFO    -  3/14 qemu_x86_64               tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (qemu 2.415s)
INFO    -  4/14 ehl_crb_sbl               tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (build)
INFO    -  5/14 qemu_x86                  tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (qemu 2.215s)
INFO    -  6/14 acrn_ehl_crb              tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (build)
INFO    -  7/14 up_squared                tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (build)
INFO    -  8/14 up_squared                tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (build)
INFO    -  9/14 ehl_crb_sbl               tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (build)
INFO    - 10/14 acrn                      tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (build)
INFO    - 11/14 acrn_ehl_crb              tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (build)
INFO    - 12/14 ehl_crb                   tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (build)
INFO    - 13/14 ehl_crb                   tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     PASSED (build)
INFO    - 14/14 qemu_x86_64               tests/subsys/cpp/libcxx/cpp.libcxx.newlib          PASSED (qemu 2.330s)
```

chen-png, 2022-03-15:
as for "some x86 platforms", this is not public, so I used this. actually, for those platforms support C++ exception, which use the bin format file "zephyr.bin" to this test, they all will meet this failure.

stephanosio, 2022-03-15:
Looking at:
https://github.com/zephyrproject-rtos/zephyr/blob/affa68bbadf47adf791d1a535f406b91aa1d6f3b/CMakeLists.txt#L1420-L1429

`.eh_frame` should not be removed when `CONFIG_EXCEPTIONS=y`, which is the case for this test.

Are you sure this happens on the upstream boards?

chen-png, 2022-03-15:
sorry for the confusion, I submit this [8c069c3](https://github.com/zephyrproject-rtos/zephyr/commit/8c069c365db739c83619f52dbfe1b44572d5a2ec) into upstream recently, but I want to backport it into v3.0 and v2.7 too, need an issue to track it for backport PR, so I created this.

stephanosio, 2022-03-15:
> 

Ok, that makes a lot more sense 🤦‍♂️ 

Thanks for clarifying.

chen-png, 2022-03-17:
fixed by #43681, #43679.

PR #43681 (2022-03-16): [Backport v2.7-branch]cmake: save eh_frame section in output with CONFIG_EXCEPTIONS.
.eh_frame section should not be removed directly in the
hex format and bin format output, it should be based on
whether we need exception handler feature.

Fixes #43807
PR #43679 (2022-03-16): [Backport v3.0-branch] cmake: save eh_frame section in output with CONFIG_EXCEPTIONS.
Backport 8c069c365db739c83619f52dbfe1b44572d5a2ec from #43483

Fixes #43807
closed 2022-03-17 by chen-png