zephyrproject-rtos/zephyr #43828: intel_adsp_cavs25: multiple tests under tests/boards/intel_adsp/smoke are failing

yerabolu, 2022-03-15:
**Describe the bug**
boards.intel_adsp.host_ipc
boards.intel_adsp.clock_calibrate
Above 2 tests are failing due to a assertion failure. 

boards.intel_adsp.ipm_cavs_host doesn't run until the twister timeout. 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p intel_adsp_cavs25 --device-testing --device-serial-pty=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh -T tests/boards/intel_adsp/smoke/ -vvv
2. see error

**Logs and console output**
```
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Trimming 16 bytes of extended manifest
INFO:cavs-fw:Resetting HDA device
INFO:cavs-fw:Powering down DSP cores
INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
INFO:cavs-fw:Mapped 2M huge page at 0x155200000 to contain 106496 bytes of firmware
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
INFO:cavs-fw:Wait for ROM startup
INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
INFO:cavs-fw:cAVS firmware load complete
--
Running test suite intel_adsp
===================================================================
START - test_smp_boot_delay
Launch cpu1
Launch cpu2
Launch cpu3
PASS - test_smp_boot_delay in 0.61 seconds
===================================================================
START - test_cpu_halt
halt/restart core 1...
halt/restart core 2...
halt/restart core 3...
PASS - test_cpu_halt in 0.452 seconds
===================================================================
START - test_post_boot_ipi
PASS - test_post_boot_ipi in 0.1 seconds
===================================================================
START - test_cpu_behavior
Per-CPU smoke test 0...
Cache behavior check
CCOUNT/WALCLK ratio 10.406
CPI = 2.506
Per-CPU smoke test 1...
Cache behavior check
CCOUNT/WALCLK ratio 10.396
CPI = 2.502
Per-CPU smoke test 2...
Cache behavior check
CCOUNT/WALCLK ratio 10.425
CPI = 2.502
Per-CPU smoke test 3...
Cache behavior check
CCOUNT/WALCLK ratio 10.409
CPI = 2.502
PASS - test_cpu_behavior in 0.2 seconds
===================================================================
START - test_host_ipc
Simple message send...
Return message request...
Return message request 2...
Return message request, async...
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:83: test_host_ipc: (i < 10000 is false)
timeout waiting for msg_flag
FAIL - test_host_ipc in 0.589 seconds
===================================================================
START - test_clock_calibrate
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/smoke/src/hostipc.c:28: ipc_done: (done_flag is true)
done called unexpectedly
FAIL - test_clock_calibrate in 0.8 seconds
===================================================================
START - test_ipm_cavs_host

```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain (e.g Zephyr SDK, ...): SDK13.2
 - Commit SHA: e65e26162199

LixinGuoX, 2022-03-22:
This is a random failure, and found the cAVS IPC Message Handler Callback function was not invoked as expected.

LixinGuoX, 2022-03-24:
@andyross Hi, could you help to look at this issue?

LixinGuoX, 2022-03-29:
Also find that test_ipm_cavs_host failed on intel_adsp_cavs18 occasionally, we can add retry-failed option when running tests to avoid this unstable issue.

andyross, 2022-04-05:
Just getting to this for triage.  This definitely seems reliable on my hardware (the Asus chromebook and a Up Xtreme).  What kind of failure rates are you seeing?  Does this happen if you do it via a manual build or only in a twister run, etc...?

andyross, 2022-04-18:
Ping?  Any update on the failure rate, etc...?

smrtos, 2022-04-22:
(more details)

There are 2 test sets:

<details>
<summary>Test Set 1</summary>

- boards.intel_adsp : boards.intel_adsp.host_ipc
- boards.intel_adsp : boards.intel_adsp.clock_calibrate
- boards.intel_adsp : boards.intel_adsp.ipm_cavs_host

</details>

<details>
<summary>Test Set 2</summary>

-  drivers.dma.chan_blen_transfer.dma_m2m_chan1_burst8
-  drivers.sensor.generic.sensor_unit_conversion
-  drivers.can.utilities.can_filter_to_zcan_filter
-  drivers.dma.chan_blen_transfer.dma_m2m_chan0_burst8
-  drivers.hwinfo.api.clear_reset_cause
-  drivers.can.utilities.can_frame_to_zcan_frame
-  drivers.dma.chan_blen_transfer.dma_m2m_chan0_burst16
-  drivers.dma.scatter_gather.dma_m2m_sg
-  drivers.can.utilities.zcan_frame_to_can_frame
-  drivers.sensor.generic.fpu.sensor_unit_conversion
-  drivers.sensor.generic.sensor_handle_triggers
-  drivers.hwinfo.api.device_id_get
-  drivers.hwinfo.api.get_reset_cause
-  boards.intel_adsp.ipm_cavs_host
-  drivers.dma.chan_blen_transfer.dma_m2m_chan1_burst16
-  drivers.can.utilities.can_bytes_to_dlc
-  drivers.can.utilities.can_dlc_to_bytes
-  drivers.sensor.generic.fpu.sensor_handle_triggers
-  drivers.can.utilities.zcan_filter_to_can_filter
-  drivers.sensor.generic.fpu.sensor_get_channels
-  drivers.hwinfo.api.get_supported_reset_cause
-  drivers.dma.loop_transfer.dma_m2m_loop
-  drivers.sensor.generic.sensor_get_channels
-  drivers.dma.loop_transfer.dma_m2m_loop_suspend_resume
-  boards.intel_adsp.mm.adsp_mem_map_array
-  boards.intel_adsp.mm.adsp_mem_map_region

</details>

For cavs25, 

Scenario 1:
If test set 1 failed completely with below screenshot (i.e. without "PROJECT EXECUTION FAILED"), it will cause test set 2 to end in error state with only delay boot banner shown.
[image]

Scenario 2:
If test set 1 failed with below screenshot (i.e. with "PROJECT EXECUTION FAILED"), it will not jeopardize test set 2.
[image]

It seems random for scenario 1 and 2 to happen.
But if we remove the CONFIG_MP_NUM_CPUS=4 added in commit https://github.com/zephyrproject-rtos/zephyr/commit/b433813f5fb5655ee18133dfac176c9e8dfe9a45, test set 2 will not be affected by test 1 even in scenario 1.

So a limited conclusion so far is:

Test set 1 and CONFIG_MP_NUM_CPUS=4 seem to cause the board enter some incorrect state and block the test set 2 which runs afterwards.

smrtos, 2022-04-28:
> (more details)
> 
> There are 2 test sets:
> 
> Test Set 1
> Test Set 2
> For cavs25,
> 
> Scenario 1: If test set 1 failed completely with below screenshot (i.e. without "PROJECT EXECUTION FAILED"), it will cause test set 2 to end in error state with only delay boot banner shown. [image]
> 
> Scenario 2: If test set 1 failed with below screenshot (i.e. with "PROJECT EXECUTION FAILED"), it will not jeopardize test set 2. [image]
> 
> It is quite often for test set 1 to fail on all cavs 15/18/25 platforms. But it seems random for which scenario to happen.
> 
> And if we remove the CONFIG_MP_NUM_CPUS=4 added in commit [b433813](https://github.com/zephyrproject-rtos/zephyr/commit/b433813f5fb5655ee18133dfac176c9e8dfe9a45), test set 2 will not be affected by test 1 even in scenario 1.
> 
> So a limited conclusion so far is:
> 
> Test set 1 and CONFIG_MP_NUM_CPUS=4 seem to cause the board enter some incorrect state and block the test set 2 which runs afterwards.

The Test Set 1 is failing for all cavs15/18/25 platforms now.

andyross, 2022-04-28:
Amber poked me to come look at this one.  Indeed, I see this reliably now (though checking the dates, it's likely that there is another problem on the CI hardware). 

But for the existing failure: this bisects to this patch which merged a few weeks back:

commit 5416abfe5f70 ("util: convert wait_for macro to uppercase")

A simple revert of that works, but as it involves other subsystems I can't submit that and need to spend a bit of time untangling.

As far as the changes in that patch: It doesn't simply rename a macro, it replaces the original cpu-local spinning in the WAIT_FOR() macro with a sleep.  That's... probably not obviously wrong in all cases, but it's not the design intent either (most of these conditions are trying to catch microsecond-scale replies from the host, as reported by interrupts potentially received on another core: we really do want to be spinning).  I suspect there's an edge case in there tripping things up.

Will untangle the deltas and submit a partial reversion when I get a chance.

PR #46314 (2022-06-07): tests/intel_adsp/smoke: Revert WAIT_FOR breakage
The evolution of the original WAIT_FOR trick in this test, first to
move it to a generic spot and then to make it take a delay
paramatrization, have had the unfortunate side effect of breaking the
original usage.

And... it's really not following the design intent anyway.  The idea
here was to spin hard waiting on external hardware (a python script
running on the host) to deliver an interrupt.  Both of the new tricks
(suspending the thread with a sleep and hammering the HDA wall clock
for time) can affect interrupt delivery.

Just restore the or [... cut]
commit 5139d052 (2022-06-07): tests/intel_adsp/smoke: Revert WAIT_FOR breakage

The evolution of the original WAIT_FOR trick in this test, first to
move it to a generic spot and then to make it take a delay
paramatrization, have had the unfortunate side effect of breaking the
original usage.

And... it's really not following the design intent anyway.  The idea
here was to spin hard waiting on external hardware (a python script
running on the host) to deliver an interrupt.  Both of the new tricks
(suspending the thread with a sleep and hammering the HDA wall clock
for time) can affect interrupt delivery.

Just restore the o [... cut]
closed 2022-06-07 by carlescufi