zephyrproject-rtos/zephyr #43830: LPC55S69 Not flashing to second core.

durinwinter, 2022-03-15:
Dual Core examples not working on Zephyr. MCU Expresso dual core examples for bare metal does work on this board.

**To Reproduce**
Attempt flashing a dual core example on the LPC55S69-EVK.

**Expected behavior**
Dual core functionality acts as expected according to sample documentation.

**Impact**
Showstopper for LPC55S69 Usage with Zephyr.

danieldegrasse, 2022-03-23:
@durinwinter How are you flashing the sample? You need to use `JLinkExe` to manually load the multicore binary, see here: https://docs.zephyrproject.org/latest/boards/arm/lpcxpresso55s69/doc/index.html#building-a-dual-core-image. Additionally, the only sample I am aware of that supports dual code on the LPC55S69 is `samples/subsys/ipc/openamp`

danieldegrasse, 2022-04-28:
@durinwinter Could you confirm if you were able to get this sample to work for you?

dleach02, 2022-05-19:
closing due to nonresponse

closed 2022-05-19 by dleach02