zephyrproject-rtos/zephyr #43836: stm32: g0: RTT doesn't work properly after stop mode

ycsin, 2022-03-16:
**Describe the bug**
The RTT stops working properly after the MCU enter stop mode. I'm on 2.7.1 LTS branch which already has #38561 patch.
It prints garbage after MCU went to sleep, it also missed some lines that would have been printed if UART backend were used:

[image]

[image]

Apart from this, I believe some of the character (next line?) sent from the log/shell cant be recognized by the RTT viewer for some reason:
```
WARNING: <Terminal 0>: ANSI CSI SGR parameter 7 not supported.
WARNING: <Terminal 0>: ANSI CSI SGR parameter 7 not supported.
WARNING: <Terminal 0>: ANSI CSI SGR parameter 7 not supported.
WARNING: <Terminal 0>: ANSI CSI SGR parameter 7 not supported.
WARNING: <Terminal 0>: ANSI CSI SGR parameter 7 not supported.
```

**Environment (please complete the following information):**
 - OS: Windows
 - gnuarmemb
 - 2.7.1 LTS

ycsin, 2022-03-16:
@erwango @ABOSTM any idea?

ABOSTM, 2022-03-17:
At first glance I would say: maybe one of the condition described in #38561 in not respected after wakeup
* one of the DMA is clocked

and sometimes also
* one of the DBGMCU bit STOP/STANDBY/SLEEP is set

@ycsin , could you check that ?

ycsin, 2022-03-17:
> @ycsin , could you check that ?

Thanks for looking into this.
How should I check this? Break point in the `power.c` exit post ops?

ABOSTM, 2022-03-17:
Anywhere but after wakeup,  `power.c` exit post ops  sounds good, but at the end of this function.

ycsin, 2022-03-22:
> @ycsin , could you check that ?

I added a breakpoint [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/st_stm32/stm32g0/power.c#L72):
[image]

[image]
[image]

erwango, 2022-04-29:
@ycsin Did you made any progress on this issue? Is it still valid ?

ycsin, 2022-04-29:
> @ycsin Did you made any progress on this issue? Is it still valid ?

I've left my previous company and I doubt if anyone made any progress on this issue, it should still be valid

erwango, 2022-04-29:
> > @ycsin Did you made any progress on this issue? Is it still valid ?
> 
> I've left my previous company and I doubt if anyone made any progress on this issue, it should still be valid

Thanks for the head's up.

erwango, 2022-08-12:
Would need another test since #48640 was merged and can have an impact

closed 2022-11-06 by github-actions[bot]