zephyrproject-rtos/zephyr #43842: tests-ci : libraries: encoding: jwt test Timeout

hakehuang, 2022-03-16:
**Describe the bug**
jwt test is Timeout on zephyr-v3.0.0-1108-ge65e26162199 on mimxrt1060_evk

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1060_evk  --sub-test libraries.encoding
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.0.0-1108-ge65e26162199  ***
Running test suite lib_jwt_test
===================================================================
START - test_jwt
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x8000a758  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0xee119966 r14/lr:  0x6000f675
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000608 (test_jwt)
E: Halting system

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.0.0-1108-ge65e26162199

LixinGuoX, 2022-03-16:
Found the same issue on mimxrt1050_evk, #43792 for reference.

hakehuang, 2022-03-16:
@dleach02 , @danieldegrasse , @mmahadevan108

PR #44093 (2022-03-23): random: rand32_ctr_drbg: fix bug preventing initialization
Bug introduced that prevented the logic from initializing the
context structure. Without this initialization, the application
will crash on repeated request for ctr_drbg random data.

Fixes: #44092
Fixes: #43792
Fixes: #43841
Fixes: #43842
closed 2022-03-23 by dleach02