zephyrproject-rtos/zephyr #43852: Support testing with ARM FVP in the CI

stephanosio, 2022-03-16:
### Summary

Support running the Zephyr tests using the ARM FVP (fixed virtual platform; emulation software) in the CI.

### Problem description

Some boards and ARM features (e.g. Ethos neural coprocessors) can only be tested using the ARM FVP emulation platform (QEMU does not currently support emulating them).

### Proposed change

1. Include ARM FVP as part of the [CI Docker image](https://github.com/zephyrproject-rtos/docker-image).
    * We need to check if ARM FVP EULA allows this.

1. Update CI workflows to run tests using the ARM FVP included in the CI Docker image.
    * Set `ARMFVP_BIN_PATH` environment variable.

carlocaione, 2022-03-16:
For emulation FVP should be more advanced and truthful to the hardware than QEMU, so maybe we can avoid all the usaul `icount` issues with QEMU.

povergoing, 2022-03-16:
While the annoying issue may be [`You are not permitted to redistribute the 'Fixed Virtual Platforms (FVPs)' 
Software (in whole or in part) or any derivatives thereof.`](https://developer.arm.com/tools-and-software/simulation-models/fixed-virtual-platforms/end-user-license-agreement-for-fixed-virtual-platforms). I am not an expert on Licenses, but it's a kind of dangerous thing I think. I already contacted the relevant guy half a year ago. But still no response...

povergoing, 2022-03-16:
> For emulation FVP should be more advanced and truthful to the hardware than QEMU, so maybe we can avoid all the usual `icount` issues with QEMU.

That's true, I 100% agree that including FVP into Docker img. Anyway, I'll try again. Fortunately, some high-level one in Arm is aware of this thing, so let's see.

stephanosio, 2022-03-16:
@povergoing The clause you quoted falls under `(b) if you are receiving a Non-Commercial Use License or version (as applicable) of the Arm Tools:`, which I suppose is referring to those who have received a non-commercial license of the normally paid tools.

The `(a) if you are receiving the Arm Tools free of charge:` does not seem to explicitly prohibit redistribution, but it does not expressly permit redistribution either.

Weichen81, 2022-03-16:
Just some personal suggestion:
another thing that need to be considered is that the instance to accept "EULA" must be a human, so the click or enter "OK" behaviors should not be done by a "BOT". 
For Cortex-A series, an architecture level FVP "Armv-A Base RevC AEM FVP" (https://developer.arm.com/tools-and-software/simulation-models/fixed-virtual-platforms/arm-ecosystem-models) can be download without "EULA". So maybe you
can integrate the URL in docker image, but not FVP binary. But for Cortex-R, current we still haven't a proper way to bypass "EULA" click.

povergoing, 2022-03-16:
> @povergoing The clause you quoted falls under `(b) if you are receiving a Non-Commercial Use License or version (as applicable) of the Arm Tools:`, which I suppose is referring to those who have [...]

I am also very confused. This is totally out of my knowledge... But IMO, since the only way to get the FVP is to "register on arm website, accept "EULA" then download" (Arm wants to record who uses FVP?), there must be some restrictions on redistribution. In another world, if the binary is included in CI docker IMG or zephyr SDK, that means anyone can download FVP without accepting the EULA so that Arm cannot record.

**Or is there any way that the host machine keeps the FVP binary instead of the docker IMG, mapping the FVP binary path to docker when running CI?**  In this way, we use FVP as a service and certainly do not redistribute FVP, so we don't need to care about the restrictions of redistribution.

stephanosio, 2022-03-16:
> Or is there any way that the host machine keeps the FVP binary instead of the docker IMG, mapping the FVP binary path to docker when running CI?

There is a long-term plan to implement something like that to get testing with the proprietary toolchains working in the CI, but it will take some time to get that implemented (no timeline yet).

povergoing, 2022-04-22:
Hi @stephanosio @carlocaione,
Update here: the FVP_BaseR_AEMv8R on [page ](https://developer.arm.com/tools-and-software/simulation-models/fixed-virtual-platforms/arm-ecosystem-models) can download without any logins or registers.

The tricky part is how we avoid redistribution? That means the docker file should not contain the fvp bins (since anyone can easily get the docker image). Can we download the FVP at runtime triggered by some event? like,  `run ci -> trigger download fvp -> run ci test`?
Any suggestions?

stephanosio, 2022-04-24:
Thanks for the update.

> Can we download the FVP at runtime triggered by some event? 

While that should work, doing so would incur a significant amount of traffic to the ARM website and also, more importantly, will slow down the CI depending on the file provider server load.

I think the best way to handle this would be to create a private CI Docker image, based on the public one, that is only available to the authenticated Zephyr CI runners (and of course, cannot be downloaded publicly).

povergoing, 2022-04-24:
> I think the best way to handle this would be to create a private CI Docker image, based on the public one, that is only available to the authenticated Zephyr CI runners (and of course, cannot be downloaded publicly).

Agreed, looks like it is the best way. So, is there anything we can help with? :)

stephanosio, 2022-04-24:
> Agreed, looks like it is the best way. So, is there anything we can help with? :)

I will implement a proof-of-concept this week and let you know if there are any issues. Thanks.

stephanosio, 2022-04-24:
Test CI run using a private CI docker image that contains ARM FVP: https://github.com/zephyrproject-rtos/zephyr-testing/actions/runs/2215429286

stephanosio, 2022-04-24:
@povergoing Testing with the latest ARM FVP available on the website, I am seeing a lot of failures:

* fvp_baser_aemv8r: [3 failures](https://gist.github.com/stephanosio/4eea54d6b48d83c3efa60bbf7d65ab42)
    * [samples/subsys/tracing/tracing.user: Timeout](https://gist.github.com/stephanosio/4eea54d6b48d83c3efa60bbf7d65ab42#file-gistfile1-txt-L6)
    * [tests/kernel/threads/thread_apis/kernel.threads.apis: Timeout](https://gist.github.com/stephanosio/4eea54d6b48d83c3efa60bbf7d65ab42#file-gistfile1-txt-L25)
    * [tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls: Timeout](https://gist.github.com/stephanosio/4eea54d6b48d83c3efa60bbf7d65ab42#file-gistfile1-txt-L270)
* fvp_baser_aemv8r_smp: [37 failures](https://gist.github.com/stephanosio/581d40cacc81a569beb56d429956fb61)
    * tests/kernel/smp/kernel.multiprocessing.smp: Timeout
    * samples/userspace/shared_mem/sample.kernel.memory_protection.shared_mem: Timeout
    * samples/userspace/hello_world_user/sample.helloworld: Timeout
    * tests/lib/sprintf/libraries.libc.sprintf: Timeout
    * tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace: Timeout
    * tests/kernel/threads/tls/kernel.threads.tls.userspace: Timeout
    * ...
* fvp_baser_aemv8r_aarch32: All passed
* fvp_base_revc_2xaemv8a: [3 failures](https://gist.github.com/stephanosio/8393439c0c7f85837034130f73ef8595)
    * [tests/ztest/error_hook/testing.ztest.error_hook: Failed](https://gist.github.com/stephanosio/8393439c0c7f85837034130f73ef8595#file-gistfile1-txt-L7)
    * [tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map: Timeout](https://gist.github.com/stephanosio/8393439c0c7f85837034130f73ef8595#file-gistfile1-txt-L109)
    * [tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls: Timeout](https://gist.github.com/stephanosio/8393439c0c7f85837034130f73ef8595#file-gistfile1-txt-L753)
* mps3_an547: All passed
* mps3_an547_ns: All passed

cc @SgrrZhf @carlocaione

povergoing, 2022-04-25:
> Test CI run using a private CI docker image that contains ARM FVP: ~https://github.com/zephyrproject-rtos/zephyr-testing/actions/runs/2215429286~ https://github.com/zephyrproject-rtos/zephyr-testing/actions/runs/2216000834
Amazing work, I didn't expect you have done it so quickly 🥇 

> @povergoing Testing with the latest ARM FVP available on the website, I am seeing a lot of failures:

I see the failures. @SgrrZhf  and I will fix them as soon as possible.

May I ask if those failures will block the CI or something for others? Can we see the results if we push a PR for fixing them in the GitHub action?

stephanosio, 2022-04-25:
> May I ask if those failures will block the CI or something for others?

@povergoing We can merge #45099 after these failures are fixed.

> Can we see the results if we push a PR for fixing them in the GitHub action?

You can try opening a PR with the fixes to [`zephyr-testing`](https://github.com/zephyrproject-rtos/zephyr-testing) to see if they pass.

povergoing, 2022-04-25:
> You can try opening a PR with the fixes to [`zephyr-testing`](https://github.com/zephyrproject-rtos/zephyr-testing) to see if they pass.

understood, thanks!

stephanosio, 2022-04-25:
Filed bug reports for the failures reported in https://github.com/zephyrproject-rtos/zephyr/issues/43852#issuecomment-1107855478:

#45108 
#45110 
#45111

microbuilder, 2022-11-15:
Required to enable CI testing of the Ethos-U55 in this PR, which is currently build only: https://github.com/zephyrproject-rtos/zephyr/pull/42580

stephanosio, 2024-02-15:
ARM FVP licence has been updated to not prohibit redistribution and therefore it is now possible to include it in the public Docker image, as done in https://github.com/zephyrproject-rtos/docker-image/pull/170.

CI will start using ARM FVP for the supported boards once the new Docker image is mainlined, after which this issue can be closed.

stephanosio, 2024-10-24:
Closing since FVP has been integrated into the public CI Docker image and subsequently in the Zephyr main CI workflow.

https://github.com/zephyrproject-rtos/docker-image/blob/e511be8a0a9551927a0f8ece1da4e0e83ffdb6ce/Dockerfile.ci#L24-L31

closed 2024-10-24 by stephanosio