zephyrproject-rtos/zephyr #43858: mcumgr seems to lock up when it receives command for group that does not exist

lairdjm, 2022-03-16:
**Describe the bug**
It seems that mcumgr is locking the smp_svr sample application up when it receives a command for which it does not have a handler for, noticed whilst testing fs checksum additions. Test command: `\06\09AAwAAAACAEYBAL//2nk=\0a`
No further looking for the cause or how this happens has been investigated as of yet. This is with the shell mcumgr transport, no other transport has been tested.

**To Reproduce**
Steps to reproduce the behaviour:
Build smp_svr sample with it set to include fs and use shell for mcumgr transport, send above test command, observe as UART will no longer respond and no output is emitted.

**Expected behavior**
For an error to be emitted and for the application to not lock up

**Impact**
Showstopper if all it takes to DoS a device is a simple invalid mcumgr command

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 2ef47509c37163a0848b72b6ab37b7dbf831661e

**Additional context**
Will update bug once I investigate more

de-nordic, 2022-03-16:
I have tried to verify that on e186ebc3143e642f69c9379abdccccf4f7fbfaf8 with following steps:
1) I have built smp_svr with OS group only, with taskstat command enabled only, for BT and uploaded it to board;
2) I have issued command "taskstat", with mcumgr-cli, got error 2, "no mem", which is correct with default conf as there is not enough space to collect taskstat;
3) I have issued command "echo 0123456789" and got rc=8, "not supported"; mcumgr-cli does not report error correctly but this is problem with mcumgr-cli;
4) I have issued command "image list" and got "Error: 8" (rc=8), because entire image group has not been compiled in.

The smp_svr seems to correctly response with rc=8 on non-existing commands.

lairdjm, 2022-03-16:
If you use `AAwAAAACAEYBAL//2nk=` does it cause the issue? This is using a 16-bit group where the upper byte is non-zero, so might be caused by that, I'll have a look at it later and see if I can find at what point it's freezing

de-nordic, 2022-03-16:
I have tried to decode your input and got this: 00 0c 00 00 00 02 00 46  01 00 bf ff da 79
I think that console frame length is not encoded in big endian (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/mcumgr/lib/transport/smp-console.md), so the smp decoder probably waits for more 04 14 frames to fill in the missing bytes.

lairdjm, 2022-03-16:
Doesn't seem to be the issue. If I send a valid command, I get a response, and it's all being done by the same underlying code. This is Qt-based:
```
void MainWindow::SendMessage(QByteArray message)
{
	QByteArray msg;
	uint16_t size = message.length();
	size += 2;
	msg.append((uint8_t)((size & 0xff00) >> 8));
	msg.append((uint8_t)(size & 0xff));
	msg.append(message);
	uint16_t crc = crc16(&message, message.length(), 0x1021, 0, true);
	msg.append((uint8_t)((crc & 0xff00) >> 8));
	msg.append((uint8_t)(crc & 0xff));

	QByteArray b64msg;
	b64msg.append((uint8_t)0x06);
	b64msg.append((uint8_t)0x09);
	b64msg.append(msg.toBase64());
	b64msg.append((uint8_t)0x0a);

	gspSerialPort.write(b64msg);
}

void MainWindow::on_btn_board_clicked()
{
    QByteArray message;
    QByteArray cbor;

    message.append((uint8_t)0x00); //Read
    message.append((uint8_t)0x00); //Flags
    message.append((uint8_t)0x00); //Length A
    message.append((uint8_t)0x00); //Length B
    message.append((uint8_t)0x00); //Group A
    message.append((uint8_t)0x46); //Group B
    message.append((uint8_t)0x01); //Sequence
    message.append((uint8_t)0x00); //Message ID

    QCborStreamWriter datamap(&cbor);
    datamap.startMap();
    datamap.endMap();

    message[3] = (uint8_t)cbor.length();
    message.append(cbor);

    SendMessage(message);
}
```

lairdjm, 2022-03-16:
Indeed seems to be a bug, it's stuck on finding the group but will not get past 70:
```
mgmt_find_group (command_id=0, group_id=70) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/src/mgmt.c:100
100             for (group = mgmt_group_list; group != NULL; group = group->mg_next) {
(gdb) bt
#0  mgmt_find_group (command_id=0, group_id=70) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/src/mgmt.c:100
#1  mgmt_find_handler (group_id=<optimized out>, command_id=0) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/src/mgmt.c:134
#2  0x00015bca in smp_handle_single_payload (handler_found=<synthetic pointer>, req_hdr=0x20004dbc <sys_work_q_stack+2076>, 
    cbuf=0x20004ddc <sys_work_q_stack+2108>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:213
#3  smp_handle_single_req (handler_found=<synthetic pointer>, req_hdr=0x20004dbc <sys_work_q_stack+2076>, 
    streamer=0x20004e34 <sys_work_q_stack+2196>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:213
#4  smp_process_request_packet (streamer=streamer@entry=0x20004e34 <sys_work_q_stack+2196>, req=0x20006518 <_net_buf_pkt_pool>)
    at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:327
#5  0x00006ca4 in zephyr_smp_process_packet (nb=<optimized out>, zst=0x20001680 <smp_shell_transport>)
    at /tmp/aa/zephyr/subsys/mgmt/mcumgr/smp.c:249
#6  zephyr_smp_handle_reqs (work=0x20001680 <smp_shell_transport>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/smp.c:265
#7  0x00010d88 in work_queue_main (workq_ptr=0x20000fe0 <k_sys_work_q>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/kernel/work.c:649
#8  0x00012f6e in z_thread_entry (entry=0x10d05 <work_queue_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/lib/os/thread_entry.c:36
#9  0x00012f6e in z_thread_entry (entry=0x10d05 <work_queue_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/lib/os/thread_entry.c:36
#10 0x2aa640a6 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
#0  mgmt_find_group (command_id=0, group_id=70) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/src/mgmt.c:100
#1  mgmt_find_handler (group_id=<optimized out>, command_id=0) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/src/mgmt.c:134
#2  0x00015bca in smp_handle_single_payload (handler_found=<synthetic pointer>, req_hdr=0x20004dbc <sys_work_q_stack+2076>, 
    cbuf=0x20004ddc <sys_work_q_stack+2108>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:213
#3  smp_handle_single_req (handler_found=<synthetic pointer>, req_hdr=0x20004dbc <sys_work_q_stack+2076>, 
    streamer=0x20004e34 <sys_work_q_stack+2196>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:213
#4  smp_process_request_packet (streamer=streamer@entry=0x20004e34 <sys_work_q_stack+2196>, req=0x20006518 <_net_buf_pkt_pool>)
    at /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/smp/src/smp.c:327
#5  0x00006ca4 in zephyr_smp_process_packet (nb=<optimized out>, zst=0x20001680 <smp_shell_transport>)
    at /tmp/aa/zephyr/subsys/mgmt/mcumgr/smp.c:249
#6  zephyr_smp_handle_reqs (work=0x20001680 <smp_shell_transport>) at /tmp/aa/zephyr/subsys/mgmt/mcumgr/smp.c:265
#7  0x00010d88 in work_queue_main (workq_ptr=0x20000fe0 <k_sys_work_q>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/kernel/work.c:649
#8  0x00012f6e in z_thread_entry (entry=0x10d05 <work_queue_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/lib/os/thread_entry.c:36
#9  0x00012f6e in z_thread_entry (entry=0x10d05 <work_queue_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>)
    at /tmp/aa/zephyr/lib/os/thread_entry.c:36
#10 0x2aa640a6 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) c
```

closed 2022-03-24 by carlescufi