zephyrproject-rtos/zephyr #43887: SystemView tracing with STM32L0x fails to compile

k0d, 2022-03-16:
**Describe the bug**
I'm trying to enable tracing using systemview on a STM32L071 board.
I can compile the same code for an nrf9160dk board and it works fine.
It fails with the console log shown below.

**To Reproduce**
have this in your project config
```
CONFIG_THREAD_NAME=y
CONFIG_SEGGER_SYSTEMVIEW=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_TRACING=y
CONFIG_SEGGER_SYSVIEW_POST_MORTEM_MODE=n
CONFIG_SEGGER_SYSTEMVIEW_BOOT_ENABLE=y
```

**Expected behavior**
It should compile without errors.

**Logs and console output**
```
❯ west build .
-- west flash: rebuilding
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): <redacted>/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: <redacted>
-- Zephyr version: 3.0.0 (<redacted>/zephyrproject/zephyr)
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: <redacted>_<redacted>
-- Cache files will be written to: <redacted>/Library/Caches/zephyr
-- Found dtc: /opt/homebrew/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/Applications/ARM)
-- Found BOARD.dts: <redacted>/boards/arm/<redacted>_<redacted>/<redacted>_<redacted>.dts
node '/board' compatible '<redacted>,<redacted>' has unknown vendor prefix '<redacted>'
-- Generated zephyr.dts: <redacted>/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: <redacted>/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: <redacted>/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: <redacted>/build/zephyr/dts.cmake
<redacted>/build/zephyr/zephyr.dts:677.5-26: Warning (i2c_bus_reg): /soc/i2c@40005400/st25dv@53:reg: I2C address must be less than 7-bits, got "0x200". Set I2C_TEN_BIT_ADDRESS for 10 bit addresses or fix the property
<redacted>/build/zephyr/zephyr.dts:73.39-78.5: Warning (interrupt_provider): /soc/interrupt-controller@40010400: Missing #address-cells in interrupt provider
Parsing <redacted>/zephyrproject/zephyr/Kconfig
Loaded configuration '<redacted>/boards/arm/<redacted>_<redacted>/<redacted>_<redacted>_defconfig'
Merged configuration '<redacted>/prj.conf'
Configuration saved to '<redacted>/build/zephyr/.config'
Kconfig header saved to '<redacted>/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: <redacted>/build
[30/154] Building C object modules/segger/CMake...odules/debug/segger/SEGGER/SEGGER_SYSVIEW.c.obj
In file included from <redacted>/zephyrproject/zephyr/include/tracing/tracing_syscall.h:11,
                 from <redacted>/build/zephyr/include/generated/syscalls/atomic_c.h:7,
                 from <redacted>/zephyrproject/zephyr/include/sys/atomic_c.h:75,
                 from <redacted>/zephyrproject/zephyr/include/sys/atomic.h:31,
                 from <redacted>/zephyrproject/zephyr/include/kernel_includes.h:21,
                 from <redacted>/zephyrproject/zephyr/include/kernel.h:17,
                 from <redacted>/zephyrproject/modules/debug/segger/Config/SEGGER_RTT_Conf.h:150,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW_ConfDefaults.h:66,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW.h:65,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW_Int.h:64,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW.c:146:
<redacted>/zephyrproject/zephyr/include/sys/atomic.h: In function 'atomic_test_and_clear_bit':
<redacted>/zephyrproject/zephyr/subsys/tracing/sysview/./tracing_sysview_syscall.h:14:2: warning: implicit declaration of function 'SEGGER_SYSVIEW_RecordU32' [-Wimplicit-function-declaration]
   14 |  SEGGER_SYSVIEW_RecordU32(TID_SYSCALL, (uint32_t)id)
      |  ^~~~~~~~~~~~~~~~~~~~~~~~
<redacted>/build/zephyr/include/generated/syscalls/atomic_c.h:234:61: note: in expansion of macro 'sys_port_trace_syscall_enter'
  234 | #define atomic_and(target, value) ({  atomic_val_t retval;  sys_port_trace_syscall_enter(K_SYSCALL_ATOMIC_AND, atomic_and, target, value);  retval = atomic_and(target, value);  sys_port_trace_syscall_exit(K_SYSCALL_ATOMIC_AND, atomic_and, target, value, retval);  retval; })
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
<redacted>/zephyrproject/zephyr/include/sys/atomic.h:151:8: note: in expansion of macro 'atomic_and'
  151 |  old = atomic_and(ATOMIC_ELEM(target, bit), ~mask);
      |        ^~~~~~~~~~
[... 150 lines cut ...]
  186 | static inline void atomic_clear_bit(atomic_t *target, int bit)
      |                    ^~~~~~~~~~~~~~~~
<redacted>/zephyrproject/zephyr/include/sys/atomic.h:186:20: error: static declaration of 'atomic_clear_bit' follows non-static declaration
In file included from <redacted>/zephyrproject/modules/debug/segger/Config/SEGGER_RTT_Conf.h:150,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW_ConfDefaults.h:66,
                 from <redacted>/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW.h:65,
                 from <redacted>/zephyrproject/zephyr/subsys/tracing/sysview/./tracing_sysview_syscall.h:10,
                 from <redacted>/zephyrproject/zephyr/include/tracing/tracing_syscall.h:11,
                 from <redacted>/build/zephyr/include/generated/syscalls/atomic_c.h:7,
                 from <redacted>/zephyrproject/zephyr/include/sys/atomic_c.h:75,
                 from <redacted>/zephyrproject/zephyr/include/sys/atomic.h:31,
                 from <redacted>/zephyrproject/zephyr/include/spinlock.h:15,
                 from <redacted>/zephyrproject/zephyr/kernel/atomic_c.c:23:
<redacted>/zephyrproject/zephyr/include/kernel.h:4034:4: note: previous implicit declaration of 'atomic_clear_bit' was here
 4034 |    atomic_clear_bit(&work->flags,
      |    ^~~~~~~~~~~~~~~~
[96/154] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/tracing/sysview/sysview.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: re-build in <redacted>/build failed (no --build-dir given)
```

**Environment (please complete the following information):**
 - OS: Mac OS 12.1
 - Toolchain GNU Arm
 - Tag: zephyr-v3.0.0

erwango, 2022-03-17:
Nice redacted log ;-)

erwango, 2022-03-17:
Ok, this seems linked to the use of CONFIG_SEGGER_SYSTEMVIEW with M0/M0+ target.
I'm hitting the same issue with `bbc_microbit`

erwango, 2022-03-17:
@microbuilder Can you have a look ?

kmeinhar, 2022-05-10:
Same issue on board `atsamd20_xpro`. Is there a quick fix or work around?

str4t0m, 2022-07-10:
@teburd Could you have a look at this. I think this broken Cortex-M0 since #39454.

ccoenenatbaumer, 2022-07-27:
This is caused by circular header dependencies. IMHO the segger systemview header should never direclty include zephyr kernel headers but what happens is 

```
In file included from ../include/zephyr/kernel_includes.h:37,
                 from ../include/zephyr/kernel.h:17,
                 from /home/user/workspace/zephyrproject/modules/debug/segger/Config/SEGGER_RTT_Conf.h:150,
                 from /home/user/workspace/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW_ConfDefaults.h:66,
                 from /home/user/workspace/zephyrproject/modules/debug/segger/SEGGER/SEGGER_SYSVIEW.h:65,
                 from ../subsys/tracing/sysview/./tracing_sysview_syscall.h:10,
                 from ../include/zephyr/tracing/tracing_syscall.h:11,
                 from zephyr/include/generated/syscalls/atomic_c.h:7,
                 from ../include/zephyr/sys/atomic_c.h:75,
                 from ../include/zephyr/sys/atomic.h:31,
                 from ../include/zephyr/kernel_structs.h:24,
                 from ../include/zephyr/arch/arm/aarch32/arch_inlines.h:10,
                 from ../include/zephyr/arch/arch_inlines.h:18,
                 from ../include/zephyr/sys/arch_interface.h:1265,
                 from ../include/zephyr/arch/cpu.h:12,
                 from /home/user/workspace/zephyrproject/zephyr/arch/common/isr_tables.c:10:
../include/zephyr/fatal.h:87:59: error: unknown type name 'z_arch_esf_t'
   87 | void k_sys_fatal_error_handler(unsigned int reason, const z_arch_esf_t *esf);
```
So regular code includes zephyr kernel code, which includes the tracing with segger, which then in SEGGER_RTT_Conf.h includes the kernel again. 

I prepared a fix which moves all the dependent code into the c files of the segger module. Unfortunately that needs changes in both zephyr and segger repositories:

[zephyr - fix_segger_dependencies](https://github.com/ccoenenatbaumer/zephyr/tree/fix_segger_dependencies)
[segger - fix_include_dependencies](https://github.com/ccoenenatbaumer/segger/tree/fix_include_dependencies)

Let me know if it works for you, then I will start a merge request.

microbuilder, 2022-07-28:
@ccoenenatbaumer Do you mind making a pair of PRs for these changes?

ccoenenatbaumer, 2022-07-29:
With the changes it builds again, but I am unable to verify if Segger SystemView is still working properly. Would be great if someone could do that.

microbuilder, 2022-07-29:
I was previously able to compile on the M33 (all I had on hand at the time with a built-in J-Link was the `nrf5340dk_nrf5340_cpuapp`), but I haven't been able to get SystemView (v3.32) to work.

KConfig Settings:

```
# Segger SystemView
CONFIG_THREAD_NAME=y
CONFIG_SEGGER_SYSTEMVIEW=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_TRACING=y
CONFIG_TRACING_BACKEND_RAM=y

# Optional: Start logging SystemView events on system start
CONFIG_SEGGER_SYSTEMVIEW_BOOT_ENABLE=y

# Optional: `CONFIG_SEGGER_SYSVIEW_POST_MORTEM_MODE` enables the debugger
# to be attached after the system has crashed using ``west attach`` for
# post-mortum analysis in SystemView.
CONFIG_SEGGER_SYSVIEW_POST_MORTEM_MODE=n
```

As per the tracing documentation, I also updated this file:

```bash
$ cp $ZEPHYR_BASE/subsys/tracing/sysview/SYSVIEW_Zephyr.txt ~/.config/SEGGER/
```

I haven't used SystemView with Zephyr before, so I'm not sure if one of these settings is wrong, or if this has been silently broken for a while, or if it's just something specific to MacOS in my case.

microbuilder, 2022-07-29:
@nordic-krch Are you responsible for SystemView as well? Just curious if the above looks right to you. Wasn't able to get any tracing input working to test on the nRF53, but don't know when this was last known to work.

str4t0m, 2022-07-29:
@ccoenenatbaumer Thanks for your proposal. I picked your changes and in a quick test with the synchronization sample everything seems to work.
So I would suggest you to open a PR. Even if you have not tested the changes fully you can always open a draft PR, because that simplifies testing and giving feedback for others.

microbuilder, 2022-07-29:
> @ccoenenatbaumer Thanks for your proposal. I picked your changes and in a quick test with the synchronization sample everything seems to work. So I would suggest you to open a PR. Even if you have [...]

What config flags did you add that you were able to get SystemView working? Couldn't get any output myself with the settings here https://github.com/zephyrproject-rtos/zephyr/issues/43887#issuecomment-1199059992

str4t0m, 2022-07-29:
I was testing with `samples/synchronization` adding an overlay file with the following options:

```
CONFIG_THREAD_NAME=y
CONFIG_TRACING=y
CONFIG_SEGGER_SYSTEMVIEW=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_SEGGER_SYSTEMVIEW_BOOT_ENABLE=y
```
With SystemView V3.32 on Linux.
Have you tried with Cortex-M4 or a Cortex-M0+? Those targets have worked quite reliable in the few tests I have performed.
Also tried with an STM32U5 target (CM33), but there it had issues finding the buffer and only worked from time to time.

ccoenenatbaumer, 2022-08-01:
I could successfully test with nucleo_g431rb and created two pull requests

#48513
zephyrproject-rtos/segger#13

PR #48513 (2022-08-02): segger: Resolve circular include dependencies
CONFIG_SEGGER_SYSTEMVIEW=y does not compile for Cortex-M targets anymore. This is caused by circular include dependencies with the SEGGER module.

Zephyr kernel is dependend on trace.
Trace is dependend on segger rtt.
Segger rtt MUST NOT be dependend on zephyr kernel.

Move lock functions from header into c file to avoid circular
dependency.

Fixes #43887.

This fix needs an update of the segger repository (see zephyrproject-rtos/segger#13), west.yml has to be updated after this.
commit 81694a25 (2022-08-02): segger: Resolve circular include dependencies

Zephyr kernel is dependent on trace.
Trace is dependent on segger rtt.
Segger rtt MUST NOT be dependent on zephyr kernel.

Move lock functions from header into c file to avoid circular
dependency. This fix needs an update of the segger repository.

Fixes #43887.
closed 2022-08-01 by carlescufi