zephyrproject-rtos/zephyr #43898: Twister:  total test case number discrepancy the result xml.

smrtos, 2022-03-17:
**Describe the bug**
Run test with twister and generate the raw result XML. 

The top row of the XML like below indicates the total case number like below:

`<testsuite errors="8" failures="7" name="intel_adsp_ace15_mtpm_sim" skipped="18352" tests="19453" time="5008.572402">`

But the total number by counting below XML node is not the same as above.

`<testcase classname=...`

Here's an example:
https://github.com/smrtos/zephyr_rtos_bug_repro/tree/main/twister.test.case.number.discrepancy)

**Expected behavior**
The `tests` attribute should indicate the same number of `<testcase>` node.

**Impact**
Inconsistent test statistics.

carlescufi, 2022-03-23:
Maybe fixed by https://github.com/zephyrproject-rtos/zephyr/pull/44094

smrtos, 2022-03-29:
Here's a script to ease the validation of the raw XML result.
https://github.com/intel-sandbox/sm_zephyr_tools/blob/main/Traceability/zRawTestXmlStats.py

nashif, 2022-03-29:
with some additions to https://github.com/zephyrproject-rtos/zephyr/pull/44094, this might address any problems now, at least we will get warnings about discrepancies

nashif, 2022-04-05:
lots of the discrepancy can be attributed to malformed tests, but also to some accounting bugs in twister, so this should be addressed once https://github.com/zephyrproject-rtos/zephyr/pull/44094 is merged.

closed 2022-05-13 by nashif