zephyrproject-rtos/zephyr #43899: can: stm32: Build issue on g4 target

erwango, 2022-03-17:
**Describe the bug**
`twister -b -p nucleo_g474re  -T samples/drivers/can/sample.drivers.can`

```
/__w/zephyr/zephyr/drivers/can/can_stm32fd.c
/__w/zephyr/zephyr/drivers/can/can_stm32fd.c: In function 'can_stm32fd_get_max_bitrate':
/__w/zephyr/zephyr/drivers/can/can_stm32fd.c:161:20: error: 'const struct can_stm32fd_config' has no member named 'mcan'
  161 |  *max_bitrate = cfg->mcan.max_bitrate;
      |                    ^~
/__w/zephyr/zephyr/drivers/can/can_stm32fd.c: In function 'can_stm32fd_line_1_isr':
/__w/zephyr/zephyr/drivers/can/can_stm32fd.c:181:37: error: 'data' undeclared (first use in this function)
  181 |  struct can_mcan_data *mcan_data = &data->mcan_data;
      |                                     ^~~~
```
**Impact**
CI blocked

erwango, 2022-03-17:
@alexanderwachter Would you  have time to have a look ?

henrikbrixandersen, 2022-03-17:
@erwango My apologies. Fix is in #43901.

PR #43901 (2022-03-17): drivers: can: stm32fd: fix compilation
Fix compilation of the STM32 CAN-FD driver front-end.

Fixes: #43899
commit e3f93d76 (2022-03-17): drivers: can: stm32fd: fix compilation

Fix compilation of the STM32 CAN-FD driver front-end.

Fixes: #43899
closed 2022-03-17 by carlescufi