zephyrproject-rtos/zephyr #43910: civetweb/http_server - DEBUG_OPTIMIZATIONS enabled

paveldedourek-eaton, 2022-03-17:
**Describe the bug**
Compilation failed during compilation of **samples/net/civetweb/http_server** when DEBUG_OPTIMIZATIONS is enabled.

Platform: nucleo_f429zi

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_f429zi zephyr/samples/net/civetweb/http_server -t menuconfig
2. enable DEBUG_PTIMIZATIONS 
3. west build -b nucleo_f429zi zephyr/samples/net/civetweb/http_server

**Expected behavior**
I expect to compile it without problems.

**Impact**
I am not able to debug my application without optimizations.

**Logs and console output**
e9924876@~/zephyrproject$ west build -b nucleo_f429zi zephyr/samples/net/civetweb/http_server
/home/e9924876/.local/bin/west

[0/1] Re-running CMake...
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/e9924876/zephyrproject/zephyr/samples/net/civetweb/http_server
-- Cache files will be written to: /home/e9924876/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/e9924876/zephyrproject/zephyr)
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nucleo_f429zi
-- Found host-tools: zephyr 0.13.1 (/home/e9924876/zephyr-sdk)
-- Found dtc: /home/e9924876/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.1 (/home/e9924876/zephyr-sdk)
-- Found BOARD.dts: /home/e9924876/zephyrproject/zephyr/boards/arm/nucleo_f429zi/nucleo_f429zi.dts
-- Generated zephyr.dts: /home/e9924876/zephyrproject/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/e9924876/zephyrproject/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/e9924876/zephyrproject/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/e9924876/zephyrproject/build/zephyr/dts.cmake
Parsing /home/e9924876/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/e9924876/zephyrproject/build/zephyr/.config'
No change to configuration in '/home/e9924876/zephyrproject/build/zephyr/.config'
Kconfig header saved to '/home/e9924876/zephyrproject/build/zephyr/include/generated/autoconf.h'
-- Target architecture - arm
-- Enabling tests in the build - ON
-- Thread Stack Size - 102400
-- Enabling server executable - ON
-- Serve no static files - OFF
-- Disable CGI support - OFF
-- Disable caching support - OFF
-- C++ wrappers - OFF
-- IP Version 6 - OFF
-- Websockets support - OFF
-- Server statistics support - OFF
-- Memory Debugging - OFF
-- ASAN in debug mode - ON
-- Force x32 / x64 architecture - OFF
-- Lua CGI support - OFF
-- Executable installation - ON
-- Build if there are warnings - ON
-- Duktape CGI support - OFF
-- SSL support - ON
-- Compile for OpenSSL 1.1 API - OFF
-- Dynamically load SSL libraries - ON
-- Configuring C Compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/e9924876/zephyrproject/build
[201/211] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/e9924876/zephyrproject/build/zephyr/zephyr_pre0.map 
: && ccache /home/e9924876/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/e9924876/zephyrproject/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/config/libsubsys__net__lib__config.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/ethernet/libdrivers__ethernet.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/civetweb/lib..__modules__lib__civetweb.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/e9924876/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/home/e9924876/zephyrproject/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/e9924876/zephyrproject/build/zephyr && /home/e9924876/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E echo
/home/e9924876/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `pull_inner':
/home/e9924876/zephyrproject/modules/lib/civetweb/src/civetweb.c:6426: undefined reference to `fileno'
/home/e9924876/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `push_inner':
/home/e9924876/zephyrproject/modules/lib/civetweb/src/civetweb.c:6267: undefined reference to `ferror'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/e9924876/.local/bin/cmake --build /home/e9924876/zephyrproject/build
e9924876@~/zephyrproject$ 

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK - 3.0.0-rc3-49-g1b6944b03e8f
 - Commit SHA or Version used - c17bc7ad2ad9f331396efc13c197d55dd63dddc1 (~/zephyrproject/zephyr)

carlescufi, 2022-03-23:
@Nukersson could you please take a look?

Nukersson, 2022-03-23:
Hi and thank you for your problem description. Unfortunately I can help only by pointing to the way for solution since I'am very busy in other projects now.
Here what happens:
1. Enabling debug means the compiler will use -O1 or -O0 at some point. This means, the "dead" or potentially unused code will be compiled also.
2. Since most everything in civertweb is compiled two things happening:
    a. The footprint increases drastically. The civertweb is very very hungry for your RAM ind may not fit in your MCU anymore.
    b. The lines of code with maybe not implemented libc (e.g. newlib) functionality will be compiled. This is what happening in your case.
3. Since 2.b. holds, linker (see led message) misses some implementations / definitions and throws an error. This is the last error message you get.

Solution:
Re-Inplementing missed code parts and compiling in the best hope that 2.a. will not destroy the allowed allowed footprint.

As I said previously, I can't invested time here anymore, cause this task may take some days for implementation. Moreover, civertweb is not optimal for Zephyr in terms of footprint. I would prefer to implement websocket server for Zephyr from scratch to save memory.

mbolivar-nordic, 2022-04-05:
@mbolivar-nordic to seek a maintainer

fkokosinski, 2022-05-31:
Hi @paveldedourek-eaton, thanks for reporting this issue.

I did a little digging and it looks like civetweb requires the usage of minimal libc that Zephyr provides, which does not implement `fileno` and `ferror` functions. I could force this sample to build by defining them directly in `modules/lib/civetweb/src/civetweb.c`:
```
#define fileno(x) (*x)
#define ferror(x) (0)
```

Could you let me know if this solves this issue on your side as well?

PR #46746 (2022-06-22): civetweb: remove obsolete code
Retrying https://github.com/zephyrproject-rtos/zephyr/pull/45988 now that we have released v3.1 and nobody has stepped up to maintain civetweb. (I can't reopen the old PR due to GitHub restrictions.)

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they pro [... cut]
commit f49d11d3 (2022-06-22): civetweb: remove obsolete code

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they promise to
maintain it going forward.

Many thanks to everyone who has contributed to civetweb support in
Zephyr while it was here. So long and thanks for all the fish.

Fixes: #45807
Fixes: #43910
Fixes [... cut]
closed 2022-06-22 by mbolivar-nordic