zephyrproject-rtos/zephyr #43912: LTSv2 Service Pack 1 WishList

cfriedt, 2022-03-17:
* LVGLv8

brgl, 2022-03-18:
Another changeset we'd be interested in getting into LTSv2 is the following PR: https://github.com/zephyrproject-rtos/zephyr/pull/43927 Fortunately this is straightforward.

brgl, 2022-03-18:
WRT LVGL v8 in 2.7: the bit about the changes needing to be atomic and minimal is a no-go. While eventually we ended up moving almost of the lvgl code out of the zephyr tree for 3.1, the changes that stayed in-tree encompas many defconfigs, samples etc. I did a minimal, squashed port of the LVGL changes to 3.0 and it still amounts to 800 LOC changed. Is this still acceptable?

cfriedt, 2022-03-20:
> WRT LVGL v8 in 2.7: the bit about the changes needing to be atomic and minimal is a no-go. While eventually we ended up moving almost of the lvgl code out of the zephyr tree for 3.1, the changes [...]

I wonder if we could remove what is necessary to adopt the approach taken in 3.1 - a module sounds far more appealing. I would definitely consider it.

Minimal in this case should also mean "minimal effort", so if the work has been done and it's not too, too hard to apply to v2.7-branch, that would work too.

cfriedt, 2023-06-01:
This is likely not going to happen - closing.

closed 2023-06-01 by cfriedt