zephyrproject-rtos/zephyr #43928: pm: going to PM_STATE_SOFT_OFF in pm_policy_next_state causes assert in some cases

emillindq, 2022-03-17:
**Describe the bug**
Using the PM, it is possible to suspend devices. This can be done in two ways:
1. use the `pm_device_action_run` on a device
2. automatically when entering a system-wide soft-off by returning this as next state in `pm_policy_next_state`

A lot of drivers implement a pm callback for setting the underlying IC in different power states. The problem is that a lot of these drivers (for example vcnl4040) assume that this is not done from the idle thread. From the idle thread, things like `k_sem_take`, `k_msleep` and so on can't be used because an assert in sched.c checks if the current thread is the idle thread.
https://github.com/zephyrproject-rtos/zephyr/blob/e8f93a915def87b9c7e1bce3323bec996d7863da/kernel/sched.c#L186

Now, there are workarounds that can resolve this, like done here https://github.com/zephyrproject-rtos/zephyr/issues/41944 but not being able to use thread objects when powering down is a huge pain. Want to use i2c? Forget it.

In my project, I call a self-developed callback to manually put the devices into SUSPEND mode before I signal to pm_policy_next_state to enter SOFT_OFF so it's done in a thread-context. But there has to be a better way to do this on a kernel level. Maybe launch a task with co-op priority which suspends all devices?

Or am I missing something?

**To Reproduce**
Steps to reproduce the behavior:
1. Build sample for vcnl4040
2. implement pm_policy_next_state and return PM_STATE_SOFT_OFF from it
3. see assert

**Expected behavior**
The kernel can be used as usual when suspending devices, when entering system-wide SOFT_OFF

**Impact**
There are workarounds, as mentioned, but there has to be a better way to do this

**Environment (please complete the following information):**
 - OS: Ubuntu 21.04
 - Zephyr SDK
 - Zephyr v3.0.0

emillindq, 2022-03-24:
@carlescufi we are building for a custom board we have made with the vcnl4040 mounted on (using the stm32h743, but I don't think that matters). But to verify, if anybody has has the adafruit feather board, that one has the vcnl4040 on it.

Thanks!

gmarull, 2022-03-24:
FYI, https://github.com/zephyrproject-rtos/zephyr/pull/43724 proposes a new API to detect wether you are allowed to block or not on the pm_action callback.

JordanYates, 2022-03-24:
> FYI, #43724 proposes a new API to detect wether you are allowed to block or not on the pm_action callback.

It allows you to detect that condition, but doesn't fix the underlying problem.
IMO the best option is to use device runtime PM for any devices that need to block in transitions, and be explicit in your application code about when they need to powered.

gmarull, 2022-03-25:
> It allows you to detect that condition, but doesn't fix the underlying problem. IMO the best option is to use device runtime PM for any devices that need to block in transitions, and be explicit in [...]

Agreed

emillindq, 2022-03-27:
In my opinion this is a fairly wide topic that would benefit from as much input as possible from experienced contributors.

My take on this is that the PM action callback when entering system-wide soft-off has to run in a task with highest priority. A work would be the best. I know the system queue is usually at a cooperative priority, and the inherent use case as such is to schedule work which is not blocking for a long time, so it should be available. How about running this work there? If some pm action callback cancels work on the system workq it's not started yet so it will not deadlock.

ceolin, 2022-03-28:
Actually it runs with the lowest priority, basically it only runs if there is nothing else to be executed. What happens if you have a separated workq suspending devices and have another task running  that require a particular device that was suspended ?

closed 2022-06-11 by github-actions[bot]

Oleh-Kravchenko, 2025-11-18:
Lol,
it reproducing on my custom board! v3.7.1

`ASSERTION FAIL [!z_is_idle_thread_object(thread)] @ WEST_TOPDIR/zephyr/kernel/sched.c:121`