zephyrproject-rtos/zephyr #43940: Support for CH32V307 devices

Mango-kid, 2022-03-17:
This Issue is for adding support for the ch32v307 series devices. This is a very interesting IoT CPU and has an integrated 10M ethernet PHY.

I am currently working on a dev board for the ch32v307 that would allow for easy and low cost real-time ethernet applications.

I am willing to maintain support for this new SOC. I am experienced with general embedded system development but this is my first time doing anything major for adding support for an SOC on Zephyr, so any help with a plan of attack would be great.  

Also if anyone is willing to work with me and/or validate any PR, I would be happy to provide hardware for testing. 

**Additional context**
There is already some peripheral support on github, but it seems very similar to the ST HAL drivers and I think there may be some controversy surrounding the licensing in the HAL. But I am also assuming that the HAL would not be used directly in Zephyr. 

[CH32V307 Github](https://github.com/openwch/ch32v307)

closed 2022-05-31 by github-actions[bot]

fdarling, 2022-07-19:
I can't find any mention of support for the CH32V307 other than in the release notes referring to this issue. Are any of the peripherals supported? I am interested specifically in Ethernet...

dragonlock2, 2024-01-25:
This might be a step in the right direction, I was adding CH32X035 support which uses a similar core. Based on the non-standard way CH32 handles interrupts, I had to do a workaround documented here [zephyrboards/soc/riscv/wch_ch32/common/irq.c](https://github.com/dragonlock2/zephyrboards/blob/main/soc/riscv/wch_ch32/common/irq.c#L25). Not sure I want to try mainlining it due to how hacky it feels 😅