zephyrproject-rtos/zephyr #43959: Support Zephyr in path containing spaces

tejlmand, 2022-03-18:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr supports specifying CONF_FILE and other user arguments with space separation, like:
```
cmake -DCONF_FILE="a.conf b.conf"
```
whereas the official way of specifying a list in CMake is using `;`, like this:
```
cmake -DCONF_FILE="a.conf;b.conf"
```

The use of space is convenient to many users, especially on Linux systems where space is rare in path.
But on Windows, a space in path is very common and creates confusion and annoyance to users who places Zephyr for example in a home dir, and that home dir contains spaces.

The main issue is that relative paths are expanded inside Zephyr CMake, for example using `${APPLICATION_CONFIG_DIR}` or `${CMAKE_CURRENT_LIST_DIR}`, not to mention the use case where a file is specified directly at application top-level CMakeLists.txt, as:
```
set(CONF_FILE ${CMAKE_CURRENT_LIST_DIR}/somefile.conf)
set(OVERLAY_CONFIG ${CMAKE_CURRENT_LIST_DIR}/extra.conf)
```
in such occasions, a space in the path is problematic when later splitting the setting based on spaces, as the.
CMake supports spaces in path, but decisions taken in the past prevents full use of CMake support.

**Describe the solution you'd like**

To both support existing usage, while also be able to make full use of CMake path handling, it would be desirable to have a global setting, for example: `SPACE_SEPARATED_ARGUMENTS=[True|False]`.
Users that needs path with spaces can the set this setting to `False`, and keep current behavior for current users, default is existing behavior.

The setting should be configurable through:
```
Required:
set(SPACE_SEPARATED_ARGUMENTS <val>)    # in CMakeLists.txt files
-DSPACE_SEPARATED_ARGUMENTS=<val>       # As command line argument
export SPACE_SEPARATED_ARGUMENTS=<val>  # As environment setting

Optional:
west config cmake.space_separated_arguments <val> # As west workspace configuration setting
```

In future, we can consider if we should deprecate the use of space, as users becomes more familiar with CMake lists.
From a interface perspective, there is little difference in writing `-DCONF_FILE="a.conf;b.conf"` vs `-DCONF_FILE="a.conf b.conf"`, but there is a big difference in the freedom to users as to where they may install Zephyr repo.

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

sslupsky, 2022-07-13:
Yeah, this is a good idea.  I encounter this problem using macOS where spaces in the path are common.  It is particularly difficult to work around if you try to use iCloud since macOS expands the iCloud path with "Mobile Documents".

Kevyn-Harder-UofA, 2024-09-05:
Any progress on this?