zephyrproject-rtos/zephyr #43964: k_timer callback timing gets unreliable with more cores active

kv2019i, 2022-03-18:
**Describe the bug**
When we increase number of active CPUs, the timing of the periodic k_timer degrade.

This seems to be related to the way each core advances system clock with sys_clock_announce().

Example of calculated average and max detected delay between two k_timer callbacks measured over 1sec period (measured in units of 38.4Mhz clock, so 38400=1msec):
```
     1024128.333333] (         923.020813) c3 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38399, max 38562
[     2048127.708333] (         922.656250) c3 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38400, max 38421
[     3072127.864583] (         922.552063) c3 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38400, max 38525
[     4096127.500000] (         922.161438) c3 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38399, max 38420
```
When four cores are loaded with load (<20% load per core, but something to run every 1ms):
```
[   282757402.708333] (         826.328125) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38840, max 40114
[   283793606.640625] (         871.770813) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38857, max 40115
[   284829285.390625] (         854.557312) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38837, max 40115
[   285865183.307292] (         845.807312) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38846, max 40106
[   286901603.046875] (         836.510437) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38865, max 40127
[   287937902.578125] (         838.984375) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38861, max 40121
[   288974742.552083] (         837.109375) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38881, max 40108
[   290011384.010417] (         854.427063) c2 zll-schedule       ......../zephyr_domain.c:152  INFO domain tick avg 38874, max 40122
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build SOF for cAVS2.5 with Zephyr main (TGL or ADL system)
2. use sof-tgl-nocodec.tplg
3. Run four concurrent streams:
```
arecord -D hw:0,10 -c 4 -r 48000 -f S16_LE /dev/null -q 
arecord -D hw:0,11 -c 4 -r 16000 -f S16_LE /dev/null -q
arecord -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q
arecord -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/null -q
```

**Expected behavior**
SOF expects k_timer timing to be consistent independently of how many cores are used and what is their load.

**Impact**
Blocker issue for cAVS2.5 use in SOF releases.

**Additional context**
Discovered with SOF, details in bug:
https://github.com/thesofproject/sof/issues/5556

kv2019i, 2022-03-18:
@andyross  @dcpleung  can you take a look and/or comment
FYI @lyakh  @lgirdwood

lgirdwood, 2022-03-18:
@kv2019i good find. I would expect the max to align with the average. I assume there is no IDC or other IRQs that could add latency that could impact the results ? 
@marcinszkudlinski fyi

kv2019i, 2022-03-18:
@andyross The driver indeed looks correct. One thing we've been looking at with @lyakh is how the conversions between ticks and 38.4Mhz cycles are done. With 4 cores active, timer interrupt is handled on all cores, all 4 cores update last_count and report advancement of time to Zephyr timer core. In all steps, there's a rounding from cycles to ticks and back and this is one suspect that could build up with more cores active.

The k_timer usage in SOF is here:
https://github.com/thesofproject/sof/blob/main/src/schedule/zephyr_domain.c#L164

# zephyr_domain_timer_fn
https://github.com/thesofproject/sof/blob/main/src/schedule/zephyr_domain.c#L98

I did not find any ready infra for interrupt firing statistics, but I did some local edits and confirmed that rate of interrupts is not unexpectedly high. We get 1-5 irqs per 1ms per DSP in this use-case. The DSP load is <25% on all DSPs, so it's not also a case where some of the DSPs are so busy that they wouldn't able to have cycles for timer bookkeeping and ISRs.

kv2019i, 2022-03-18:
Curiously, if I load 3 cores out of 4, the average stays as expected:
```
core0 compare_isr avg 38414, max 42120
core1 compare_isr avg 38415, max 44789
core3 compare_isr avg 38426, max 44678
core0 compare_isr avg 38424, max 40817
core1 compare_isr avg 38423, max 41878
core3 compare_isr avg 38412, max 41561
core0 compare_isr avg 38414, max 42128
core1 compare_isr avg 38415, max 44798
```

With two DSPs with active audio pipelines:
```
core0 compare_isr avg 38410, max 41850
core1 compare_isr avg 38410, max 44285
core3 compare_isr avg 38408, max 42358
core0 compare_isr avg 38408, max 42085
core1 compare_isr avg 38408, max 44277
core3 compare_isr avg 38410, max 42276
core0 compare_isr avg 38410, max 42274
core1 compare_isr avg 38410, max 44274
```

And just single DSP loaded:
```
core0 compare_isr avg 38402, max 40863
core3 compare_isr avg 38402, max 40705
core0 compare_isr avg 38402, max 40788
core3 compare_isr avg 38402, max 40691
core0 compare_isr avg 38402, max 40783
core3 compare_isr avg 38402, max 40710
core0 compare_isr avg 38402, max 40789
core3 compare_isr avg 38402, max 40703
```

And with four cores loaded the avg jumps up:
```
core0 compare_isr avg 39033, max 43064
core3 compare_isr avg 39038, max 43160
core1 compare_isr avg 39037, max 43910
core2 compare_isr avg 39032, max 43313
core0 compare_isr avg 39047, max 42905
core3 compare_isr avg 39048, max 43351
core1 compare_isr avg 39048, max 42748
core2 compare_isr avg 39052, max 43865
core0 compare_isr avg 39016, max 43232
core3 compare_isr avg 39015, max 43289
core1 compare_isr avg 39016, max 42800
core2 compare_isr avg 39016, max 43450
```

andyross, 2022-03-18:
That's an interesting clue.  I know core 3 has a weird runtime configuration due to the clock stuff.  Maybe... there's an interrupt latched on it that is firing but not getting cleared and causing bus contention?  Eh... no, we'd expect a spurious interrupt to panic.  But maybe something along that axis...

andyross, 2022-03-18:
I wrote up a really quick rig to try to isolate this, and it seems like it's more or less working for me.  The hello_world app in this branch:

https://github.com/andyross/zephyr/pull/new/cavs-latency-rig

https://github.com/andyross/zephyr/blob/cavs-latency-rig/samples/hello_world/src/main.c

...has been modified to duplicate what I believe (from a **very** quick read of the zephyr_domain.c file, so please let me know what I got wrong) is the SOF scheduler architecture.  It spawns two threads pinned to each core, one is a preemptible worker thread that does nothing but spin on the CPU (I have these running at 100%, polling on CCOUNT to get delays independent of the HDA timer).  The other is a maximum priority thread that polls on a semaphore that is triggered out of a timer interrupt.

And letting this run for a while gives fairly reasonable results (errors are in wall clock cycles):

```
...
work0: worst 5258 avg 704 (N = 54085)
work1: worst 4964 avg 698 (N = 54085)
work2: worst 5258 avg 691 (N = 55083)
work3: worst 5310 avg 713 (N = 55083)
work0: worst 5258 avg 694 (N = 55084)
work1: worst 4964 avg 688 (N = 55084)
work2: worst 5258 avg 695 (N = 56062)
work3: worst 5310 avg 716 (N = 56062)
work0: worst 5258 avg 697 (N = 56063)
work1: worst 4964 avg 691 (N = 56063)
work2: worst 5258 avg 694 (N = 57047)
work3: worst 5310 avg 715 (N = 57046)
work0: worst 5258 avg 696 (N = 57047)
work1: worst 4964 avg 690 (N = 57047)
work2: worst 5258 avg 691 (N = 58034)
work3: worst 5310 avg 712 (N = 58034)
work0: worst 5258 avg 694 (N = 58035)
work1: worst 4964 avg 688 (N = 58035)
```

Now... those worst case numbers are indeed pretty bad and point to some kind of spurious latency glitch in the system.  It's possible that they are artifacts of app startup I guess (i.e. measuring the latency incurred by thread setup, cache filling, etc...), but even then the measured 5k cycles (~= 130us) is a **LONG** time for the kernel to be hanging a highest priority thread.  This is definitely something to investigate.

But what it's not seeing is the kind of deterministic delays that you're measuring.  Also, it looks basically identical with MP_NUM_CPUS=2 or 3.

My guess is we probably have at least two bugs in play: the one I see above which is producing high latencies spuriously, and the one visible in SOF which I'm guessing is going to turn out to be an interaction with some other driver or subsystem.

andyross, 2022-03-18:
One kibitz on the design, FWIW:  I note that you're creating a separate k_timer instance for each CPU in the system, and initializing them all asynchronously as you start up.  That means that if the calls to k_timer_start() happen on different system ticks, they're going to require multiple ISRs on adjacent ticks to process, which is wasteful.  Since the system is designed to run on a 1 kHz cadence, it might make sense to register just one timer and poke all the semaphores from that.

I don't think this is a root cause here, but it's worth noting.

Also: I'm noticing that we're probably being too aggressive with the tick rate.  At 50 kHz, we have only 768 HDA cycles per tick, which is right at the level of the average latencies measured above (though those are being pulled up by the spurious errors, I think).  Also, in cavs_timer, I see that the MIN_DELAY setting (how close to "now" the driver will attempt to set a timeout before rounding up to the next tick) is 384 cycles, or half a tick anyway.  Again, I don't see how this is causing an issue, but it's false precision.  My intuition would be that a ~12-24 kHz tick rate (and maybe a lower MIN_DELAY) would be a better choice, which keeps the tick rate well above the latency noise floor.  And AFAICT SOF doesn't care anyway, all timing is either done in cycles or out of the 1 kHz timer callbacks.

kv2019i, 2022-03-18:
@andyross  Thanks! This might indeed have some subtle factors. The failing test case in suite is actually passing most of the time, and this error occurred in a stress test that runs much longer iterations of the test suite. 

The results are actually not all that different. Your "avg %d" is a average of the error and 650-750 error actually is close to results I guess (38400+640=39050). So while 1ms would ideally be exactly 50 Zephyr ticks, we seem to regularly wake up one tick late, pushing the average up. But curiously, when we run 1-3 core loads, the timer fires like clockwork every 34800 cycles of the 38.4Mhz clock. The worst-case seem to also be aligned.

Some differences which might explain why you get the different avg also with less cores:
  - in SOF code, we create ONE timer, and FOUR threads, so the timer callback is called (on some CPU) once, and it then calls k_sem_give() on all cores  --- in your test rig, a separate timer is created for every core (shouldn't make a difference, but still pointing out as an implementation difference)
  -  your test rig actually loads the DSPs more heavily than the SOF case in question. maybe that's the reason why the avg error increase is seen with less cores

Hmm, maybe this is a feature of the timer that uses CONFIG_SYS_CLOCK_TICKS_PER_SEC (50000 in these builds) as its granularity.  

At 50000 ticks/sec, one tick error in timer delivery is not a big deal. What is somewhat unexpected is that error adds up cumulatively. I.e. we'd expect to get occasional late timer callbacks, but catch up on the next one, maintainig a steday 1000Hz rate. This does not seem to be happening.

andyross, 2022-03-18:
Yeah, we raced with the comments.  And indeed I was wrong and you share a single k_timer already; I was just confused.  I'll update my code, though I doubt that matters much.

The problem with the "added error" theory is that in a k_timer, this is all happening in an ISR.  Slow processing in the ll thread won't change anything, the only way you can get a slipped tick is if the interrupt itself (not the ISR handler, but the actual interrupt: the timer reset happens before the callback, see kernel/timer.c:z_timer_expiration_handler()) is more than MIN_DELAY (which is currently "half a tick") late.  And even at the current 10ns setting, that is kinda hard to imagine.

Which is sorta pointing back to the same place, I think.  We may have a latency hog in the system somewhere, maybe something keeping a spinlock way too long, etc...

andyross, 2022-03-18:
Though, if we do have a latency hog, one workaround would indeed be to tune down the tick rate to 12000 or so and try to give the ISR more time.  Worth an experiment anyway.

kv2019i, 2022-03-18:
@andyross how about the theory that we have a cumulative rounding error between HDA clock and 50000Hz sys ticks? I guess we could also increase the tick and the error should reduce this way.

andyross, 2022-03-18:
There's no rounding error though, 38400000 == 768 * 50000 exactly. :)

But in this case it doesn't matter, because the math in the driver is modular.  It checks the number of whole ticks that have elapsed, announces those, and sets the last_count variable to the time of the last announced tick.  Any extra is "left" in the hardware counter to be counted later.  The only opportunity I can see for slippage is one level up, in k_timer, where it just naively assumes that it's running at the start of an ISR and that "N more ticks from now" is a valid way to get a consistent stride of N ticks.  (It wouldn't be a bad idea to improve this code to do that though!).

andyross, 2022-03-18:
OK, so I think things are becoming clearer.  The most recent version of the branch corrects a bug in the averages: I was averaging the absolute error, but that's wrong.  If a single event is 100 cycles late (an expected/tolerable error), and then the system correct and the next one looks 100 cycles "early", then that's correct.  But it was reporting two errors of 100 cycles instead.  Changing the math to signed gives a correct aggregate error.

And indeed, doing this shows that the typical/median error is quite small, but that there are excursions.  And when those happen, we end up slipping the k_timer update by one tick.  So the average as computed does indeed settle out a little higher than zero, meaning that it's slipping time.

So add that to two changes:

1. Use a smaller MIN_DELAY of 256 cycles (based on some quick testing of how long it really takes).

2. Set CONFIG_SYS_CLOCK_TICKS_PER_SEC to 12000 instead

And now things look much better!  Here's a representative run:

```
work1: last   149 worst    748 avg     0 (N = 52000)
work0: last     9 worst   1066 avg     0 (N = 52000)
work3: last  -931 worst   1351 avg     0 (N = 52000)
work2: last   420 worst   1638 avg     0 (N = 52000)
work0: last     6 worst   1066 avg     0 (N = 54000)
work3: last -1032 worst   1351 avg     0 (N = 54000)
work2: last   -77 worst   1638 avg     0 (N = 54000)
work1: last  -136 worst    748 avg     0 (N = 54000)
work0: last    -8 worst   1066 avg     0 (N = 55999)
work3: last   278 worst   1351 avg     0 (N = 55999)
work2: last   141 worst   1689 avg     0 (N = 56000)
work1: last -1149 worst   1149 avg     0 (N = 56000)
```

Note that there is no slippage, the average settles out to exactly zero.

So I think this more or less proves out the "k_timer timeout handler is running much later than expected" theory.  And my guess is that the workaround above would be acceptable for SOF in the near term.

So then the question becomes: why is that handler late?  And I don't have an answer.  But... it's really late.  The worst case inter-event latency in that log is 1689 HDA cyles, which comes to 17600 CPU cycles!  That is a HUGE interrupt latency, even if in the context of an audio app it might be tolerable.  We need to find this thing.

Alternatively, we could augment the k_timer handler to correct its rescheduling.  This wouldn't be hard and would be a good robustness feature.  But it still doesn't fix the huge latency hog.

kv2019i, 2022-03-21:
@andyross  I can confirm that indeed with modified MIN_DELAY and 12000 sys-clock, I now see a stable 38400 avg also with the SOF test and measured in the SOF timer callback.

Not so good news is that the SOF test suite still fails. So either the rootcause is not the timer timing, or it's related to the occasional timing spike. Trying to dig up some data for either hypothesis today.

andyross, 2022-03-29:
I'm back at work finally.  If the workarounds didn't work, then my guess is that this timer symptom is co-causal with whatever is causing that giant interrupt latency.

@kv2019i I see that ISR timing rig.  Does that show the expected very-late-interrupts?  Or maybe is the problem something like a spinlock underneath the ISR entry?

kv2019i, 2022-03-30:
@andyross I got sidetracked a bit as well as the main bug I was debuggined turns out to be fixed by the k_sem coherency fix (https://github.com/thesofproject/sof/pull/5588). I.e. even if I remove all the timer workarounds and have the drift in the 1ms tick, the 4core test case still passes with the ksem allocation fix. 

The workaround to sys tick (and min-delay) does fix the problem of "higher than expected average", but I can still see the unexpected worst cases and this is also visible in the ISR timing rig. No idea on the rootcause of this.

andyross, 2022-04-05:
Seeing a little progress here.  The not-quite-root-but-rootish cause seems to be that IPI delivery on cavs25 (at least) has bursty/nondeterministic in some circumstances.  I'm having a terrible time getting this problem to show itself in a rig smaller than the one above, but it seems real.  What happens is that occasionally the IPIs that get sent by one timer interrupt end up tying up the hardware (presumably something in the interrupt delivery path) in such a way that the timer interrupt for the next tick doesn't arrive on core 0 until more than a tick late.

If I hack the system to not send the IPI and just poll a flag from the worker threads and yield, the issue goes away and everything proceeds normally.  Obviously that's not a fix, as the preemption of those workers is the design requirement.  But it shows that the IDC hardware is the culprit.

But the good part is: we send a **lot** of IPIs.  The design expectation was that these should be fast, so we issue one any time the scheduler state changes.  Which means that in this particular case, that timer interrupt is unpending four threads, and calling arch_sched_ipi() once for each.  And each of those then needs to ping the other three cores, so that's **twelve** interrupts, when we really only needed three.

So I think I have an idea: we can batch the "must send IPI" flag in the scheduler such that we don't execute it until we reach either a call to reschedule() or the end of an interrupt, which are already documented scheduling points.  No code will be relying on synchronous preemption at any other location, as that would break on uniprocessor setups anyway.  Should be a small patch, and a quick prototype seems to be working.

kv2019i, 2022-04-07:
It seems @teburd 's #44605 solves the average error aspect of this bug. With that patch and #43967 for measurement, I get following on a all-core (4) DSP load on cAVS2.5 system:

Without #44605 :

```
core0 compare_isr avg 39197, max 43412
core3 compare_isr avg 39196, max 42814
core1 compare_isr avg 39186, max 43166
core2 compare_isr avg 39186, max 43175
core0 compare_isr avg 39200, max 42799
core3 compare_isr avg 39200, max 42922
core1 compare_isr avg 39195, max 43093
core2 compare_isr avg 39194, max 43164
```

With #44605 : 
```
core0 compare_isr avg 38408, max 42441
core1 compare_isr avg 38408, max 42387
core3 compare_isr avg 38407, max 42621
core2 compare_isr avg 38407, max 46491
core0 compare_isr avg 38407, max 42138
core1 compare_isr avg 38403, max 41779
core3 compare_isr avg 38403, max 42133
```

So the average values now look fine and adjusting MIN_DELAY and SYSTICK are no longer needed.

The worst case spikes are still there, so that part of problem seems to remain (potentially explained by IPI/IDC delays as @andyross  described above).

andyross, 2022-04-07:
See #44653 for an evolved version of the same workaround.  Instead of skipping the announce call, all that's required is that we make sure that all the callbacks happen on a single core (something that app code is going to clearly expect anyway).

andyross, 2022-04-08:
#44653 is merged, and turns out to address the root cause even though I didn't see it at the time (see final comment in that bug, the commit message is an incomplete explanation!).  We can close this.

closed 2022-04-08 by andyross

kv2019i, 2022-04-11:
@andyross How about the worst-case spikes? The avg is now fixed, but I'm still seeing suprisingly high worst-case spikes:

core0 compare_isr avg 38399, max 39274
core3 compare_isr avg 38400, max 42168
core1 compare_isr avg 38400, max 42748
core2 compare_isr avg 38399, max 42660
core0 compare_isr avg 38400, max 39275
core3 compare_isr avg 38399, max 42291
core1 compare_isr avg 38399, max 41908
core2 compare_isr avg 38400, max 41987
core0 compare_isr avg 38400, max 39272
core3 compare_isr avg 38399, max 42285
core1 compare_isr avg 38399, max 41775
core2 compare_isr avg 38399, max 42332

Should we have a a separate bug for this? The spikes are still quite high, that's almost 10% delay (42332 max versus 38399 avg).