zephyrproject-rtos/zephyr #43976: [lwm2m_engine / sockets] Possibility to decrease timeout on connect()

petterssonandreas, 2022-03-18:
This is more of a question than feature request or anything else really.

I am using the `lwm2m_rd_client` and `lwm2m_engine`. I have noticed that the timeout for `connect()` when I try to register to a lwm2m server is quite long, about 2 minutes. I would like to decrease this.

Where does the ~2 minutes timeout come from? Is that from some standard/protocol?

I have tried to modify the code slightly, to try to update `SO_SNDTIMEO` just before connecting, using

```c
struct timeval timeout;
timeout.tv_sec  = 20;
timeout.tv_usec = 0;
ret = setsockopt(client_ctx->sock_fd, SOL_SOCKET, SO_SNDTIMEO, &timeout, sizeof(timeout));
```

Is this a valid approach? I get no errors, but still see no decrease in timeout of the `connect()` function.

I did notice in `include/net/socket.h` the following:

```c
/**
 * sockopt: Receive timeout
 * Applies to receive functions like recv(), but not to connect()
 */
#define SO_RCVTIMEO 20
/** sockopt: Send timeout */
#define SO_SNDTIMEO 21
```

Does that mean that there is no way to set the timeout of `connect()`?  If so, is that a limitation of Zephyr, or of sockets in general?

I am using NCS 1.8.0, and the nRF9160 modem, if that changes anything. Maybe Nordic hasn't implemented all sockopt support?

rlubos, 2022-03-23:
Are you using DTLS or not (I'm assuming that yes, as for regular UDP connect should not trigger any communication). So, if DTLS is used, the timeout is the effect of DTLS handshake and the retansmissions that take place in there. You can try to set `TLS_DTLS_HANDSHAKE_TIMEO` socket option to limit the amount of retransmissions taken (note that this option is specific to nRF91 modem, Zephyr took slightly different approach here).

Also, for the future, please ask NCS related questions on Nordic Devzone (https://devzone.nordicsemi.com/), nRF91 modem is not supported by vanilla Zephyr.

petterssonandreas, 2022-03-23:
@rlubos Thanks for the answer! Yes, I am using DTLS.

> Are you using DTLS or not (I'm assuming that yes, as for regular UDP connect should not trigger any communication).

Oh, OK, is this the reason for the comment in `include/net/sockets.h`? That since `connect()` doesn't trigger communication, it actually has no timeout, because it is immediate? And therefore `SO_RCVTIMEO` has no effect on it?

Thanks for the tip on `TLS_DTLS_HANDSHAKE_TIMEO`, I will try that!

Oh, sorry for asking in the wrong forum, I was a bit unsure myself. Since the comment I spoke about was in Zephyr code, I thought this had nothing to do with Nordic specific stuff.

rlubos, 2022-03-23:
> Oh, OK, is this the reason for the comment in `include/net/sockets.h`? That since `connect()` doesn't trigger communication, it actually has no timeout, because it is immediate?

No, the reason for the comment is simply that the timeout is not implemented for `connect()`. Not that my comment only applied for UDP socket, TCP socket does perform a TCP handshake on `connect()`.

petterssonandreas, 2022-03-23:
@rlubos OK, thanks!

I will close this issue, and possibly open a ticket on Nordic Devzone if I need more help.

closed 2022-03-23 by petterssonandreas