zephyrproject-rtos/zephyr #43980: No PWM signal on Nucleo F103RB using TIM1 CH2 PA9

GeorgeCGV, 2022-03-18:
**Describe the bug**

No PWM signal is observed in Zephyr using Nucleo F103RB board with enabled Timer 1 and CH2 on PA9.

**To Reproduce**

Create an overlay for the `nucleo_f103rb` board:

```
&timers1 {
    status = "okay";
    st,prescaler = <0>;
    pwm1: pwm {
        status = "okay";
        pinctrl-0 = <&tim1_ch1_pwm_pa8
                             &tim1_ch2_pwm_pa9
                             &tim1_ch3_pwm_pa10>;
        pinctrl-names = "default";
    }; 
};
```

Start PWM on channel 2 or all (doesn't matter) with `pwm_pin_set_<...>`.

Observe no signal on PA9.

[image]

**Expected behavior**

PWM signal on CH2 PA9 shall be generated as it happens in HAL generated project.

[image]

**Impact**

No CH2 PWM signal.

**Environment (please complete the following information):**
 - OS: `Windows`
 - Toolchain `arm-none-eabi-10.3-2021.07`
 - Commit SHA:  [49d08c2 ](https://github.com/zephyrproject-rtos/zephyr/commit/49d08c25fe86b2849f331a28dd73714a7c1f1e88)

FRASTM, 2022-03-21:
In the DTS of the nucleo_f103rb, there is 
```
&usart1 {
	pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};
```

So that the pb could be a pin conflict on **PA9** between usart1_tx_pa9 and tim1_ch2_pwm_pa9.
Could you try disabling the usart1 in the overlay 
```
&usart1 {
	status = "disabled";
};
```

GeorgeCGV, 2022-03-21:
@FRASTM, indeed, missed that... works 👍

closed 2022-03-21 by GeorgeCGV