zephyrproject-rtos/zephyr #44005: add strtoll and strtoull to libc minimal

MeisterBob, 2022-03-21:
## Origin

I don't know where the original source is, I took it from [strtol](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/source/stdlib/strtol.c)/[strtoul](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/source/stdlib/strtoul.c) and changed the types from `long` to `long long`.

## Purpose

`strtoll` and `strtoull` convert strings to `(unsigned) long long`.

## Mode of integration

The addition would be in the main tree.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/39612

## Description

The actual minimal libc has no strtoll/strtoull support, but i.e. civetweb requires those functions. One could use newlibc to gain access to strtoll/strtoull but civetweb has compatibility issue with newlibc. Even if those issues where solved the addition memory required by newlibc is way to much just for those two functions.

Also the lack of string conversion functions lead to multiple local implementations i.e. in https://github.com/zephyrproject-rtos/zephyr/pull/43289 and (libc_extension.h)[https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/civetweb/common/include/libc_extensions.h].

For further reference there is a [thread in discord](https://discord.com/channels/720317445772017664/954308485489786921) discussing this topic.

# Dependencies

none

## Revision

see PR

## License

BSD-4-Clause-UC

stephanosio, 2022-03-21:
cc @carlescufi @nashif

carlescufi, 2022-03-21:
Thank you @MeisterBob. This will be discussed on Wednesday.

kestewart, 2022-03-24:
@carlescufi - BSD-4-Clause-UC is NOT an approved OSI license.   It is going to impose a conditions on Zephyr downstream users to be compliant with using Zephyr in the future.   Have you planned for this to be communicated in the source releases?   as well as binaries created by Zephyr users?    
 2. Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the following disclaimer in the documentation and/or other materials provided with the distribution.
3. All advertising materials mentioning features or use of this software must display the following acknowledgement: This product includes software developed by the University of California, Berkeley and its contributors.
4. Neither the name of the University nor the names of its contributors may be used to endorse or promote products derived from this software without specific prior written permission.

For future reference cheat sheet of which licenses are OSI approved can be found https://spdx.org/licenses/preview/, and there is a programatic interface if you want to pull a "is license OSI approved" into your checking scripts for commits going forward.

stephanosio, 2022-03-24:
https://github.com/zephyrproject-rtos/zephyr/pull/39612#pullrequestreview-889510429
https://github.com/zephyrproject-rtos/zephyr/pull/39612#issuecomment-1047758863

@kestewart There were already some files that were licensed `BSD-4-Clause-UC` in the Zephyr repository, and we assumed this license was already approved or at least acceptable in some way; though, these were added in the the very early days of the project, possibly before such measures were in place: https://github.com/zephyrproject-rtos/zephyr/commit/bddd72602c44936c2f943068c178d77e369f76bc

How do you suggest we go about it? It affects both this new addition and the existing files that included since the Zephyr v1.0.0 release.

kestewart, 2022-03-24:
@stephanosio - it looks like those prior files did not get their licensing identified properly until last year, and I can't find traces of any reviews of this licensing by the board, so I think we've got a bigger issue here.   As you point out, that advertising clause is the problem.   At this point, I suggest reverting the recent merge, and let's either plan out the steps we'll need to do in release & documentation, or replacing the affected code.

carlescufi, 2022-06-28:
This is now done, see https://github.com/zephyrproject-rtos/zephyr/pull/45810

closed 2022-06-28 by carlescufi