zephyrproject-rtos/zephyr #44010: frdm_k64f: failed to run testcase samples/kernel/metairq_dispatch/

Zhaoningx, 2022-03-21:
To Reproduce

Failed to run test case samples/kernel/metairq_dispatch/, but it's weird, I want to look into why the test case is failed, I added some `printk ` to show the log, after doing this, the test case can be tested successfully. So I would like to report a bug.

Steps to reproduce the behavior:

```
west build -b frdm_k64f samples/kernel/metairq_dispatch/ -p always

west flash
```

see error

```
*** Booting Zephyr OS build zephyr-v3.0.0-1365-gbc72fb67b1c8  ***
I: Starting Thread0 at priority -2
I: Starting Thread1 at priority -1
II: M0 T1 mirq 4483 disp 7448 proc 28275 real 29560
I: M3 T1 mirq 4273 disp 403952 proc 23659 real 24700
I: M4 T1 mirq 4273 disp 911462 proc 54809 real 56056
I: MI: Thread 3 qERROR: Queue full, event dropped!
ueue fulERROR: Queue full, event dropped!
l, messaERROR: Queue full, event dropped!
ge 35 dropped
9 T1 mirq 4273 disp 1270240 proc 25308 real 25915
I: M10 T1 mirq 4273 disp 2592285 proc 18628 real 19840
I: M13 T1 mirq 4273 disp 3020566 proc 54308 real 55075
I: M14 T1 mirq 4273 disp 3556263 proc 55770 real 56290
I: M32 T1 mirq 4273 disp 3276999 proc 23507 real 24700
I: M36 T1 mirq 729122 disp 3674487 proc 49369 real 50215
I: M37 T1 mirq 682379 disp 4261804 proc 38369 real 39280
I: M11 T0 mirq 4273 disp 6183822 proc 44261 real 45355
I: M24 T0 mirq 4273 disp 6194083 proc 22030 real 22270
I: M28 T0 mirq 4273 disp 6553486 proc 11800 real 12550
I: M29 T0 mirq 4273 disp 7124728 proc 13993 real 14980
: Starting Thread2 at priority 0
I: M15 T2 mirq 4273 disp 8698025 proc 53732 real 55075
I: M16 T2 mirq 4273 disp 9245757 proc 47954 real 49000
I: M20 T2 mirq 4273 disp 9679200 proc 57232 real 57505
I: M21 T2 mirq 4273 disp 10252967 proc 3787 real 4045
I: M23 T2 mirq 4273 disp 10741816 proc 11436 real 12550
I: M30 T2 mirq 4273 disp 11023460 proc 6065 real 6475
I: M34 T2 mirq 4273 disp 11284293 proc 50637 real 51430
I: Starting Thread3 at priority 1
I: M1 T3 mirq 4273 disp 13979747 proc 43637 real 44140
I: M2 T3 mirq 4273 disp 14577059 proc 33802 real 34420
I: M5 T3 mirq 4273 disp 15019920 proc 17388 real 18625
I: M6 T3 mirq 4273 disp 15567182 proc 13010 real 13765
I: M7 T3 mirq 4273 disp 16090404 proc 56479 real 57505
I: M8 T3 mirq 4273 disp 16640176 proc 21164 real 22270
I: M12 T3 mirq 4273 disp 17023579 proc 137 real 400
I: M17 T3 mirq 4273 disp 17311579 proc 4945 real 5260
I: M18 T3 mirq 4273 disp 17860416 proc 31483 real 31990
I: M19 T3 mirq 4273 disp 18443557 proc 42387 real 42925
I: M22 T3 mirq 4273 disp 18869153 proc 12902 real 13765
I: M25 T3 mirq 4273 disp 19318809 proc 40079 real 40495
I: M26 T3 mirq 4273 disp 19898336 proc 50474 real 51430
I: M27 T3 mirq 4273 disp 20513318 proc 3192 real 4045
I: M31 T3 mirq 4273 disp 20846166 proc 17611 real 18625
I: M33 T3 mirq 4273 disp 21271800 proc 21444 real 22270
```

Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.13.2
Commit ID: bc72fb67b1c8

Zhaoningx, 2022-03-28:
Hi @dleach02, any comments here? Thanks.

mmahadevan108, 2022-04-08:
Below is the serial output with the latest code from main. I was using commit `e0274e2c54198b5e35f0d5ae997b8f0b5dd9099a`

```
*** Booting Zephyr OS build zephyr-v3.0.0-2521-ge0274e2c5419  ***
I: Starting Thread0 at priority -2
I: Starting Thread1 at priority -1
II: M0 T0 mirq 4423 disp 7383 proc 13695 real 14715
I: M3 T0 mirq 4218 disp 376448 proc 54552 real 55238
I: M12 T0 mirq 4213 disp 493857 proc 11260 real 12325
I: M13 T0 mirq 4218 disp 969215 proc 766 real 1570
I: M14 T0 mirq 4218 disp 1437862 proc 50076 real 50565
I: M15 T0 mirq 4218 disp 2029404 proc 43958 real 44590
I: M17 T0 mirq 4218 disp 2542668 proc 43752 real 44590
I: M26 T0 mirq 4218 disp 2719961 proc 24287 real 25470
I: M27 T0 mirq 4218 disp 3262473 proc 7895 real 8740
I: M28 T0 mirq 4218 disp 3752895 proc 38244 real 38615
I: M33 T0 mirq 4218 disp 4079930 proc 25535 real 26665
I: M34 T0 mirq 4218 disp 4575497 proc 14529 real 14715
I: M37 T0 mirq 4228 disp 5098733 proc 40581 real 41005
I: M40 T0 mirq 3109 disp 5560894 proc 14914 real 15910
I: M8 T1 mirq 4218 disp 7705210 proc 38785 real 39810
I: M9 T1 mirq 4218 disp 8227857 proc 6908 real 7545
I: M10 T1 mirq 4228 disp 8743854 proc 44347 real 44590
I: M11 T1 mirq 4218 disp 9341131 proc 25774 real 26665
I: M20 T1 mirq 4218 disp 9428704 proc 34009 real 35030
I: M24 T1 mirq 4228 disp 9848602 proc 53593 real 54150
I: M29 T1 mirq 4218 disp 10215257 proc 13936 real 14715
I: M30 T1 mirq 4218 disp 10772924 proc 18164 real 19495
I: M36 T1 mirq 4223 disp 11055621 proc 5897 real 6350
: Starting Thread2 at priority 0
I: M2 T2 mirq 4218 disp 13782518 proc 3675 real 3960
I: M5 T2 mirq 4218 disp 14116969 proc 6677 real 7545
I: M7 T2 mirq 4218 disp 14559373 proc 40685 real 41005
I: M16 T2 mirq 4218 disp 14733516 proc 30989 real 31445
I: M18 T2 mirq 4218 disp 15219680 proc 41589 real 42200
I: M21 T2 mirq 4218 disp 15657291 proc 12082 real 12325
I: M22 T2 mirq 4218 disp 16190943 proc 13184 real 13520
I: M23 T2 mirq 4218 disp 16760605 proc 49021 real 49370
I: M35 T2 mirq 4228 disp 16761504 proc 1811 real 2765
I: M38 T2 mirq 4218 disp 17298269 proc 33884 real 35030
I: Starting Thread3 at priority 1
I: M1 T3 mirq 4218 disp 20181070 proc 12512 real 13520
I: M4 T3 mirq 4218 disp 20536351 proc 9500 real 9935
I: M6 T3 mirq 4218 disp 20966755 proc 56888 real 57735
I: M19 T3 mirq 4218 disp 20905561 proc 35598 real 36225
I: M25 T3 mirq 4218 disp 21264923 proc 52159 real 52955
I: M31 T3 mirq 4218 disp 21580950 proc 1679 real 2765
I: M32 T3 mirq 4218 disp 22093787 proc 55513 real 56540
I: M39 T3 mirq 4218 disp 22401976 proc 20188 real 20690
I:         ---------- Latency (cyc) ----------
I:             Best    Worst     Mean    Stdev
I: MetaIRQ     4213     4423     4224       32
I: Thread0     7383  5098733  2411256  1619504
I: Thread1  7705210 11055621  9482129  1059991
I: Thread2 13782518 17298269 15508067  1154519
I: Thread3 20181070 22401976 21241422   707014
I: MetaIRQ Test Complete
```

dleach02, 2022-04-12:
I am unable to reproduce your failure.

Will close this issue. @Zhaoningx if you determine a reproduceable method to show this bug please reopen this ticket with details.

closed 2022-04-12 by dleach02