zephyrproject-rtos/zephyr #44013: Bluetooth: ISO: ISO over USB

Thalley, 2022-03-21:
**Is your enhancement proposal related to a problem? Please describe.**
Currently it is not possible to transmit or receive ISO data using e.g. the `native_posix` board on a board running the `hci_usb` or `hci_usb_h4` samples. 

**Describe the solution you'd like**
Make it possible to send and receive ISO data over USB using the `hci_usb` or `hci_usb_h4` samples. 

**Describe alternatives you've considered**
N/A

**Additional context**

ISO USB from the core spec (version 5.3, vol 4, part B, 2.5: 

> 2.5 ISOCHRONOUS ENDPOINTS EXPECTATIONS
> These isochronous endpoints transfer synchronous data to and from the Controller of the radio.
> 
> Time is the critical aspect for this type of data. The USB firmware should transfer the contents of the data to the Controllers' synchronous FIFOs. If the FIFOs are full, the data should be overwritten with new data. 
> 
> These endpoints have a one (1) ms interval, as required by Chapter 9 of the USB Specification, Versions 1.0 and 1.1.
> 
> The radio is capable of three (3) 64 kb/s voice channels (and can receive the data coded in different ways – 16-bit linear audio coding is the method that requires the most data). A suggested max packet size for this endpoint would be at least 64 bytes. (It is recommended that max packet sizes be on power of 2 boundaries for optimum throughput.) However, if it is not necessary to support three voice channels with 16-bit coding, 32 bytes could also be considered an acceptable max packet size.

Casper-Bonde-Bose, 2022-03-21:
@Thalley For USB operation `native_posix` relies on BlueZ to load the USB driver and forward the ISO data channel to the socket interface that Zephyr uses. Current Linux kernel do not have support for this - hence one would need to wait for support @Vudentz might be able to comment as he knows more about BlueZ than I do.
A workaround could be to have the `hci_usb` sample application use the ACL channel for ISO data and implement multiplexing in the `native-posix `zephyr build based on the connection handle - which will be different for an ACL and the ISO. Such a solution will not require BlueZ support for ISO data transport.
I have only ben using the `hci_uart` sample application.

Thalley, 2022-03-21:
@jhedberg @Vudentz If you have any input on this, please share :D

Thalley, 2024-02-09:
This is still valid