zephyrproject-rtos/zephyr #44062: Need a way to deal with stack size needed when running coverage report.

enjiamai, 2022-03-22:
**Describe the bug**
This is an extension of [#41388](https://github.com/zephyrproject-rtos/zephyr/issues/41388), when we run the code coverage report, some of the tests will fail due to not have sufficient stack size.
(Ex. tests/net, it needs more stack size like that to pass the test: 
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_NET_TCP_WORKQ_STACK_SIZE=2048
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096

Just enlarge the stack size or extra coverage.conf to decide stack size needed seems not a good solution for this issue.
So a better way is needed to make it won't need to add extra stack sizes manually when running with coverage report generated.

The point of this requirement might be: We don't need to add extra stack size case by case, and boards.

Zhaoningx, 2022-05-16:
Hi, I'm working with @enjiamai, we did some analysis together, and found that this is because the memory applied by some platforms for coverage was overwritten by the memory of the prj.conf (if configured), resulting in memory overflow. After discussion, the memory size shouldn't only decide by the test case stack if the coverage feature is enabled. We are working on how to apply extra memory for coverage whatever the platform is.

Zhaoningx, 2022-05-25:
Hi, when coverage is enabled, GCC will insert the necessary code in the source code to count the operations, which causes the usage of each stack to increase. If configuring the stack size manually by the user, the stack size will be different from other test cases, so if using the idea above to add a fixed size `COVERAGE_STACK_SIZE` value to increase stacks, it may cause stacks to be larger than our expected, maybe resulting in stack overflow or a CPU exception. So far I think it's hard to find a uniform way to increase the stack size so that all cases fit. Need discuss with @enjiamai for the next step.

I triaged the part of the tests/net, and I find that the failure case can be tested successfully by adjusting the value of some macros manually. The main affected macros are as follows for all net cases:
```
CONFIG_NET_TX_STACK_SIZE
CONFIG_NET_RX_STACK_SIZE
CONFIG_MAIN_STACK_SIZE
ZTEST_STACK_SIZE
```

Zhaoningx, 2022-08-01:
Hi, sorry for the long time no update, the main issue about this bug is the stack usage is different from the test case. So my first solution is to create a new macro `COVERAGE_STACK_SIZE` to increase the stack that is needed when coverage is enabled.

Let me use the main stack as an example:
**defined**
```
config COVERAGE_STACK_SIZE
	  int "extra size of stack for coverage"
	  default 8192 if COVERAGE_GCOV
	  default 0
	  help
		When the macro COVERAGE_GCOV is enabled, the test case needs more
		memory to mark the testing code and store the info in the BSS
		segment.
```
**arch/arm/core/aarch32/cortex_m/reset.S**
`    ldr r0, =z_main_stack + CONFIG_MAIN_STACK_SIZE + CONFIG_COVERAGE_STACK_SIZE`

**kernel/init.c**
`   K_THREAD_PINNED_STACK_DEFINE(z_main_stack, ADJUST_TEST_STACK_SIZE(CONFIG_MAIN_STACK_SIZE));`

when I enabled the coverage, I'll add the macro COVERAGE_STACK_SIZE to the stack used, and when coverage is disabled, the CONFIG_COVERAGE_STACK_SIZE is 0 by default.