zephyrproject-rtos/zephyr #44084: build: multiple unused static function

ycsin, 2022-03-22:
**Describe the bug**
Some unused static functions (dead code) exist in:
- `shed.c` - unnecessarily compiles `is_aborting` & `set_current` for some configurations
- `mpsc_pbuf.c` - `idx_dec`
- `ring_buffer.c` - `wrap`

cfriedt, 2022-03-23:
@ycsin Is there a warning generated? Can you include any additional info in the description?

ycsin, 2022-03-23:
> @ycsin Is there a warning generated? Can you include any additional info in the description?

I just learned that no warning will be generated if they are `inline`

ycsin, 2022-03-23:
> @ycsin - we have some fields that are typically filled out with bug reports, so I just wanted to capture that information here.

I had a look at the bug report fields again, but I'm not sure what to write. I think this is more like something that we know should be fixed, but doesnt cause any issue if we don't. Probably it doesn't worth a backport?

closed 2022-03-30 by ycsin