zephyrproject-rtos/zephyr #44092: rand32_ctr_drbg fails to call the respective initialization routing

dleach02, 2022-03-22:
**Describe the bug**
f44b3dc4dfe changed where initialization of entropy_dev to occur at compile time but the logic in z_impl_sys_csrand_get() calls the initialization routine IF entropy_dev is NULL. In the init routine it will assign the device object to entropy_dev and properly initialize the context object. 

Using an initialized context object results in a crash.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mimxrt1050_evk -p always tests/net/socket/tls/
2. See error

**Impact**
Code crashes from an indirect call to a NULL address when getting sys_csrand_get() data.

dleach02, 2022-03-22:
ping @gmarull

gmarull, 2022-03-22:
Thanks for reporting @dleach02 , will submit a fix tomorrow

dleach02, 2022-03-22:
I have a fix... posting now

PR #44093 (2022-03-23): random: rand32_ctr_drbg: fix bug preventing initialization
Bug introduced that prevented the logic from initializing the
context structure. Without this initialization, the application
will crash on repeated request for ctr_drbg random data.

Fixes: #44092
Fixes: #43792
Fixes: #43841
Fixes: #43842
commit a282e930 (2022-03-23): random: rand32_ctr_drbg: fix bug preventing initialization

Bug introduced that prevented the logic from initializing the
context structure. Without this initialization, the application
will crash on repeated request for ctr_drbg random data.

Fixes: #44092
closed 2022-03-23 by dleach02