zephyrproject-rtos/zephyr #44108: CONFIG_ZTEST_NEW_API=y broken with CONFIG_TEST_USERSPACE=y

henrikbrixandersen, 2022-03-23:
**Describe the bug**
The new ztest API (`CONFIG_ZTEST_NEW_API=y`) is broken when used in combination with `CONFIG_TEST_USERSPACE=y`. There are no in-tree tests of this combination.

One some platforms (e.g. `qemu_cortex_m3`) the test added below fails the assert, indicating the test case is not running in user space after all. On other platforms (e.g. `frdm_k64f`), the test below causes a bus fault.

**To Reproduce**
Steps to reproduce the behavior:

Apply the following patch to create a test case:

```
diff --git a/tests/ztest/base/prj_verbose_0.conf b/tests/ztest/base/prj_verbose_0.conf
index b4903cd10c..7233d69b58 100644
--- a/tests/ztest/base/prj_verbose_0.conf
+++ b/tests/ztest/base/prj_verbose_0.conf
@@ -1,3 +1,4 @@
 CONFIG_ZTEST=y
 CONFIG_ZTEST_NEW_API=y
 CONFIG_ZTEST_ASSERT_VERBOSE=0
+CONFIG_TEST_USERSPACE=y
diff --git a/tests/ztest/base/src/main.c b/tests/ztest/base/src/main.c
index 07fc9eb474..21d7c0e14d 100644
--- a/tests/ztest/base/src/main.c
+++ b/tests/ztest/base/src/main.c
@@ -35,6 +35,11 @@ ZTEST(framework_tests, test_assert_mem_equal)
        zassert_mem_equal(actual, expected, sizeof(expected), NULL);
 }
 
+ZTEST_USER(framework_tests, test_userspace_is_user)
+{
+       zassert_true(k_is_user_context(), NULL);
+}
+
 /***************************************************************************************************
  * Sample fixture tests
  **************************************************************************************************/
```

1. `west build -b qemu_cortex_m3 tests/ztest/base -- -DCONF_FILE=prj_verbose_0.conf`
2. See error:

```
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.0.0-1639-ga4d438444771  ***
Running test suite fixture_tests
===================================================================
START - test_fixture_pointer
 PASS - test_fixture_pointer in 0.1 seconds
===================================================================
Test suite fixture_tests succeeded
Running test suite framework_tests
===================================================================
START - test_assert_mem_equal
 PASS - test_assert_mem_equal in 0.1 seconds
===================================================================
START - test_assert_tests
 PASS - test_assert_tests in 0.1 seconds
===================================================================
START - test_empty_test
 PASS - test_empty_test in 0.1 seconds
===================================================================
START - test_userspace_is_user

    Assertion failed at WEST_TOPDIR/zephyr/tests/ztest/base/src/main.c:40
 FAIL - test_userspace_is_user in 0.1 seconds
===================================================================
Test suite framework_tests failed.
Running test suite rules_tests
===================================================================
START - test_rules_before_after
 PASS - test_rules_before_after in 0.1 seconds
===================================================================
Test suite rules_tests succeeded
===================================================================
PROJECT EXECUTION FAILED
```

1. `west build -t flash -b frdm_k64f tests/ztest/base -- -DCONF_FILE=prj_verbose_0.conf`
2. See error:3. 

```
*** Booting Zephyr OS build zephyr-v3.0.0-1639-ga4d438444771  ***
Running test suite fixture_tests
===================================================================
START - test_fixture_pointer
 PASS - test_fixture_pointer in 0.1 seconds
===================================================================
Test suite fixture_tests succeeded
Running test suite framework_tests
===================================================================
START - test_assert_mem_equal
 PASS - test_assert_mem_equal in 0.1 seconds
===================================================================
START - test_assert_tests
 PASS - test_assert_tests in 0.1 seconds
===================================================================
START - test_empty_test
 PASS - test_empty_test in 0.1 seconds
===================================================================
START - test_userspace_is_user
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x20001f6c
E:   NXP MPU error, port 3
E:     Mode: User, Data Address: 0x20001f6c
E:     Type: Read, Master: 0, Regions: 0x8100
E: r0/a1:  0x20001f64  r1/a2:  0x20001fec  r2/a3:  0x00000000
E: r3/a4:  0x00000001 r12/ip:  0x0000ae43 r14/lr:  0x0000ae4f
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x00002c76
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000020 (test_userspace_is_user)
E: Halting system
```

```
$ addr2line -e build/zephyr/zephyr.elf 0x00002c76
zephyr/subsys/testsuite/ztest/src/ztest_new.c:403
```

**Expected behavior**
`CONFIG_ZTEST_NEW_API=y` should work with `CONFIG_TEST_USERSPACE=y`.

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 20.04 LTS
 - Toolchain: Zephyr SDK v0.13.2
 - Commit SHA: a4d4384447710e9885c0248385db0d5a5d197d50

stephanosio, 2022-03-23:
cc @yperess

yperess, 2022-03-23:
@henrikbrixandersen The issue is that `config TEST_USERSPACE` in `subsys/testsuite/Kconfig` does have the line `depends on USERSPACE`. The definition of `ZTEST_USER` in `ztest_test_new.h` switches off of `CONFIG_USERSPACE` to see if userspace exists.

I might even argue that `CONFIG_TEST_USERSPACE` shouldn't even exist?

Possible solutions for you:
1. Add `CONFIG_USERSPACE=y`
2. Same as 1, but also add the dependency to `TEST_USERSPACE`
3. Get rid of `TEST_USERSPACE`

I would opt for 3, because I don't see the benefit of having a test specific config for userspace.

henrikbrixandersen, 2022-03-24:
`CONFIG_TEST_USERSPACE=y` is a "soft" version of `CONFIG_USERSPACE=y` which only enables userspace if the architecture/SoC supports it, allowing tests to run in userspace where possible but not preventing the same test to run on a platform without userspace support.

Having to force `CONFIG_USERSPACE=y` in order to use `CONFIG_ZTEST_NEW_API=y` is a step backwards compared to our older ztest framework.

yperess, 2022-03-25:
Gotcha, it's easy to swap it over. I wasn't aware of the "soft" version.

yperess, 2022-05-19:
The fix that closed this was reverted because it caused failures on the nightly build. There's a new PR https://github.com/zephyrproject-rtos/zephyr/pull/45262 for review to fix this.

PR #44562 (2022-04-26): ztest: Update userspace tests in new API
Update the new API to use K_USER as the flags for both
CONFIG_USERSPACE and CONFIG_TEST_USERSPACE.

Fixes #44108
commit 900b6723 (2022-04-26): ztest: Fix userspace ztests in new API

Update the new API to use K_USER as the flags for both
CONFIG_USERSPACE and CONFIG_TEST_USERSPACE. Also, fix the linker
script to properly include the suites, tests, and rules.

Fixes #44108
PR #45262 (2022-05-25): Ztest userspace
A second attempt at getting userspace to work with ztest's new API. It looks like the addition of the sections to the linker was still breaking the MMU/MPU. This attempt uses the entry point as a `__syscall` which should work around that issue.

Fixes #44108
commit 86cadf92 (2022-05-25): ztest: Fix userspace ztests in new API

Update the new API to use K_USER as the flags for both
CONFIG_USERSPACE and CONFIG_TEST_USERSPACE. Also, fix the linker
script to properly include the suites, tests, and rules.

Fixes #44108
closed 2022-05-25 by stephanosio