zephyrproject-rtos/zephyr #44113: LE Audio: Remove specific size for bt_audio_base et.al.

Thalley, 2022-03-23:
**Is your enhancement proposal related to a problem? Please describe.**
The following structs have a specified size controlled by Kconfigs. 

```c
struct bt_audio_base
struct bt_audio_base_subgroup
struct bt_audio_base_bis_data
struct bt_codec_data
struct bt_codec
```

Which makes sense when setting the values and storing them in the stack. 

However, for many use cases, e.g. when parsing a BASE from a broadcast source, we do not need to store them and thus the specific size can easily become a limitation (not to mention a significant stack allocation). 

**Describe the solution you'd like**
In the above structs, we should be able to set the size of the arrays to `0` and simply cast the buffer (on receival) to the correct struct. That way we can still parse them, but without requiring a memory copy or being limited to a specified size. 

For this to work, we need to find a different way of storing the data in the stack, as the structs will not longer be able to fit the data stored there. Alternatively we can avoid storing the data in the stack, and simply retrieve it from the application directly, and let the application deal with storing them. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Casper-Bonde-Bose, 2022-03-23:
This is similar to #42860 for meta-data and codec-config data which uses the same storage type.
The overall problem with having a fixed size for LTV formatted data is that there is no requirement in the specification to place the most important data first in the payload - hence a zephyr based product could end up not working in the future when additional LTV values are added and not all will fit into the fix-ed size storage. The application layer should parse and extract what is needed - allowing it to ignore the rest.
The alternative would be to only parse and store what is currently known and be sure all sizes are defined to accommodate the maximum size - maybe truncating the dynamic length UTF-8 types.

Thalley, 2023-10-11:
Cannot store them with [0] as array sizes due to limiation in C.
Alternatives:
1) Make role-agnostic Kconfigs for them (e.g. BT_BAP_BASE_SUBGROUP_COUNT)
	Pros
		Simple
		Similar to existing 
	Cons
		Still high memory usage and partially lack of supporting all BASEs
2) Treat the BASE as an opaque structure and implement getter functions, e.g. `bt_bap_base_get_subgroup(base, index)` and `bt_bap_base_get_bis(base, index)` which will just traverse the advertising data instead of storing it (at all). 
	Pros
		Supports all BASES regardless of what we can store
		No memory usage
		Cleans up some Kconfigs
	Cons
		Less intuitive to use (for many)
		Still need to store some data in the broadcast sink (e.g. BIS indexes per subgroup for the receive states)
		May need to traverse the advertising several times => more processing in the RX thread
			Could make a copy and provide it via the workqueue => Requires 255 octets of memory