zephyrproject-rtos/zephyr #44145: Zephyr Panic dump garbled on Intel cAVS platforms

kv2019i, 2022-03-23:
**Describe the bug**
When Zephyr hits a fatal error, the OS panic dump is not printed out correctly in many circumstances. This is happening regularly with multicore configurations and SOF application, in bugs like https://github.com/thesofproject/sof/issues/5556

**To Reproduce**
Aside https://github.com/thesofproject/sof/issues/5556 , no simple steps yet available. A simple panic test with a adsp smoke test is ok (output fine), so this does not happen in all cases.

**Expected behavior**
The panic dump is readable/parseable.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
A few example dumps (from use-case of above bug)

EXAMPLE1
```
--[00000000] <inf> main: SOF on intel_adsp_cavs25
[00000000] <inf> main: SOF initialized
555357: SHM: FW ABI 0x3015000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-593-gb7aee6182a91-dirty zephyr:zephyr-v3.0.0-1016-g43f2c22e5efb src hash 0x61b977eb (ldc hash 0x61b977eb)
X000116[473]0 0<11e6r4r7>3 ] o<s:e rr >* * oFsA:T A L* *E XCCPEUP TI0O NE
 CCAUSE [9 0(01l1o6a4d7/3s]t o<ree rarl>i gnomse:n t )**
sxb[e020604181f6 4V7A3D]D R< er0rx>1 31o
::  ** [ PS0 01106x467036]2 0<rrr> [os:0 0 1*1*6 4 7 3 ]( IN<TLEVeErLr:> o0s :E XCM :* *  A0 0 UM:1 RI0NxGb:e02fe907  W OSEP: 1 OWB:0x6b eC0A6LbL6I6N0C : A2 2)
1x9e07[b7000
1] <e[rr>0 01o1s6:4 7 3*]*  < Ae4r r> s9:6 d4   *A*5   A8 0x100x1b e 0A297 428  A6 00xx3b e 0A67b 630 A100x 15
7 0x[9e0070b191060
0err>[ o0s0:1 1 6*4*7 3A]1 2 <e0rxr7>  A13 os:  ** LBE0Gx be0725c0 A14 0xbe01d080 LE0NxDb e072618 A15 0xb(en0i1ld)09
 9
A[00101061417634]7 3<] e<rre>r r>o s:o s :* * S>A>R>  ZEPHY0Rx 1FaAT
s0: CP[U e0x0c1e1p6t4i7o3n]  o<n eCrPrU>  0o
s: Current[ th0r0e1a1d6:4 74]0 x<9eer0r7>1 b8o8s :( uHnaklntoiwnng)
yystem
```

EXAMPLE2
```
�[00000000] <inf> main: SOF on intel_adsp_cavs25
[00000000] <inf> main: SOF initialized
584633: SHM: FW ABI 0x3015000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-586-g8e85d0018c7e-dirty zephyr:zephyr-v3.0.0-1014-gfa7f5ad3686c src hash 0xef16be95 (ldc hash 0xef16be95)
[0[000039933775544]]  <<eerrrr>>  ooss::    ****  CFPAUT AL0  EEXXCCECPATUISOEN 
113 (lo[ad/0s0t3o9r3e7 5P4I]F  <daetrar >e rroosr:)
P **  [PC 00309x3b7e5042]6 4<b6e rVrA>D DRo s:0 x 3*0*  
0x606[20
b03937[54]0 0<3e9r3r7>5 4]o s<: r*r*>     (IoNsT:L EVE L*:*  A0 0 EXCM: 00 xUbMe:02feaf1   RSIPN G:0 WOE:0xbe016 bO6W6B0:  A2 6 CALLINC:0x92e)07

[[000309339735745]4 ]< e<rre>r r>o s:o s :* *  * *A 4  A8 (nil0)x 2 1Aa98 c  A5 0xbe060bx693e00 7Ab160d 4  A6 0x30 x 9Ae70 7b900 0Ax1115 
*  * *A 4  A8 (nil0)x 2 1Aa98 c  A5 0xbe060bx693e00 7Ab160d 4  A6 0x30 x 9Ae70 7b900 0Ax1115 
0x9e07b900
003[93705043]9 3<75e4r]r >< orsr:>   **o sA:1 2  ** 0LxB7E GA 13 0xb0ex0b1ed0078205 cL0E NAD1 4 0x0bxeb0e10d7029601 8L CAO1U5N T (nil)
orsr:>   **o sA:1 2  ** 0LxB7E GA 13 0xb0ex0b1ed0078205 cL0E NAD1 4 0x0bxeb0e10d7029601 8L CAO1U5N T (nil)
0xbe0264b0
H039[375040]3 9<37e5r4r]>  <oesr:r >  ** SAoRs : >>0>x 1ZaEP
uROR 0[: 0C0P3U9 3e7x5c4e]p t<ioenr ro>n  CoPsU:  0C
grrent th[ead0:0 39307x595e]0 7<1be8r8r >( uonsk:n oHwanl)tin
```

EXAMPLE3
```
[00000000] <inf> main: SOF on intel_adsp_cavs25
[00000000] <inf> main: SOF initialized
547963: SHM: FW ABI 0x3015000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-586-g8e85d0018c7e-dirty zephyr:zephyr-v3.0.0-1014-gfa7f5ad3686c src hash 0xfc2445c4 (ldc hash 0xfc2445c4)
[0011733[80] 0<01e7r3r3>8 0]o s<: er r*>*  FoAsT:A L  E*X*C ECPPTUI 0 EXCOCNU
S
0E 28 (load prohibi[ted)0
D17338[0] <01733e8r0r]>  <oesr:r >  **o s :P C  **  PS 0xbe003xb6000eb2 0VA
0fff[f80
1173380][ <0e0r1r7>3 38o0s]:  < e*r*r >  @  o(sI:N  TL E*V*E L :/Am0 0E nXtC/M:0 sxb0e 0U2M3:ef1f 2R I NSGP:a g04 W0OxEb:0e10 6ObW8Bf:00 1 4A 2C 0AcLL0IxN9Ce:_027)22/
 chao0 [m eA0/3k80 v] 0<ehxe6rr>
ma n*e*/ p A[r4/ f033-zephyrsof/sof/zep001x7f3f3f8f0f]f 00<  Ae5r r> 0xocs :  A6  ** ( nAi8l )  A7 0xbe01d02x6f4f f fAf9f ff
hyr0/x.[.be00061b7933308 0A]1 1</src/incl err> os: 0x 9*e*0 7A21d24 0
u dAe1/4so 0[fx/1 0A0s151 c7h3308x09]e e07<d23ecr0url
>>e/ toa[ss:0 k0.17 3*3*8 0L]B E<h:94
Ge rr> os: 0 x*b*e 0S3AbR0 08 LEN0Dx 1e
3b016 [COU0N0T1 733800x]b e<02e3rfrf>0
0os: >>[> Z0E0P1H7Y3R3 8F0A]T A<L eErRrR>O R os0::  CurrCePnUt  etxhcreepatdi:o n on CPU 00x9e
7n)
```

**Environment (please complete the following information):**
 - OS: Zephyr, cAVS2.5 hardware
 - Toolchain: Zephyr SDK 0.13.1
 - Zephyr main commit b4dfdb5afb5e4e16d9207e79b0b542ee9d128067

**Additional context**
Workarounds tried:
 - CONFIG_PRINTK_SYNC=y

FYI @nashif @teburd @andyross

kv2019i, 2022-03-23:
@teburd I tried a simple forced divide-by-zero and when I put it in the intel_adsp smoketest, the panic was still ok. I'll try to play with more options later.

teburd, 2022-03-23:
I was thinking about trying that exact same thing... what happens if you try causing some chaos in an ISR with a divide by zero or write out of valid memory?

teburd, 2022-03-29:
@kv2019i it seems maybe there was a coherency issue with semaphores looking at the SOF thread, is this still an issue?

https://github.com/thesofproject/sof/pull/5588

kv2019i, 2022-03-30:
@teburd The garbled dumps seem to be still happening with https://github.com/thesofproject/sof/pull/5588 merged, so it's not that.

kv2019i, 2022-03-30:
One way to force a panic that causes garbled output: https://github.com/kv2019i/sof/commit/6b177c3c6880fd8ed62a8434f04c1d889c78b4f5

teburd, 2022-04-05:
This is caused by a combination of printk and LOG_ calls, where the LOG_ calls update the underlying memory window buffer (winstream) 1 character at a time and has a different synchronization mechanism than printk.

I have some work to actually solve the problem, but I believe I've found the issue

teburd, 2022-04-05:
I've got a minimal reproducer based on hello_world, from everything I can tell its caused by a printk and LOG racing against eachother on the underlying memory window buffer.

https://github.com/teburd/zephyr/tree/intel_adsp_fatal_corrupt

Try running

samples/boards/intel_adsp/fatal_error

``` sh
west build -p always -b intel_adsp_cavs25 samples/boards/intel_adsp/fatal_error
./soc/xtensa/intel_adsp/tools/cavstwist.sh build/zephyr/zephyr.elf

Hello World! intel_adsp_cavs25
000000000] <err> os:  ** FATAL EXCEPPTI[ON0
0000000[] 0<00e0r0r0>0 0]o s<: e r*r*>  CPoUs : 0  *E*X C CPACU SE 9 (0xbleo0a1d0/7s7t9o rVeA DaDlRi gnment)
0  *E*X C CPACU SE 9 (0xbleo0a1d0/7s7t9o rVeA DaDlRi gnment)
0xbe0189a1
L0000[00000]0 0<00e0r0r]>  <oesr:r >  **o s :P S  ** 0 x 6 0(cI2N0
 0 E[XC0M0:0 0000 0U0M]: <1 eRrIrN>G :o0s :W OE :**1   OAW0B :102x bCeA0L1L3I1N4C0:  2S)P
xx9e[010b000f090 0 0A02]  <e0rxrb>e 01o8s9:a 1  * *A 3  A0 x770
bbe016[89000 0 0A050 000]x 1<00e r rA>6  o0sx:f f  * *A 7  A8 0(xnbiel0)1 7 6A890
6xbe[01080905000 0A0100]  <e0rxrb>e 01o6s2:b b  *A*1 1A 12( nil0)
cA13 [0000x0b0e000107]6 3<8 eAr1r4>  os0:x b e*0*1 7L6B5E0G  A15 0x(bnei0l1)
A39 LEN[D 000000x0b0e00]1 2<c4efr rL>C OUoNsT:  ( n*i*l )
R0x1d
 000000[000]0 0<00e0r0r0>]  <ose:r r>>> > oZsE:P HCYuRr rFeAnTtA Lt hErReRaOdR:  0: 0xC9PeU0 1ebx6c4e8p t(iounn konno wCnP)U

0
[00000001] <err> os: Halting system
```

PR #44563 (2022-04-13): intel_adsp: Fix corrupt fatal logs
Adds a lock around the logger backend for adsp and changes the buffer size to 80 to avoid 1 character a time writes to the trace buffer

fixes #44145
closed 2022-04-13 by nashif