zephyrproject-rtos/zephyr #44192: esp32 flash custom partition table

huarunlin, 2022-03-25:
**Is your feature request related to a problem? Please describe.**
When I implemented the ota function, I found that when zephyr uses the esp-idf bootloader, the flash partition table uses the default "Single factory app, no OTA", and zephyr does not seem to provide a way to modify it.
 
**Describe the solution you'd like**
I hope to configure the flash partition table through cmake configure.

sylvioalves, 2022-03-26:
Can you describe which OTA function you implemented?  
In Zephyr, to work with OTA feature you have to use MCUBOOT instead of default IDF_BOOTLOADER.

huarunlin, 2022-03-29:
> Can you describe which OTA function you implemented? In Zephyr, to work with OTA feature you have to use MCUBOOT instead of default IDF_BOOTLOADER.

I'm going to use the ESP-IDF firmware update API because I can't find any data or examples of MCUBOOT updates in the application.

rftafas, 2022-03-29:
@huarunlin Have in mind that extended support to IDFboot on Zephyr is not on roadmap, which means Secureboot/FlashEncryption won't be available. Same applies for partition customization. Chances are that we will not have an ESP32S3 IDFboot option for Zephyr, provided we already have it working on MCUboot.

Although we don't have that in plans, it is feasible to do it. As for using ESP-IDF update APIs, as long as you understand the limitations above, should be ok. It could be difficult to port them, as they probably depend on FreeRTOS.

As for MCUboot usage, maybe you can start with this: https://docs.zephyrproject.org/latest/guides/device_mgmt/ota.html

We did some tests and they seem to work well.

closed 2022-04-14 by sylvioalves