zephyrproject-rtos/zephyr #44196: Unify C++ source and header file extensions

stephanosio, 2022-03-25:
## Overview

Currently, we have different file extensions for the C++ source and header files in the Zephyr repository:

**Source Files**

* [`cpp`](https://github.com/zephyrproject-rtos/zephyr/search?q=extension%3Acpp)
* [`cxx`](https://github.com/zephyrproject-rtos/zephyr/search?q=extension%3Acxx)
* [`cc`](https://github.com/zephyrproject-rtos/zephyr/search?q=extension%3Acc)

**Header Files**

* [`h`](https://github.com/zephyrproject-rtos/zephyr/tree/bdae9399be532f9fd04770597b20bca37e784c37/samples/modules/tflite-micro/magic_wand/src)
* [`hh`](https://github.com/zephyrproject-rtos/zephyr/pull/41735/files)

For the sake of consistency, the project should decide on and use only single extension each for the C++ source and header files.

## Context

### Widely-used conventions

**C++ source Files:** `.cpp`, `.cxx`, `.cc`, `C`
**C++ header Files:** `.hpp`, `.hxx`, `.hh`, `h`

### From the perspective of the Standard C++ Foundation

* [What source-file-name convention is best? foo.cpp? foo.C? foo.cc?](https://isocpp.org/wiki/faq/coding-standards#src-file-ext)
* [What header-file-name convention is best? foo.H? foo.hh? foo.hpp?](https://isocpp.org/wiki/faq/coding-standards#hdr-file-ext)

## Opinions

Amongst the "widely-used conventions" listed above, some to avoid are:

* `C` (capital C) for source files
    * This will not play well with the operating systems that do not strictly enforce the case sensitiveness of the file names (e.g. Windows, macOS).
    * This is more prone to human errors than having alphabetically different names for the C++ source files and the C source files.
* `h` for C++ header files
    * This provides no distinction between the C++ header files and the C header files.

## Proposed Change

TBD (to be discussed and finalised in the relevant meetings)

stephanosio, 2022-03-25:
Please feel free to update this issue with more contexts and opinions.

stephanosio, 2022-03-25:
cc @tejlmand @yperess @carlescufi @nashif @MaureenHelm @alexanderwachter

tknapstad, 2022-03-25:
I would suggest the `.hpp`/`.cpp` standard adopted by the Boost libraries. Gcc and Clang will automatically recognize all alternatives, but at least the Arm Keil compiler requires an additional command line switch to recognize the source file as C++ code unless the filename extension is `.cpp` (see https://www.keil.com/support/man/docs/armcc/armcc_chr1359124195123.htm). I don't know if IAR has similar quirks (or if Zephyr supports/wants to support these toolchains) but I think that `.hpp`/`.cpp` are the most widely adopted C++ filename extensions.

cfriedt, 2022-03-25:
`.cpp` / `.hpp` is fine with me, although the fact is, 3rd party code will likely be named `.cpp` / `.h` and we will need to deal with that as well (CMake handles this just fine).

stephanosio, 2022-03-25:
I also prefer `.hpp`/`.cpp`.

By the way, the scope of this is the files that are being committed to the main Zephyr repository (i.e. zephyrproject-rtos/zephyr) only. Third-party code in the modules can keep whatever convention their upstream repo uses.

alexanderwachter, 2022-03-25:
+1 for .cpp .hpp

carlescufi, 2022-03-25:
Another vote for `.hpp`/`.cpp` for files in the main zephyr tree, as mentioned by @stephanosio.

ppryga, 2022-03-31:
In my experience `.hpp`/`.cpp` are most widely used. Let's use these.

mkschreder, 2022-03-31:
I also vote for .hpp / .cpp. 

Make sure that you also add a compliance check that fails all the others so we don't forget we made this decision.