zephyrproject-rtos/zephyr #44199: (U)INT{32,64}_C macro constants do not match the Zephyr stdint types

stephanosio, 2022-03-25:
**Describe the bug**
The Zephyr RTOS has its own C standard integer type definitions ([`zephyr_stdint.h`](https://github.com/zephyrproject-rtos/zephyr/blob/0521aa0fbce835022fbd65d0d438f4a20d5af893/include/toolchain/zephyr_stdint.h#L11-L16)) that override the toolchain defaults.

The `__INT{32,64}_C` internal constant macros, which are defined by the toolchain, are not overridden by the `zephyr_stdint.h` of the toolchain abstraction layer, and this causes the C library `stdint.h` header to infer incorrect types (i.e. those that are native to the toolchain rather than those that are specified by the toolchain abstraction layer) for the `INT{32,64}_C` constant macros, leading to type mismatches between the constant macros and the rest.

Affected macros:
* `INT32_C`
* `UINT32_C`
* `INT64_C`
* `UINT64_C`

**To Reproduce**
Check out and run #44193

**Expected behavior**
The integer constants generated by the stdint `(U)INT{32,64}_C` macros should correspond to the types specified by `stdint.h` and `inttypes.h`.

**Impact**
`INT{32,64}_C` macros cannot be used.

**Logs and console output**
Building #44193 for `qemu_cortex_m3`:
```
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c: In function 'main':
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:13:9: warning: format '%d' expects argument of type 'int', but argument 2 has type 'long int' [-Wformat=]
   13 |  printk("d%" PRId32 "\n", INT32_C(32));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:15:19: note: format string is defined here
   15 | #define PRId32   "d"  /* int32_t */
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:18:9: warning: format '%i' expects argument of type 'int', but argument 2 has type 'long int' [-Wformat=]
   18 |  printk("i%" PRIi32 "\n", INT32_C(32));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:21:19: note: format string is defined here
   21 | #define PRIi32   "i"  /* int32_t */
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:23:9: warning: format '%o' expects argument of type 'unsigned int', but argument 2 has type 'long int' [-Wformat=]
   23 |  printk("o%" PRIo32 "\n", INT32_C(32));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:27:19: note: format string is defined here
   27 | #define PRIo32   "o"  /* int32_t */
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:28:9: warning: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'long unsigned int' [-Wformat=]
   28 |  printk("u%" PRIu32 "\n", UINT32_C(32));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:33:19: note: format string is defined here
   33 | #define PRIu32   "u"  /* uint32_t */
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:33:9: warning: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'long unsigned int' [-Wformat=]
   33 |  printk("x%" PRIx32 "\n", UINT32_C(32));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
[... 65 lines cut ...]
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:40:21: note: format string is defined here
   40 | #define PRIx64   "llx"  /* uint64_t */
/home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:39:9: warning: format '%llX' expects argument of type 'long long unsigned int', but argument 2 has type 'long unsigned int' [-Wformat=]
   39 |  printk("X%" PRIX64 "\n", UINT64_C(64));
      |         ^~~~
In file included from /home/stephanos/Dev/zephyrproject/zephyr/include/sys/printk.h:14,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel_includes.h:32,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/stephanos/Dev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/stephanos/Dev/zephyrproject/zephyr/tests/misc/print_format/src/main.c:7:
/home/stephanos/Dev/zephyrproject/zephyr/lib/libc/minimal/include/inttypes.h:46:21: note: format string is defined here
   46 | #define PRIX64   "llX"  /* uint64_t */
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.14.0
 - Commit SHA: 0521aa0fbce835022fbd65d0d438f4a20d5af893 (with #44193)

**Additional context**

* This problem affects almost all architectures.
* Toolchain version is not an issue here.

stephanosio, 2022-03-25:
FYI @yperess

stephanosio, 2022-03-25:
The fix is actually quite simple -- adding the following to `zephyr_stdint.h`:

```
#undef __INT32_C
#undef __UINT32_C
#undef __INT64_C
#undef __UINT64_C

#define __INT32_C(c) c
#define __UINT32_C(c) c ## U
#define __INT64_C(c) c ## LL
#define __UINT64_C(c) c ## ULL
```

yperess, 2022-03-25:
Thanks, I'll add to my PR shortly

stephanosio, 2022-03-25:
@yperess Here is 2f6c98235cfbf49e22fd5ee6b114657495680a7d.

commit fab5bb9e (2022-03-29): printk: Add tests for print format

Add tests to verify that the different print formats and constant
macros match.

Fixes #44199