zephyrproject-rtos/zephyr #44213: xtensa arch_cpu_idle not correct on cavs18+ platforms

kv2019i, 2022-03-25:
**Describe the bug**

A few issues found when comparing arch/xtensa/core/cpu_idle.c:arch_cpu_idle() against SOF mainline
implementation:

- code to set interrupt level to 5 is missing ()
           - https://github.com/thesofproject/sof/blob/main/src/arch/xtensa/include/arch/lib/wait.h
           - SOF tree commit "arch: xtensa: Support wait_for_interrupt on LX6"
- CONFIG_XTENSA_WAITI_BUG is not defined for cavs18+ platforms in Zephyr mainline
           - this is a hw property so should be defined in the soc Kconfig defaults

kv2019i, 2022-03-25:
FYI @nashif @lgirdwood

andyross, 2022-03-29:
I have a vague memory that this was a problem only on cAVS 1.5 (or maybe just on APL or something?).  It's entirely possible I was wrong.  If that's wrong then clearly we need to set that on all the affected hardware.

The interrupt level should be OK as-is though.  arch_cpu_idle() is only ever called with interrupts masked by arch_irq_lock(), whose action is precisely to set PS.INTLVL to XCHAL_EXCM_LEVEL.

kv2019i, 2022-04-08:
Ack @andyross . I submitted a patch here https://github.com/zephyrproject-rtos/zephyr/pull/44694 . Actually the workaround is needed for all LX6 platforms, e.g. cAVS1.8 and newer.

andyross, 2022-04-11:
Sounds great.  I see that merged, so closing this bug.

closed 2022-04-11 by andyross