zephyrproject-rtos/zephyr #44215: tests: subsys: cpp: over half of tests failing on macOS but do not fail on Linux

cfriedt, 2022-03-25:
**Describe the bug**
About 1/2 of `twister` jobs fail for `tests/subsys/cpp` under macOS but succeed in Linux. 

Currently, most platform testing is skipped because the only `integration_platform` listed is `mps2_an521`. I removed that because I was seeing unexpected failure while working on cfriedt/thrift-for-zephyr#67.

**Please also mention any information which could help others to understand the problem you're facing:**
** - What target platform are you using?**
 Many (`twister`)
** - What have you tried to diagnose or workaround this issue?**
Not a lot.

**To Reproduce**
Steps to reproduce the behavior:
1. apply this patch:
```diff
diff --git a/tests/subsys/cpp/cxx/testcase.yaml b/tests/subsys/cpp/cxx/testcase.yaml
index cf16410071..5610b5d035 100644
--- a/tests/subsys/cpp/cxx/testcase.yaml
+++ b/tests/subsys/cpp/cxx/testcase.yaml
@@ -1,6 +1,4 @@
 tests:
   cpp.main:
-    integration_platforms:
-      - mps2_an385
     tags: cpp
     toolchain_exclude: xcc
diff --git a/tests/subsys/cpp/libcxx/testcase.yaml b/tests/subsys/cpp/libcxx/testcase.yaml
index 2d55752e29..48e5f9741f 100644
--- a/tests/subsys/cpp/libcxx/testcase.yaml
+++ b/tests/subsys/cpp/libcxx/testcase.yaml
@@ -1,6 +1,3 @@
-common:
-  integration_platforms:
-    - mps2_an385
 tests:
   cpp.libcxx.newlib:
     filter: TOOLCHAIN_HAS_NEWLIB == 1
```
2. `twister -G -T tests/subsys/cpp`
4. See error(s)

**Expected behavior**
I would expect that `native_posix` and `native_posix_64` fail in addition to the preexisting Linux failures (`qemu_malta`, `qemu_malta_be`) because `native_posix` is not yet supported on Zephyr, but to me this indicates something else is severely wrong.

Linux:
```shell
twister -G -T tests/subsys/cpp
Renaming output directory to /home/cfriedt/workspace/zephyrproject/zephyr/twister-out.2
INFO    - Zephyr version: zephyr-v3.0.0-1888-g1568b665a8ed
INFO    - JOBS: 32
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (125 configurations) selected, 45 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   46/ 125  36%  skipped:   46, failed:    0
ERROR   - qemu_malta_be             tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     FAILED: Build failure
ERROR   - see: /home/cfriedt/workspace/zephyrproject/zephyr/twister-out/qemu_malta_be/tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano/build.log
INFO    - Total complete:   47/ 125  37%  skipped:   46, failed:    1
ERROR   - qemu_malta                tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano     FAILED: Build failure
ERROR   - see: /home/cfriedt/workspace/zephyrproject/zephyr/twister-out/qemu_malta/tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano/build.log
INFO    - Total complete:   83/ 125  66%  skipped:   48, failed:    2
ERROR   - qemu_malta                tests/subsys/cpp/libcxx/cpp.libcxx.newlib          FAILED: Build failure
ERROR   - see: /home/cfriedt/workspace/zephyrproject/zephyr/twister-out/qemu_malta/tests/subsys/cpp/libcxx/cpp.libcxx.newlib/build.log
INFO    - Total complete:   84/ 125  67%  skipped:   48, failed:    3
ERROR   - qemu_malta_be             tests/subsys/cpp/libcxx/cpp.libcxx.newlib          FAILED: Build failure
ERROR   - see: /home/cfriedt/workspace/zephyrproject/zephyr/twister-out/qemu_malta_be/tests/subsys/cpp/libcxx/cpp.libcxx.newlib/build.log
INFO    - Total complete:  125/ 125  100%  skipped:   48, failed:    4
INFO    - 73 of 125 test configurations passed (94.81%), 4 failed, 48 skipped with 0 warnings in 19.91 seconds
[... 57 lines cut ...]
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/cfriedt/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Error at /Users/cfriedt/workspace/zephyrproject/zephyr/cmake/modules/extensions.cmake:2033 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  /Users/cfriedt/workspace/zephyrproject/zephyr/cmake/modules/kernel.cmake:142 (assert)
  /Users/cfriedt/workspace/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:130 (include)
  /Users/cfriedt/workspace/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /Users/cfriedt/workspace/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/cfriedt/workspace/zephyrproject/zephyr/twister-out/mps2_an521/tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano/CMakeFiles/CMakeOutput.log".
See also "/Users/cfriedt/workspace/zephyrproject/zephyr/twister-out/mps2_an521/tests/subsys/cpp/libcxx/cpp.libcxx.newlib_nano/CMakeFiles/CMakeError.log".
```

**Environment (please complete the following information):**
 - OS: macOS 12.3 (Monterey)
 - Toolchain(s): `zephyr-sdk-0.13.1`, `zephyr-sdk-0.14.0`
 - Commit SHA or Version used: dfe05d289897e7b37b97ce503417f0d81e230b96

**Additional context**

stephanosio, 2022-03-25:
> Assertion failed: The toolchain is unable to build a dummy C file.

@cfriedt This rings a bell. Have you tried clearing the cache (`/Users/cfriedt/Library/Caches/zephyr`)?

cfriedt, 2022-03-25:
> > Assertion failed: The toolchain is unable to build a dummy C file.
> 
> @cfriedt This rings a bell. Have you tried clearing the cache (`/Users/cfriedt/Library/Caches/zephyr`)?

@stephanosio - will give that a shot, thanks

stephanosio, 2022-03-25:
I am unable to reproduce this on my M1 mac:

```
stephanos@devamac1:~/Dev/zephyrproject/zephyr$ scripts/twister -G -T tests/subsys/cpp
ZEPHYR_BASE unset, using "/Users/stephanos/Dev/zephyrproject/zephyr"
Renaming output directory to /Users/stephanos/Dev/zephyrproject/zephyr/twister-out.17
INFO    - Zephyr version: zephyr-v3.0.0-1901-geabb4e25a429
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    3/   4  75%  skipped:    0, failed:    0
INFO    - 3 of 4 test configurations passed (75.00%), 0 failed, 0 skipped with 0 warnings in 20.05 seconds
INFO    - In total 12 test cases were executed, 1 skipped on 1 out of total 448 platforms (0.22%)
INFO    - 3 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /Users/stephanos/Dev/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /Users/stephanos/Dev/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

cfriedt, 2022-03-25:
This was user error!

I went through the annoying gatekeeper UI for (what I thought was) every single binary in the SDK, but there were still binaries that it missed, annoyingly.

Strongly suggest adding the following to Release Notes / Getting Started Guide
```
find ~/zephyr-sdk-0.14.0 -type f -perm +111 -print | xargs -I % /bin/bash -c 'xattr -r -d com.apple.quarantine $(dirname %)'
```

closed 2022-03-25 by cfriedt