zephyrproject-rtos/zephyr #44231: Problems trying to configure the environment

bchunga, 2022-03-25:
Im trying to build the basic examples of zephyr but get stuck with this error. 
'''
[6/172] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (C:/zephyrproject/zephyr), build: zephyr-v3.0.0-1828-g69238dbfeeef
[155/172] Performing configure step for 'EspIdfBootloader'
FAILED: esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure C:/zephyrproject/zephyr/build/esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure
cmd.exe /C "cd /D C:\zephyrproject\zephyr\build\esp-idf\build\bootloader && "C:\Program Files\CMake\bin\cmake.exe" -GNinja -S C:/zephyrproject/modules/hal/espressif/components/bootloader/subproject -B C:/zephyrproject/zephyr/build/esp-idf/build/bootloader -DSDKCONFIG=C:/zephyrproject/zephyr/build/esp-idf/build/sdkconfig -DIDF_PATH=C:/zephyrproject/modules/hal/espressif -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DCMAKE_C_COMPILER=C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_CXX_COMPILER=C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_ASM_COMPILER=C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_SYSTEM_NAME=Generic -DPYTHON=C:/Users/USUARIO/AppData/Local/Programs/Python/Python310/python.exe && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/zephyrproject/zephyr/build/esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure"
CMake Warning at C:/zephyrproject/modules/hal/espressif/tools/cmake/idf.cmake:18 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build:
                              C:/esp-idf/esp-idf
Call Stack (most recent call first):
  C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:34 (include)

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.1.windows.2")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/USUARIO/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/esp-idf/esp-idf/tools/cmake/build.cmake:224 (depgraph_add_edge):
  Unknown CMake command "depgraph_add_edge".
Call Stack (most recent call first):
  C:/esp-idf/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:41 (project)

-- Configuring incomplete, errors occurred!
See also "C:/zephyrproject/zephyr/build/esp-idf/build/bootloader/CMakeFiles/CMakeOutput.log".
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\zephyrproject\zephyr\build'
'''

I dont know if there if there is a problem with my environment variables configuration (Windows), the ESP-IDF path configured works for other projects (with FreeRTOS).

sylvioalves, 2022-03-25:
@bchunga Have you followed https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html?

bchunga, 2022-03-25:
> @bchunga Have you followed https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html?

Changed the toolchain path, i found that in my case the route was %USERPROFILE%\.espressif\tools\tools\zephyr, but now the CMake throw this error: 

   [155/172] Performing configure step for 'EspIdfBootloader'
FAILED: esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure C:/zephyrproject/zephyr/build/esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure
cmd.exe /C "cd /D C:\zephyrproject\zephyr\build\esp-idf\build\bootloader && "C:\Program Files\CMake\bin\cmake.exe" -GNinja -S C:/zephyrproject/modules/hal/espressif/components/bootloader/subproject -B C:/zephyrproject/zephyr/build/esp-idf/build/bootloader -DSDKCONFIG=C:/zephyrproject/zephyr/build/esp-idf/build/sdkconfig -DIDF_PATH=C:/zephyrproject/modules/hal/espressif -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DCMAKE_C_COMPILER=C:/Users/USUARIO/.espressif/tools/tools/zephyr/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_CXX_COMPILER=C:/Users/USUARIO/.espressif/tools/tools/zephyr/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_ASM_COMPILER=C:/Users/USUARIO/.espressif/tools/tools/zephyr/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -DCMAKE_SYSTEM_NAME=Generic -DPYTHON=C:/Users/USUARIO/AppData/Local/Programs/Python/Python310/python.exe && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/zephyrproject/zephyr/build/esp-idf/src/EspIdfBootloader-stamp/EspIdfBootloader-configure"
CMake Warning at C:/zephyrproject/modules/hal/espressif/tools/cmake/idf.cmake:18 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build:
                              C:/esp-idf/esp-idf
Call Stack (most recent call first):
  C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:34 (include)

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.1.windows.2")
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:308 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:41 (project)

CMake Error at C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:308 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:41 (project)

CMake Error at C:/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:308 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:41 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/zephyrproject/zephyr/build/esp-idf/build/bootloader/CMakeFiles/CMakeOutput.log".
See also "C:/zephyrproject/zephyr/build/esp-idf/build/bootloader/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\zephyrproject\zephyr\build'

sylvioalves, 2022-03-25:
Add this into the environment: `set PATH=%PATH%;%ESPRESSIF_TOOLCHAIN_PATH%\bin`

tanzhilangnw, 2022-03-27:
when I tried command:west build -b nrf52dk_nrf52832 samples/boards/nrf/clock_skew it throws the problem:
```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/xp/zephyrproject/zephyr/samples/boards/nrf/clock_skew
-- Cache files will be written to: /home/xp/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/xp/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
CMake Error at /home/xp/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake:79 (find_package):
  Could not find a package configuration file provided by "Zephyr-sdk"
  (requested version 0.13.1) with any of the following names:

    Zephyr-sdkConfig.cmake
    zephyr-sdk-config.cmake

  Add the installation prefix of "Zephyr-sdk" to CMAKE_PREFIX_PATH or set
  "Zephyr-sdk_DIR" to a directory containing one of the above files.  If
  "Zephyr-sdk" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /home/xp/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/xp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/xp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:81 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
-- Configuring incomplete, errors occurred!
```
but actually I have edited the ~/.bashrc file like 
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_INSTALL_SDK_DIR=/home/xp/zephyr-sdk-0.13.2

 why would this happened?

bchunga, 2022-03-28:
@sylvioalves thanks for your help, I was able to build, the  problem: I was trying to use the same idf that I use with FreeRTOS (installed by VS Code plugin) so i think there is something different into the zephyr cmake or the idf requirements.

closed 2022-03-28 by bchunga