zephyrproject-rtos/zephyr #44244: Bluetooth: Controller: ISO BIS payload counter rollover

carlst, 2022-03-26:
**Describe the bug**
The ISO _bisPayloadCounter_ (value of `tx->payload_count` as set in `subsys/bluetooth/controller/hci/hci.c`, `hci_iso_handle()`) can rollover (e.g. after transmitting ISO data for ~10-11 minutes at 10ms rate, or 64K packets).

In `subsys/bluetooth/controller/ll_sw/lll_adv.h`, `struct lll_adv_iso_stream`
changing the line:
`uint16_t pkt_seq_num` 
to:
`uint64_t pkt_seq_num:39`
seems to resolve the issue.

**To Reproduce**
1. Use nRF52840DK
2. Remove `k_sleep(K_SECONDS(1))` in `samples/bluetooth/iso_broadcast`  and log the various `payload_count` values near the `memq_dequeue()` calls in `subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_iso.c`
3. AFter 64K packets and  `tx->payload_count` rollover, 2 packets with same _bisPayloadCounter_ occur.

**Impact**
Low priority

carlescufi, 2022-03-26:
@carlst thanks for reporting! Why don’t you open a Pull Request with the suggested fix?

carlst, 2022-03-27:
@carlescufi OK. Done. [#44244](https://github.com/zephyrproject-rtos/zephyr/issues/44244)

closed 2022-03-27 by carlst