zephyrproject-rtos/zephyr #44247: west build -b nrf52dk_nrf52832 samples/boards/nrf/clock_skew failed

tanzhilangnw, 2022-03-27:
when I tried command:west build -b nrf52dk_nrf52832 samples/boards/nrf/clock_skew it throws the problem:

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/xp/zephyrproject/zephyr/samples/boards/nrf/clock_skew
-- Cache files will be written to: /home/xp/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/xp/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
CMake Error at /home/xp/zephyrproject/zephyr/cmake/modules/verify-toolchain.cmake:79 (find_package):
  Could not find a package configuration file provided by "Zephyr-sdk"
  (requested version 0.13.1) with any of the following names:

    Zephyr-sdkConfig.cmake
    zephyr-sdk-config.cmake

  Add the installation prefix of "Zephyr-sdk" to CMAKE_PREFIX_PATH or set
  "Zephyr-sdk_DIR" to a directory containing one of the above files.  If
  "Zephyr-sdk" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /home/xp/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/xp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/xp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:81 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
-- Configuring incomplete, errors occurred!

but actually I have edited the ~/.bashrc file like
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_INSTALL_SDK_DIR=/home/xp/zephyr-sdk-0.13.2

why would this happened?

henrikbrixandersen, 2022-03-28:
Have you done `west zephyr-export`?

tanzhilangnw, 2022-03-28:
> Have you done `west zephyr-export`?
Thanks for your answer.
  I have done `west zephyr-export`.All right it's my fault，my virtual machine has no enough space so that the sdk I downed is not complete.After I larged the disk space and redownloaded the sdk,build command works well. 
But now the new problem occurs, I'm trying to flash the board with command `west flash` but it met another problem :
```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner nrfjprog
FATAL ERROR: required program nrfjprog not found; install it or add its location to PATH
```
  Do I need to install the nrfjprog to fix the problem？But I have found nrfjprog.py file in zephyrproject/zephyr/scripts/west_commands/runners/.

tanzhilangnw, 2022-03-28:
> Have you done `west zephyr-export`?

1

henrikbrixandersen, 2022-03-28:
> Do I need to install the nrfjprog to fix the problem?

Yes. Please see https://docs.zephyrproject.org/latest/boards/arm/nrf52dk_nrf52832/doc/index.html#programming-and-debugging

gl-chenxiaosheng, 2023-12-11:
> Thanks for your answer. I have done `west zephyr-export`.All right it's my fault，my virtual machine has no enough space so that the sdk I downed is not complete.After I larged the disk space and [...]

how to fix the required program nrfjprog not found? I also have this issue.

henrikbrixandersen, 2023-12-11:
> how to fix the required program nrfjprog not found? I also have this issue.

https://docs.zephyrproject.org/latest/develop/flash_debug/nordic_segger.html