zephyrproject-rtos/zephyr #44250: Can't build WiFi support on esp32, esp32s2, esp32c3

brentalanmiller, 2022-03-27:
**Describe the bug**
When trying to enable WiFi support on the Espressif line of boards, the build always fails because of mbedtls errors in the espressif hal library.

This seems to be across all supported boards, and versions of Zephyr that have support for these devices. (3.0.0, 2.7.0, etc.)

I'm building on Ubuntu 20.04. While trying to research the error, I found a [user on the zephyr-users list](https://lists.zephyrproject.org/g/users/message/2610) who had the same issue back in 2.6.0, so maybe the driver never worked correctly, or that we're both trying to enable the driver incorrectly.

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. git checkout v3.0.0
3. west update
4. west espressif update
5. west espressif install
6. export ZEPHYR_TOOLCHAIN_VARIANT="espressif"
7. export ESPRESSIF_TOOLCHAIN_PATH="/home/bmiller/.espressif/tools/zephyr"
8. west build -b esp32c3_devkitm samples/net/wifi/ -- -DCONFIG_WIFI_ESP32=y
9. The following error occurs:
```
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/bmiller/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/bmiller/zephyrproject/zephyr/samples/net/wifi
-- Zephyr version: 3.0.0 (/home/bmiller/zephyrproject/zephyr)
-- Found Python3: /home/bmiller/zephyrproject/.venv/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: esp32c3_devkitm
-- Cache files will be written to: /home/bmiller/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/bmiller/zephyrproject/zephyr/boards/riscv/esp32c3_devkitm/esp32c3_devkitm.dts
-- Generated zephyr.dts: /home/bmiller/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/bmiller/zephyrproject/zephyr/build/zephyr/dts.cmake
/home/bmiller/zephyrproject/zephyr/build/zephyr/zephyr.dts:82.37-86.8: Warning (unit_address_format): /soc/flash-controller@60002000/flash@0/partitions/partition@0000: unit name should not have leading 0s

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/bmiller/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/bmiller/zephyrproject/zephyr/boards/riscv/esp32c3_devkitm/esp32c3_devkitm_defconfig'
Merged configuration '/home/bmiller/zephyrproject/zephyr/samples/net/wifi/prj.conf'
Merged configuration '/home/bmiller/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/bmiller/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found toolchain: espressif (/home/bmiller/.espressif/tools/zephyr)
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/bmiller/.espressif/tools/zephyr/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
CMake Warning at /home/bmiller/zephyrproject/zephyr/CMakeLists.txt:775 (message):
  No SOURCES given to Zephyr library: subsys__net__pkt_filter

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/bmiller/zephyrproject/zephyr/build
-- west build: building application
[1/370] Preparing syscall dependency handling

[10/370] Generating include/generated/version.h
-- Zephyr version: 3.0.0 (/home/bmiller/zephyrproject/zephyr), build: zephyr-v3.0.0
[174/370] Building C object zephyr/CMakeFiles/zephyr.dir/home/bmiller/zephyrproject/modules/hal/espressif/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/home/bmiller/zephyrproject/modules/hal/espressif/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c.obj 
ccache /home/bmiller/.espressif/tools/zephyr/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DCONFIG_DPP -DCONFIG_ECC -DCONFIG_IEEE80211W -DCONFIG_SHA256 -DCONFIG_WPA3_SAE -DEAP_MSCHAPv2 -DEAP_PEAP -DEAP_PEER_METHOD -DESP32_WORKAROUND -DESPRESSIF_USE -DESP_PLATFORM -DESP_SUPPLICANT -DIEEE8021X_EAPOL -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DUSE_WPA2_TASK -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -D__ets__ -I/home/bmiller/zephyrproject/zephyr/kernel/include -I/home/bmiller/zephyrproject/zephyr/arch/riscv/include -I/home/bmiller/zephyrproject/zephyr/include -I/home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated -I/home/bmiller/zephyrproject/zephyr/soc/riscv/esp32c3 -I/home/bmiller/zephyrproject/zephyr/lib/util/fnmatch/. -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/include/bt -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/include/wifi -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/include/crypto -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../esp_shared/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_common/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/include/esp32c3 -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_hw_support/port/esp32c3/private_include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_hw_support/port/esp32c3 -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_hw_support/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/riscv/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/hal/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/hal/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/soc/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/soc/src/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/soc/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/driver/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/efuse/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/efuse/esp32c3/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/efuse/private_include/esp32c3 -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_system/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_event/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_timer/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_timer/private_include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/esp_netif/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/log/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/spi_flash/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/spi_flash/private_include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/../../components/bootloader_support/include -I/home/bmiller/zephyrproject/modules/hal/espressif/zephyr/esp32c3/ [... cut ...]
Cannot use wifi on the espressif line of modules

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: espressif sdk, as installed by west
 - Zephyr version v3.0.0, but similar errors in previous versions

**Additional context**
The example above shows the esp32c3 dev board, but the error is the same regardless of board used. The same errors occur on earlier versions of Zephyr as well, although some versions require the enabling of ethernet for some reason.

sylvioalves, 2022-04-14:
@brentalanmiller Thsi has been fixed already, would you retry?

jhardyswym, 2022-04-14:
@brentalanmiller there's a couple configs not set in the sample that esp boards need, try adding these to samples/wifi/prj.conf:

`
# ESP-Specific Configs
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_ECP_ALL_ENABLED=y

CONFIG_NET_L2_ETHERNET=y
`

brentalanmiller, 2022-04-14:
@sylvioalves I tried running `west update` again, rebuilt, and I got the same error as before. I checked out the main branch of zephyr (at c5b59282d6bcbcc5eedaf29b386c90c77036440b) and the error still occurs.

I even just tried installing zephyr on a new machine also running Ubuntu 20.04 and still get the error.

@jhardyswym Thanks, it looks like those options got me closer, but now I'm getting:
```
-- west build: generating a build system
-- Application: /home/bmiller/zephyrproject/zephyr/samples/net/wifi
-- Zephyr version: 3.0.0 (/home/bmiller/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.0", minimum required is "0.7.1")
-- Board: esp32c3_devkitm
-- Cache files will be written to: /home/bmiller/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/bmiller/zephyrproject/zephyr/boards/riscv/esp32c3_devkitm/esp32c3_devkitm.dts
-- Generated zephyr.dts: /home/bmiller/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/bmiller/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/bmiller/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/bmiller/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/home/bmiller/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/home/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found toolchain: espressif (/home/bmiller/.espressif/tools/zephyr)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bmiller/zephyrproject/zephyr/build
-- west build: building application
[1/79] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[2/79] Building C object zephyr/CMakeFiles/offsets.dir/arch/riscv/core/offsets/offsets.c.obj
[3/79] Generating include/generated/offsets.h
[4/79] Building ASM object zephyr/arch/arch/riscv/core/CMakeFiles/arch__riscv__core.dir/isr.S.obj
[5/79] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/device_service.c.obj
[6/79] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/devmem_service.c.obj
[7/79] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/kernel_service.c.obj
[8/79] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/shell_utils.c.obj
[9/79] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/backends/shell_uart.c.obj
[10/79] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_esp32.c.obj
[11/79] Building C object zephyr/drivers/console/CMakeFiles/drivers__console.dir/uart_console.c.obj
[12/79] Building ASM object zephyr/CMakeFiles/zephyr.dir/soc/riscv/esp32c3/soc_irq.S.obj
[13/79] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/pinmux/pinmux_esp32.c.obj
[14/79] Building ASM object zephyr/arch/arch/riscv/core/CMakeFiles/arch__riscv__core.dir/swap.S.obj
[15/79] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/malloc.c.obj
[16/79] Building C object zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/ethernet_stats.c.obj
[17/79] Building C object zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/arp.c.obj
[18/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/utils.c.obj
[19/79] Building C object zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/ethernet.c.obj
[20/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_core.c.obj
[21/79] Building C object zephyr/subsys/net/l2/wifi/CMakeFiles/subsys__net__l2__wifi.dir/wifi_mgmt.c.obj
[22/79] Building C object zephyr/subsys/net/l2/wifi/CMakeFiles/subsys__net__l2__wifi.dir/wifi_shell.c.obj
[23/79] Linking C static library zephyr/drivers/console/libdrivers__console.a
[24/79] Generating linker_zephyr_pre0.cmd
[25/79] Generating linker_zephyr_pre1.cmd
[26/79] Linking C static library zephyr/drivers/entropy/libdrivers__entropy.a
[27/79] Linking C static library zephyr/arch/arch/riscv/core/libarch__riscv__core.a
[28/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_if.c.obj
[29/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/connection.c.obj
[30/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/udp.c.obj
[31/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_shell.c.obj
[32/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_mgmt.c.obj
[33/79] Linking C static library zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a
[34/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_context.c.obj
[35/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_pkt.c.obj
[36/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/icmpv4.c.obj
[37/79] Linking C static library zephyr/lib/libc/minimal/liblib__libc__minimal.a
[38/79] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_tc.c.obj
[39/79] Linking C static library zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a
[... cut ...] wpa_supplicant/src/utils/common.c:287: more undefined references to `k_malloc' follow
/home/bmiller/.espressif/tools/zephyr/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.bfd: zephyr/libzephyr.a(wpabuf.c.obj): in function `wpabuf_alloc':
/home/bmiller/zephyrproject/modules/hal/espressif/components/wpa_supplicant/src/utils/wpabuf.c:127: undefined reference to `k_calloc'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

brentalanmiller, 2022-04-14:
I just tried the above configuration options and they *do* work with the `esp32` board. It seems the above failure only happens with the `esp32c3_devkitm` board.

brentalanmiller, 2022-04-16:
Would it make sense to open a small PR that makes those modules a dependency of the WIFI_ESP32 driver like this? https://github.com/brentalanmiller/zephyr/commit/66cd6bf57bdedd12034cb7369f8693c552b9e1a8

Also, I just came across the samples/boards/esp32/wifi_station example and I'm able to successfully compile this project across on all of three of the supported Espressif modules. (It has the configuration options that @jhardyswym mentioned, and the C3 builds successfully with that sample project.) The C3 won't actually connect to wifi with that project, but that's probably another discussion.

cfriedt, 2022-04-17:
@sylvioalves - will this require backport for v2.7.2?

sylvioalves, 2022-04-18:
I guess no. WiFi support is available only when using the `boards/esp32/wifi_station` sample as it uses `hal_espressif`calls. `samples/net/wifi` isn't integrated yet. I plan to submit this during next week, which will then make `samples/net/wifi` to work properly.

sylvioalves, 2022-04-18:
@brentalanmiller Regarding your comment about ESP32C3 not connecting to wifi station, check #43655. A few ESP32C3 boards do connect to wifi, others don't. I have been also working on getting this fixed. Is that the same issue you are facing?

brentalanmiller, 2022-04-18:
@sylvioalves, yes the c3 issue appears to be the same one in the other ticket.

brentalanmiller, 2022-04-18:
Thanks for all the help and clarification everyone, I appreciate it.

closed 2022-04-18 by brentalanmiller