zephyrproject-rtos/zephyr #44255: While thread is running [thread_state] is in _THREAD_QUEUED

jernejp21, 2022-03-28:
**Describe the bug**
While doing thread profiling I have noticed that when a thread is running, state is "queued". From profiler it is visible that thread goes from "suspended" state to "0x00" state before entering "queued" state. On thread end, state goes from "queued" to "0x00" to "suspended". I couldn't find any reference to what state is value 0x00.

HW used is nRF52840DK.

I have created a demo application which has 4 threads:
- idle (system thread)
- main is aborted after system initialisation
- ShutDnBtn (custom thread); thread is preemptive (prio=3) and runs every 50 ms
- adcSwitchBtn (custom thread); thread is preemptive (prio=4) and runs every 50 ms

This is the code inside ShutDnBtn:
```C
uint8_t cnt = 0;
uint8_t stop = 10;
while(1)
{
  while(cnt < stop)
  {
    cnt++;
  }

  k_sleep(K_MSEC(50));
}
```

**Expected behavior**
I would expect to have defined "running" state for threads. When checking *kernel_structs.h* there isn't any mention of running state.

**Impact**
When profiling OS threads it is hard to interpret when a thread is running, when is suspended or in some other state. These state names also differ from state names described in the [documentation](https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html#thread-states).

**Logs and console output**
For profiling states I used `<thread>.base.thread_state` variable.

[image]

**Environment (please complete the following information):**
 - OS: Windows.
 - Toolchain is nRF Connect SDK v1.9.0.
 - I have tried with OS versions 2.7.99 and 3.0.0 and the result is the same.

nashif, 2022-03-29:
probably need to increase size of thread_state and add more states, uint8_t is hitting the limit right now.

jernejp21, 2022-03-31:
1. I also noticed that "idle" thread never changes [thread_state]. For CPU load calculations it would be great help if you could see how much time CPU is idle (not working on other threads or IRQs) - idle thread running.
2. What is the benefit of using single bit for state? Why not use just numbers, for example 0x01 run, 0x02 suspend 0x03 queued, 0x04 dead and so on.

peter-mitsis, 2022-04-05:
Sorry for the delay ... looking into this now.

jernejp21, 2022-04-07:
@peter-mitsis Thank you for mentioning *k_current_get()*. I looked into it and found out that if I profile value of `_kernel.cpus[0].current`, I'm able to see when which thread is running and for how long. If I replace address values with thread names (done in profiling SW), I can see exact timing of running threads, CPU load and thread switching. Just what I was looking for. Thank you.

[image]

closed 2022-04-20 by nashif