zephyrproject-rtos/zephyr #44259: intel_adsp_cavs18: tests/lib/icmsg failed

LixinGuoX, 2022-03-28:
**Describe the bug**
tests/lib/icmsg failed on intel_adsp_cavs18

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p intel_adsp_cavs18 --device-testing --device-serial-pty=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh -T tests/lib/icmsg/ -vvv

**Logs and console output**
```
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Trimming 16 bytes of extended manifest
INFO:cavs-fw:Resetting HDA device
INFO:cavs-fw:Powering down DSP cores
INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
INFO:cavs-fw:Mapped 2M huge page at 0x140a00000 to contain 81920 bytes of firmware
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
INFO:cavs-fw:Wait for ROM startup
INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
INFO:cavs-fw:cAVS firmware load complete
--
Running test suite icmsg_buf
===================================================================
START - test_icmsg_buf_ut
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 9 (load/store alignment)
E:  **  PC 0xbe0117b7 VADDR 0x9e01b3c2
E:  **  PS 0x60020
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
E:  **  A0 0xbe010914  SP 0xbe018ec0  A2 0x9e01b3c2  A3 0xc8
E:  **  A4 0xe  A5 (nil)  A6 0x9e01b000  A7 0xbe017517
E:  **  A8 0xffffffff  A9 0xbe0123d4 A10 0x9e018300 A11 0xffffffff
E:  ** A12 0xbe019e60 A13 0xbe019e40 A14 0x9e01b000 A15 0x9e01b2dc
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR (nil)
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x9e01b000 (test_icmsg_buf_ut)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit 6a5520c62693

LixinGuoX, 2022-03-28:
This issue also happens on intel_adsp_cavs25.

carlescufi, 2022-03-29:
Also, could you please move the test to `test/subsys` instead of `test/lib`?

emob-nordic, 2022-03-31:
> Also, could you please move the test to `test/subsys` instead of `test/lib`?

Will move it.

> @emob-nordic why is `qemu_xtensa` excluded and why does this test fail with the adsp_cavsXX platforms?

I wasn't able to debug the issue there so i removed the platform.

emob-nordic, 2022-03-31:
I will have a look into this issue and into `qemu_xtensa`. I suspect there could be a problem with alignment.

PR #44401 (2022-04-19): tests: Fix alignment
@LixinGuoX could you verify if this fixes #44259?
closed 2022-04-19 by nashif