zephyrproject-rtos/zephyr #44261: ACRN: twister: some changes make test cases work abnormally.

Zhaoningx, 2022-03-28:
**Describe the bug**
Some test cases failed on the ACRN platform, the first bad commit is 340da2ecf3122a1d0b3031f3b4708754e49485fa. I checked the log of a failure case, it showed me `PROJECT EXECUTION SUCCESSFUL`, but the case was marked as a failed case after the commit above got merged. Using command `west`, it works to test.

**To Reproduce**
twister -p acrn_ehl_crb --device-testing -T tests/kernel/mem_protect/mem_map/  --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11-2" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11-2" -j 1 -v

**see error:**

INFO    - Zephyr version: zephyr-v3.0.0-1803-g340da2ecf312
INFO    - JOBS: 1
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 0 configurations discarded due to filters.

Device testing on:

| Platform     | ID   | Serial device   |
|--------------|------|-----------------|
| acrn_ehl_crb |      |                 |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/4 acrn_ehl_crb              tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map.x86_64.coverage.exec SKIPPED (filter)
INFO    - 2/4 acrn_ehl_crb              tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map.x86_64.coverage SKIPPED (filter)
INFO    - 3/4 acrn_ehl_crb              tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map.x86_64 FAILED **Testsuite mismatch (device 68.132s)**
ERROR   - see: /home/ningzhax/zephyrproject/zephyr/twister-out/acrn_ehl_crb/tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map.x86_64/handler.log
INFO    - 4/4 acrn_ehl_crb              tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map SKIPPED (filter)

INFO    - 0 of 4 test configurations passed (0.00%), 1 failed, 3 skipped with 0 warnings in 83.09 seconds
INFO    - In total 8 test cases were executed, 24 skipped on 1 out of total 448 platforms (0.22%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board        | ID   |   Counter |
|--------------|------|-----------|
| acrn_ehl_crb |      |         1 |

INFO    - Saving reports...
INFO    - Writing xunit report /home/ningzhax/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ningzhax/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

**the handle.log:**

```
...
E: RIP: 0x00000000001006dc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_k_mem_map_guard_after)
Caught system error -- reason 0
PASS - test_k_mem_map_guard_after in 0.9 seconds
===================================================================
START - test_k_mem_map_exhaustion
Free memory: 7114752
E: insufficient virtual address space (requested 12288)
Mapped 1256 pages
Free memory now: 1970176
Unmapped 1256 pages
Free memory now: 7114752
PASS - test_k_mem_map_exhaustion in 0.44 seconds
===================================================================
Test suite test_mem_map succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

Environment:
repo: github main branch
commit: 340da2ecf3122a1d0b3031f3b4708754e49485fa
zephyr SDK: 13.2

yerabolu, 2022-03-29:
This issue happens on multiple platforms like mec17x, intel-adsp, sam_e70b_xplained.

chen-png, 2022-03-29:
this is due to that the judgement process of test pass/failure is changed, now the **twister will check the test suite name whether is correct or not**, this is the reason why you can see log "Testsuite mismatch".
so please see the complete results in handler.log of this test to check whether is a line " Running test suite mem_map", and check whether the suite name is the same as the test you tested.

Zhaoningx, 2022-03-29:
Hi @spoorthy, it seems this is not a twister issue, as @chen-png said,  the twister judgment process of test pass/failure is changed, it will match the test suite name for the result log, and I find that some test cases log is not complete(the first part missing) on ACRN platform, so all of them will be marked as failed cases whether the case is successful to test.

gopiotr, 2022-03-29:
> I find that some test cases log is not complete(the first part missing) on ACRN platform

@Zhaoningx - what is the reason than `ACRN` platform do not print all logs?

Let's assume (hypothetically) that we will remove verification of test suite name (which was introduced by me). If platform is not able to print all logs, then Twister are not able to detect all performed testcases (especially this one from the beginning). So in final report you will not have information about all performed testcases. Moreover if first testcase will fail and this will cause, that all test suite will fail (`PROJECT EXECUTION FAILED` will be printed) you will not able to see what is the source of problem (due to the lacks in logs at the beginning). This can lead to misleading situations, when you could observe, that all testcases in report passed, but test suite will be marked as failed (similar to current situation). Do you think that, this is correct and desirable behavior? Perhaps correct logs printing should be fixed at first - especially that current Twister version are strongly based on correctly printed output logs and this is the fundament of any performed pass or fail verification. Without assumption of correctly printed output we will never be able to be sure that all testcases was performed correctly. 

How in your opinion should looks like verification process? Should it based only on final `PROJECT EXECUTION FAILED` or `PROJECT EXECUTION SUCCESSFUL` status? This approach is insufficient in many cases and in the past was a source of many problems - especially when some hardware got stuck and still printed `PROJECT EXECUTION SUCCESSFUL` status whether or not a test pass or fail. That was source of many "false negatives" and "false positives" cases.

If the `ACRN` platform are not able to correctly print all logs how are you able to run some tests from `samples` directory, which demand for verification multiline regex? Like for example here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/philosophers/sample.yaml
I asked, because even if "test suite name verification" will be removed, you could still have a problem with such tests, which demands full output logs...

gopiotr, 2022-03-29:
> This issue happens on multiple platforms like mec17x, intel-adsp, sam_e70b_xplained.

@yerabolu could you explain it better? Which exactly tests fail? What is the output logs? Is this the same problem with partially printed logs by hardware like in @Zhaoningx case, or rather problem with output parsing performed by Twister?

Zhaoningx, 2022-03-30:
Hi @gopiotr, I totally agree with the changes you introduced, it's insufficient to only verify on final PROJECT EXECUTION FAILED or PROJECT EXECUTION SUCCESSFUL status. Just as I said above, I think maybe this is not twister's issue but ACRN's. So I think we should close this and report another issue(why the log print is not complete by the ACRN platform).

Zhaoningx, 2022-03-30:
Closed this issue and report another issue: #[44347](https://github.com/zephyrproject-rtos/zephyr/issues/44347)

closed 2022-03-30 by Zhaoningx